libfive/libfive #129: Bounds Problem

tir-kaval, 2018-02-15:
### Description

I am getting unexpected bounds using ```shape-find-bounds```.

When I run this code on the command line:

    (let* ((profile (taper-x-y
                     (rectangle #[-1/8 0] #[1/8 7/20])
                    #[0 0] 7/20
                    1/2 1))
           (solid (move
                   (rotate-z
                    (extrude-z profile -1/24 1/24)
                    (- (* 5 (/ pi 8))))
                   #[99/100 1/2 0]))
           (bounds (shape-find-bounds solid)))
     (display bounds) (newline)
     (shape->mesh solid "solid.stl" 10 '((-2 . 2) (-2 . 2) (-2 . 2))))

 I get

    (#[-1.409487924475718e38 
        -3.402797707725092e38 
        -0.041656494140625] . 
      #[1.409487924475718e38 
         3.402797707725092e38
         0.041656494140625])

The mesh looks OK when I look at in in a viewer.

I would expect tighter bounds.

### Versions

- Operating system: Ubuntu 14.04
- `libfive` commit hash: fa02db2
  - Is this the latest commit to `master`?  yes
- Qt version (if applicable): N/A

mkeeter, 2018-02-16:
I mean, those bounds are *technically* correct 😛 

Yeah, this is a common issue with the bounds-tracking, which uses interval math; intervals tend to explode if you have many nested operations, even if they're correlated, e.g. `(X + Y) * (X - Y)`.

If you could reduce this to a more minimal example, I can take a look, but no guarantees – this algorithm tends to be fragile.  You could also look at artificially clamping the bounds (e.g. by taking the intersection with a box, which should make `shape-bounds` happier).

tir-kaval, 2018-02-16:
If I already HAD a bounding box to intersect it with, why would I call ```shape-bounds``` ? :smiley:

Humor aside, this thought led me to the idea that it might be better to calculate the bounds at a higher level, i.e., transform the Scheme expression to a bounding box calculated at the Scheme level. If, as you said, there are inherent numerical difficulties doing it at the tree level, it might be worth investigating doing it "top-down"...

One technique that could be used in Scheme/Lisp (and potentially integrating [model validation](https://github.com/libfive/libfive/issues/128) is generic functions.  If you are familiar with the visitor pattern, what I had in mind is similar:

    (define-generic shape-eval)
    ...
    (shape-eval shape bounding-box-context) => <bounding-box>
    (shape-eval shape mesh-context) => <mesh>
    (shape-eval shape validate-context) => #t

In this example, each shape would then implement 3 specialized ```shape-eval``` methods--a shape "specification".  This technique might be outside the scope of libfive, but I thought I'd mention it.

For now, I'll just avoid ```shape-bounds```.

closed 2018-02-16 by tir-kaval