libfive/libfive #134: Another shape cut off by autobounds

omgitsraven, 2018-02-18:
```
(sequence
  (sphere 0.8)
  (twirl-x 3 1 #[0 0 -1])
  (rotate-z 0.3)
  (move #[0.3 1 1.2])
)
```

Is a shape being cut off a case where it's still worth reporting autobounds problems, or is autobounds basically completely at the limit of what it can be used for at this point?

omgitsraven, 2018-02-18:
Maybe off-topic, but: it's my understanding that auto-bounds works by transforming a bounding box successively, right? Is there any possibility of trying alternate approaches, like doing a very low resolution pass on the full extent of the manual bounds, then bringing it in to roughly the size of the shape that emerges (plus a margin)? (And re-extending that margin every time a higher-resolution pass adds geometry too close to the existing margin? Is it difficult to just re-calculate whatever's in the new area without having to recalculate the rest of the shape along with it?)

mkeeter, 2018-02-20:
Like issue #129, it looks like the bounds are *technically* correct, but are too big to be useful:

```
(define s (sequence
  (sphere 0.8)
  (twirl-x 3 1 #[0 0 -1])
  (rotate-z 0.3)
  (move #[0.3 1 1.2])
))
(shape-find-bounds s)
--> (#[-1289.328125 -1945.703125 -11.2574462890625] .
     #[120.2314453125 3729.46875 11.8204345703125])
```

I'm going to disable autobounding for the moment, to prevent people from running into this and being confused.

The bounds code emphasizes catching everything, at the cost of being *very* conservative.  For example, if you've got a model that's mostly centered around the origin, but has a single sphere floating far away in space, it will detect that:
```
(shape-find-bounds
  (union (sphere 1 [1 1 1])
         (box [-2 -2 -2] [1 1 1])
         (sphere 0.2 [1032 500 3])))
--> (#[-2.00006103515625 -2.00006103515625 -2.00006103515625] .
     #[1032.203125 500.19921875 3.20001220703125])
```

I'm going to make a general-purpose tracking issue for "shapes that give too-large bounds", and accumulate things there.

closed 2018-02-20 by mkeeter

mkeeter, 2018-02-20:
FYI, I wrote a little about how the bounding box estimation works [here](http://www.mattkeeter.com/blog/2017-09-29-bounds/).