libfive/libfive #136: shape-find-bounds is too conservative

mkeeter, 2018-02-20:
This is a tracking issue for cases where automatic bounds detection gives technically correct, but too-large-to-be-useful results.

From issue #134:
```
(define s (sequence
  (sphere 0.8)
  (twirl-x 3 1 #[0 0 -1])
  (rotate-z 0.3)
  (move #[0.3 1 1.2])
))
``` 

From issue #129:
```
(let* ((profile (taper-x-y
                 (rectangle #[-1/8 0] #[1/8 7/20])
                #[0 0] 7/20
                1/2 1))
       (solid (move
               (rotate-z
                (extrude-z profile -1/24 1/24)
                (- (* 5 (/ pi 8))))
               #[99/100 1/2 0]))
       (bounds (shape-find-bounds solid)))
```

omgitsraven, 2018-02-20:
Should the inability to use `blend-poly` with autobounds count as an example? https://github.com/libfive/libfive/issues/105 (Not a particularly urgent one since `blend-expt` works fine, but maybe a useful test case...)

omgitsraven, 2018-02-20:
Here's a freebie:
```
(sequence
  (sphere 1)
  (rotate-x 1)
  (rotate-y 1)
  (rotate-z 1)
)
```
This is the one I'm running into all the time; anything that I want to apply an arbitrary rotation to, I wind up having to go `rotate-x`, `rotate-y`, `intersection` with a `box`, `rotate-z`

sj-muelmont, 2018-02-22:
Possibly relevant:

[What every...floating point](http://citeseerx.ist.psu.edu/viewdoc/summary?doi=10.1.1.22.6768) p. 191, "Rounding Modes".

It looks like you are sampling or "fishing" for the bounds after creating the op tree.  Why not have a transform of the shape s-exp which returns the bounds rather than a tree?  Each shape would have to have an auxiliary function to return it's bounds, among other mods, but it should give you better bounds.  Note that even this most likely won't give you "exact" bounds.  My $.02.

sj-muelmont, 2018-02-23:
So I see you are using Boost interval, with "rounded_transc_std" policy and a type "float".  Goldberg suggests towards the end of the "Rounding Modes" section using an adaptive approach.  If you get an unreasonable interval, calculate it at increasingly higher precision until the interval is "acceptable".

omgitsraven, 2018-02-24:
Also, if we're proposing alternate methods: I'm still kind of partial to the idea of just doing a low resolution pass on the default bounds, and bringing in the auto bounds to fit around whatever turns up, plus a margin... the only complication might be expanding that margin if higher-resolution passes reveal details that come too close to the existing margins (I don't know if it would be possible to render only the new area, or if it would necessitate repeating the whole render every time that it realises the margins were too tight...)

mkeeter, 2019-08-04:
While doing more experiments with interval math, I found yet another way that bounds tracking was incorrect.  I've decided to remove it entirely (7520c6c4d1ead4bde5ed545f721d7d1053222305), rather than leaving a half-baked implementation in the codebase.  It's in the Git history for folks to dig up, if anyone else wants to take up the fight.

closed 2019-08-04 by mkeeter