libfive/libfive #137: Save As dialog appears *behind* the Studio window

doug-moen, 2018-02-21:
### Description

[Description of the issue]
Sometimes, the Save As dialog (and the Open dialog) pop up behind the Studio window, where they cannot be seen.

[Steps to reproduce]
1. Start Studio
2. Open a file, using File >> Open...
3. Save the file using File >> Save as...
The Save as... dialog window will be invisible, because it will open behind the Studio window.

[Are there any error messages printed to the console?]
no

### Versions

- Operating system: Ubuntu 16.04
- `libfive` commit hash: a4cc6c71966466a0722f36e4333364d285679843
  - Is this the latest commit to `master`?  [yes]
- Qt version (if applicable): 5.9.2

mkeeter, 2018-02-21:
This seems suspiciously similar to [this issue](https://github.com/notepadqq/notepadqq/issues/52), so I've applied the same fixes that they did.  Can you see if this helps?

doug-moen, 2018-02-21:
That appears to work.

closed 2018-02-21 by doug-moen