libfive/libfive #138: Bad stack trace/error message from Guile

doug-moen, 2018-02-21:
### Description

[Description of the issue]
I have a program that doesn't work, that I cannot debug.
This issue is about the fact that the Studio window does not display a proper stack trace. All it is telling me is "something went wrong", but it won't tell me where in the program the error was detected.

[Steps to reproduce]
Here is the program. It is gibberish; I reduced my program to the smallest code example that demonstrates the problem.
```
(define (outsideness a amin amax)
  (if (< a amin)
    (- amin a)
    (if (> a amax)
      (- a amax)
      0))
)
(define (recta lo hi)
  (lambda-shape (x y z)
    (let* ((ind (max (- (.x lo) x) (- x (.x hi)) (- (.y lo) y) (- y (.y hi))))
           ;(ox (outsideness x (.x lo) (.y hi)))
           )
      ind
    )
))

(intersection (box #[-10 -10 0] #[10 10 1])
(recta #[0 1] #[10 10])
)
```
If you delete the `;` character from `;(ox (outsideness x (.x lo) (.y hi)))` then I get an error message in the bottom of the Studio window:
```
wrong-type-arg: Wrong type argument in position 1: #<<shape> 34c9c20>

Stack trace:
0: (intersection (box (vec3 -10 -10 0) (vec3 10 10 1)) (recta (vec2 0 1) (vec2 10 10)))
```
The so called "Stack trace" actually gives zero information as to where in the program the error was detected. I've tested the "outsideness" function in a regular Guile interpreter and it works. I have no clue how to debug this further.

[What did you expect to happen?  What happened?  All the time?]
I would like a better error message indicating what went wrong, and specifically where in the program the error occurred.

[Are there any error messages printed to the console?]
no

### Versions

- Operating system: Ubuntu 16.04
- `libfive` commit hash: a4cc6c71966466a0722f36e4333364d285679843
  - Is this the latest commit to `master`?  [yes]
- Qt version (if applicable): 5.9.2
- Guile version: 2.2.3

### Common questions
[Delete this section if not relevant]

If this is a behavior issue, does the test suite (`libfive/test/libfive-test`) pass?
yes

If there are error messages in the console, have you searched for them and followed up on any promising leads?
* "Wrong argument type in position 1" apparently means something like (f x) where f is not a function.
* So maybe lambda-shape fails to capture the value of the binding "outsideness"? If so, that's a bug. Scheme is a lexically scoped language, and that would indicate a failure of lexical scoping.
* But the bug I'm reporting is the lack of a meaningful stack trace and the lack of a line number that would indicate which line of code is failing.

doug-moen, 2018-02-21:
The problem is related to calling a function (in this case `outsideness`) with `x` as an argument, inside of `lambda-shape`.

I think I may recognize the problem from my work on Curv. Distance functions are compiled into a representation that can execute on the GPU, and within a distance function, only a subset of the full Curv language is supported. If a language construct that is not within this subset is used, you get an error message that identifies the span of characters containing the bad code and a stack trace. I looked at your xtree evaluator a bit, and I hypothesize that the same kind of thing is happening, except that (at least in this case) a meaningful error message is not being generated by whatever code is translating Scheme into your internal expression tree format.

I want to write a complicated distance function, but apparently I can't factor my code into functions, so I just need to inline expand all of my code into an arithmetic expression.

doug-moen, 2018-02-21:
Inline expanding my function call didn't help. Here's what I've got now:
```
(define (rect lo hi)
  (lambda-shape (x y z)
    (let* ((a (- lo (vec2 x y)))
           (b (- (vec2 x y) hi))
           (ind (max (max (.x a) (.y a)) (max (.x b) (.y b))))
           (ox (if (< x (.x lo))
                 (- (.x lo) x)
                 (if (> x (.y hi))
                   (- x (.y hi))
                   0)))
           (oy (if (< y (.y lo))
                 (- (.y lo) y)
                 (if (> y (.x hi))
                   (- y (.x hi))
                   0)))
           (outd (norm (vec2 ox oy))))
      (if (<= ind 0)
        ind
        outd
      )
    )
  )
)
```
Maybe `if` expressions are not supported inside `lambda-shape`? There seems to be no IF in `opcode.hpp`, so maybe that's it. Also, I don't see any conditionals in the standard shape library.

mkeeter, 2018-02-21:
Okay, there are three different things going on here that are being conflated.

You're correct that there's a compilation step going on, and the full language isn't supported.  Like you found out, conditionals aren't supported, and neither are comparison operators.  This isn't going to change in the near future, because I'm compiling everything to two-argument functions in a flat tree.

The second issue is that error messages aren't useful.  I've fixed this in the commit above, so your original shape now says:
```
In function <:
wrong-type-arg: Wrong type argument in position 1: #<<shape> 114a26bc0>
```
which tells you that `wrong-type-arg` is being thrown by the comparison operator `<`.

The third issue is that backtraces aren't useful.  I agree, and will keep this open as a tracking issue for improvement.