libfive/libfive #139: Studio randomly dumps core on startup

doug-moen, 2018-02-21:
### Description

[Description of the issue]
Sometimes when I start Studio, it dumps core immediately. Most of the time, it starts up okay.

[Steps to reproduce]
It's happened twice, but I cannot reproduce it on demand.

[Are there any error messages printed to the console?]
Here's a stack trace from gdb:
```
#0  sysmalloc (nb=nb@entry=1638416, av=av@entry=0x7f285a7f0b20 <main_arena>)
    at malloc.c:2738
#1  0x00007f285a4ae743 in _int_malloc (
    av=av@entry=0x7f285a7f0b20 <main_arena>, bytes=bytes@entry=1638400)
    at malloc.c:3827
#2  0x00007f285a4b0184 in __GI___libc_malloc (bytes=1638400) at malloc.c:2913
#3  0x00007f2858f0bba9 in ?? () from /usr/lib/nvidia-384/libGL.so.1
#4  0x00007f2854ef2808 in ?? ()
   from /usr/lib/nvidia-384/libnvidia-glcore.so.384.111
#5  0x00007f2854f67766 in ?? ()
   from /usr/lib/nvidia-384/libnvidia-glcore.so.384.111
#6  0x00007f285531998c in ?? ()
   from /usr/lib/nvidia-384/libnvidia-glcore.so.384.111
#7  0x00007f28552f0146 in ?? ()
   from /usr/lib/nvidia-384/libnvidia-glcore.so.384.111
#8  0x00007f2858eacb09 in ?? () from /usr/lib/nvidia-384/libGL.so.1
#9  0x00007f2858edbc61 in ?? () from /usr/lib/nvidia-384/libGL.so.1
#10 0x00007f2858ee0cd8 in ?? () from /usr/lib/nvidia-384/libGL.so.1
#11 0x00007f2858ed94d6 in glXCreateContextAttribsARB ()
   from /usr/lib/nvidia-384/libGL.so.1
#12 0x00007f284b9e9cd6 in ?? ()
   from /usr/local/Qt5.9.2/5.9.2/gcc_64/plugins/xcbglintegrations/libqxcb-glx-integration.so
#13 0x00007f284b9e7d37 in ?? ()
   from /usr/local/Qt5.9.2/5.9.2/gcc_64/plugins/xcbglintegrations/libqxcb-glx-integration.so
#14 0x00007f2852c57f61 in QXcbIntegration::createPlatformOpenGLContext(QOpenGLContext*) const ()
   from /usr/local/Qt5.9.2/5.9.2/gcc_64/plugins/platforms/../../lib/libQt5XcbQpa.so.5
#15 0x00007f285b93a58d in QOpenGLContext::create() ()
   from /usr/local/Qt5.9.2/5.9.2/gcc_64/lib/libQt5Gui.so.5
#16 0x00007f285c0ff5fe in QWidgetPrivate::shareContext() const ()
   from /usr/local/Qt5.9.2/5.9.2/gcc_64/lib/libQt5Widgets.so.5
#17 0x00007f285c12e899 in ?? ()
   from /usr/local/Qt5.9.2/5.9.2/gcc_64/lib/libQt5Widgets.so.5
#18 0x00007f285c12ed3a in QOpenGLWidget::resizeEvent(QResizeEvent*) ()
   from /usr/local/Qt5.9.2/5.9.2/gcc_64/lib/libQt5Widgets.so.5
#19 0x00007f285c111152 in QWidget::event(QEvent*) ()
   from /usr/local/Qt5.9.2/5.9.2/gcc_64/lib/libQt5Widgets.so.5
#20 0x00007f285c0d45ec in QApplicationPrivate::notify_helper(QObject*, QEvent*)
    () from /usr/local/Qt5.9.2/5.9.2/gcc_64/lib/libQt5Widgets.so.5
#21 0x00007f285c0dba17 in QApplication::notify(QObject*, QEvent*) ()
   from /usr/local/Qt5.9.2/5.9.2/gcc_64/lib/libQt5Widgets.so.5
#22 0x00007f285b3102b8 in QCoreApplication::notifyInternal2(QObject*, QEvent*)
    () from /usr/local/Qt5.9.2/5.9.2/gcc_64/lib/libQt5Core.so.5
#23 0x00007f285c109e02 in QWidgetPrivate::sendPendingMoveAndResizeEvents(bool, bool) () from /usr/local/Qt5.9.2/5.9.2/gcc_64/lib/libQt5Widgets.so.5
#24 0x00007f285c10d843 in QWidgetPrivate::show_helper() ()
   from /usr/local/Qt5.9.2/5.9.2/gcc_64/lib/libQt5Widgets.so.5
#25 0x00007f285c110505 in QWidget::setVisible(bool) ()
   from /usr/local/Qt5.9.2/5.9.2/gcc_64/lib/libQt5Widgets.so.5
#26 0x00007f285c10d7f8 in QWidgetPrivate::showChildren(bool) ()
   from /usr/local/Qt5.9.2/5.9.2/gcc_64/lib/libQt5Widgets.so.5
#27 0x00007f285c10d85f in QWidgetPrivate::show_helper() ()
   from /usr/local/Qt5.9.2/5.9.2/gcc_64/lib/libQt5Widgets.so.5
#28 0x00007f285c110505 in QWidget::setVisible(bool) ()
   from /usr/local/Qt5.9.2/5.9.2/gcc_64/lib/libQt5Widgets.so.5
#29 0x00007f285c10d7f8 in QWidgetPrivate::showChildren(bool) ()
   from /usr/local/Qt5.9.2/5.9.2/gcc_64/lib/libQt5Widgets.so.5
#30 0x00007f285c10d85f in QWidgetPrivate::show_helper() ()
   from /usr/local/Qt5.9.2/5.9.2/gcc_64/lib/libQt5Widgets.so.5
#31 0x00007f285c110505 in QWidget::setVisible(bool) ()
   from /usr/local/Qt5.9.2/5.9.2/gcc_64/lib/libQt5Widgets.so.5
#32 0x00000000004597ae in Window::Window (this=0x7fff07987580, target=...)
    at /home/doug/res/src/libfive/studio/src/window.cpp:188
#33 0x0000000000422bf3 in App::App (this=0x7fff07987570, argc=<optimized out>, 
    argv=0x7fff079876d8) at /home/doug/res/src/libfive/studio/src/app.cpp:26
#34 0x000000000042041c in main (argc=1, argv=0x7fff079876d8)
    at /home/doug/res/src/libfive/studio/src/main.cpp:45
```

### Versions

- Operating system: Ubuntu 16.04
- `libfive` commit hash: a4cc6c71966466a0722f36e4333364d285679843
  - Is this the latest commit to `master`?  [yes]
- Qt version (if applicable): 5.9.2
- Nvidia proprietry driver: 384

### Common questions

If this is a behavior issue, does the test suite (`libfive/test/libfive-test`) pass?
yes

If this is a graphics issue on Linux, report the results of updating your graphics drivers (and switching to binary graphics drivers if applicable).
I appear to be running the latest Nvidia driver in the Ubuntu repositories.

If there are error messages in the console, have you searched for them and followed up on any promising leads?
Haven't found anything so far.

mkeeter, 2018-02-21:
This seems similar to #42.

@nickodell, did you ever get farther into debugging this?

nickodell, 2018-02-21:
Nope, never figured it out. It does look like a similar problem, though. In each stack trace, there's a malloc being called from a graphics driver.

Maybe the problem is thread safety? Not sure.

mkeeter, 2018-02-21:
The only thing I can think of is the Guile interpreter starting up at the same time as the graphics driver is being initialized, and some kind of thread safety issue (like @nickodell said).  If they can interfere with each other, it seems like a bug somewhere else in the stack, but we could try to mitigate it.

Can you try switching the order of `show()` and `interpreter->start()` [here](https://github.com/libfive/libfive/blob/master/studio/src/window.cpp#L186-L188) and see if that makes any difference?

nickodell, 2018-02-23:
Here's a stacktrace from a recent version. Unfortunately, I seem to have lost the scripts I was using for gdb.

```
Studio: malloc.c:2406: sysmalloc: Assertion `(old_top == initial_top (av) && old_size == 0) || ((unsigned long) (old_size) >= MINSIZE && prev_inuse (old_top) && ((unsigned long) old_end & (pagesize - 1)) == 0)' failed.

Thread 6 "QThread" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffda7a9700 (LWP 8032)]
0x00007ffff699fab0 in GC_register_disappearing_link_inner ()
   from /usr/lib/x86_64-linux-gnu/libgc.so.1
(gdb) bt
#0  0x00007ffff699fab0 in GC_register_disappearing_link_inner ()
   from /usr/lib/x86_64-linux-gnu/libgc.so.1
#1  0x00007ffff6cc334e in weak_set_add_x (set=set@entry=0x555555b75f60, 
    hash=1069761575586958091, hash@entry=534880787793479045, 
    pred=pred@entry=0x7ffff6cae6d0 <symbol_lookup_predicate_fn>, 
    closure=closure@entry=0x555555cbc9a0, obj=obj@entry=0x555555cbc9a0)
    at weak-set.c:580
#2  0x00007ffff6cc38e0 in scm_c_weak_set_add_x (set=<optimized out>, 
    raw_hash=raw_hash@entry=534880787793479045, 
    pred=pred@entry=0x7ffff6cae6d0 <symbol_lookup_predicate_fn>, 
    closure=0x555555cbc9a0, obj=0x555555cbc9a0) at weak-set.c:775
#3  0x00007ffff6cae4b5 in scm_i_str2symbol (str=0x555555cbc9e0)
    at symbols.c:254
#4  0x00007ffff6caeca3 in scm_from_utf8_symboln (
    sym=sym@entry=0x7ffff6ce1a30 <s_scm_open_input_string> "open-input-string", len=17, len@entry=18446744073709551615) at symbols.c:543
#5  0x00007ffff6caecbc in scm_from_utf8_symbol (
    sym=sym@entry=0x7ffff6ce1a30 <s_scm_open_input_string> "open-input-string")
    at symbols.c:526
#6  0x00007ffff6c57c3e in create_subr (define=define@entry=1, 
    name=name@entry=0x7ffff6ce1a30 <s_scm_open_input_string> "open-input-string", nreq=nreq@entry=1, nopt=nopt@entry=0, rest=rest@entry=0, 
    fcn=fcn@entry=0x7ffff6cac790 <scm_open_input_string>, generic_loc=0x0)
   ubr.c:249
#7  0x00007ffff6c57ebd in scm_c_define_gsubr (name=name@entry=0x7ffff6ce1a30 <s_scm_open_input_string> "open-input-string", req=req@entry=1, opt=opt@entry=0, rst=rst@entry=0, 
    fcn=fcn@entry=0x7ffff6cac790 <scm_open_input_string>) at gsubr.c:344
#8  0x00007ffff6cacb14 in scm_init_strports () at ../libguile/strports.x:4
#9  0x00007ffff6c5d786 in scm_i_init_guile (base=<optimized out>) at init.c:436
#10 0x00007ffff6cb0068 in scm_i_init_thread_for_guile (base=0x7fffda7a8c28, dynamic_state=0x0) at threads.c:585
#11 0x00007ffff6cb0405 in scm_init_guile () at threads.c:622
#12 0x000055555559e95e in _Interpreter::init (this=0x555555acbab0) at /home/nick/libfive/studio/src/interpreter.cpp:41
#13 0x00007ffff5bcc95e in QMetaObject::activate(QObject*, int, int, void**) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#14 0x00007ffff59d0d8f in ?? () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#15 0x00007ffff6f2e494 in start_thread (arg=0x7fffda7a9700) at pthread_create.c:333
#16 0x00007ffff5150aff in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:97
(gdb) bt
#0  0x00007ffff699fab0 in GC_register_disappearing_link_inner () from /usr/lib/x86_64-linux-gnu/libgc.so.1
#1  0x00007ffff6cc334e in weak_set_add_x (set=set@entry=0x555555b75f60, hash=1069761575586958091, hash@entry=534880787793479045, pred=pred@entry=0x7ffff6cae6d0 <symbol_lookup_predicate_fn>, 
    closure=closure@entry=0x555555cbc9a0, obj=obj@entry=0x555555cbc9a0) at weak-set.c:580
#2  0x00007ffff6cc38e0 in scm_c_weak_set_add_x (set=<optimized out>, raw_hash=raw_hash@entry=534880787793479045, pred=pred@entry=0x7ffff6cae6d0 <symbol_lookup_predicate_fn>, closure=0x555555cbc9a0, 
    obj=0x555555cbc9a0) at weak-set.c:775
#3  0x00007ffff6cae4b5 in scm_i_str2symbol (str=0x555555cbc9e0) at symbols.c:254
#4  0x00007ffff6caeca3 in scm_from_utf8_symboln (sym=sym@entry=0x7ffff6ce1a30 <s_scm_open_input_string> "open-input-string", len=17, len@entry=18446744073709551615) at symbols.c:543
#5  0x00007ffff6caecbc in scm_from_utf8_symbol (sym=sym@entry=0x7ffff6ce1a30 <s_scm_open_input_string> "open-input-string") at symbols.c:526
#6  0x00007ffff6c57c3e in create_subr (define=define@entry=1, name=name@entry=0x7ffff6ce1a30 <s_scm_open_input_string> "open-input-string", nreq=nreq@entry=1, nopt=nopt@entry=0, rest=rest@entry=0, 
    fcn=fcn@entry=0x7ffff6cac790 <scm_open_input_string>, generic_loc=0x0) at gsubr.c:249
#7  0x00007ffff6c57ebd in scm_c_define_gsubr (name=name@entry=0x7ffff6ce1a30 <s_scm_open_input_string> "open-input-string", req=req@entry=1, opt=opt@entry=0, rst=rst@entry=0, 
    fcn=fcn@entry=0x7ffff6cac790 <scm_open_input_string>) at gsubr.c:344
#8  0x00007ffff6cacb14 in scm_init_strports () at ../libguile/strports.x:4
#9  0x00007ffff6c5d786 in scm_i_init_guile (base=<optimized out>) at init.c:436
#10 0x00007ffff6cb0068 in scm_i_init_thread_for_guile (base=0x7fffda7a8c28, dynamic_state=0x0) at threads.c:585
#11 0x00007ffff6cb0405 in scm_init_guile () at threads.c:622
#12 0x000055555559e95e in _Interpreter::init (this=0x555555acbab0) at /home/nick/libfive/studio/src/interpreter.cpp:41
#13 0x00007ffff5bcc95e in QMetaObject::activate(QObject*, int, int, void**) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#14 0x00007ffff59d0d8f in ?? () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#15 0x00007ffff6f2e494 in start_thread (arg=0x7fffda7a9700) at pthread_create.c:333
#16 0x00007ffff5150aff in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:97
```

doug-moen, 2018-02-24:
If my core dump was caused by a thread safety issue, there should be a second thread that is interfering with the thread that caused the core dump. Using the original core dump, I found a candidate for this second thread. It is suggestive that two threads are making calls related to memory allocation, while all of the remaining threads are quiescent.
```
#0  0x00007f2859d92aea in GC_base () from /usr/lib/x86_64-linux-gnu/libgc.so.1
#1  0x00007f285c87d7dd in weak_set_add_x (set=set@entry=0x2585f60, 
    hash=8482915901750041685, hash@entry=4241457950875020842, 
    pred=pred@entry=0x7f285c868520 <symbol_lookup_predicate_fn>, 
    closure=closure@entry=0x29242c0, obj=obj@entry=0x29242c0) at weak-set.c:580
#2  0x00007f285c87de00 in scm_c_weak_set_add_x (set=<optimized out>, 
    raw_hash=raw_hash@entry=4241457950875020842, 
    pred=pred@entry=0x7f285c868520 <symbol_lookup_predicate_fn>, 
    closure=0x29242c0, obj=0x29242c0) at weak-set.c:775
#3  0x00007f285c868305 in scm_i_str2symbol (str=0x7f282ed984d0)
    at symbols.c:254
#4  0x00007f285c87a641 in vm_regular_engine (thread=0x2925000, vp=0x2626f30, 
    registers=0x0, resume=43139776) at vm-engine.c:2301
#5  0x00007f285c87c6b2 in scm_call_n (proc=proc@entry=0x2919ce0, 
    argv=argv@entry=0x0, nargs=nargs@entry=0) at vm.c:1257
#6  0x00007f285c7fc619 in scm_call_0 (proc=proc@entry=0x2919ce0) at eval.c:481
#7  0x00007f285c81c7bb in load_thunk_from_memory (
    data=0x7f282ece4000 "\177ELF\002\001\001\377", len=1142597, 
    is_read_only=<optimized out>) at loader.c:463
#8  0x00007f285c87963b in vm_regular_engine (thread=0x2925000, vp=0x2626f30, 
    registers=0x0, resume=43139776) at vm-engine.c:784
#9  0x00007f285c87c6b2 in scm_call_n (proc=proc@entry=0x2911a00, 
    argv=argv@entry=0x0, nargs=nargs@entry=0) at vm.c:1257
#10 0x00007f285c7fc619 in scm_call_0 (proc=proc@entry=0x2911a00) at eval.c:481
#11 0x00007f285c86b5a9 in catch (tag=tag@entry=0x404, thunk=0x2911a00, 
    handler=0x29119e0, pre_unwind_handler=0x4) at throw.c:137
#12 0x00007f285c86b915 in scm_catch_with_pre_unwind_handler (
    key=key@entry=0x404, thunk=<optimized out>, handler=<optimized out>, 
    pre_unwind_handler=<optimized out>) at throw.c:254
#13 0x00007f285c86bacf in scm_c_catch (tag=tag@entry=0x404, 
    body=body@entry=0x7f285c81a930 <do_load_thunk_from_file>, 
    body_data=body_data@entry=0x2911a20, 
    handler=handler@entry=0x7f285c81a940 <load_thunk_from_file_catch_handler>, 
    handler_data=handler_data@entry=0x2911a20, 
    pre_unwind_handler=pre_unwind_handler@entry=0x0, 
    pre_unwind_handler_data=0x0) at throw.c:377
#14 0x00007f285c81b1f7 in try_load_thunk_from_file (filename=0x2911a20)
    at load.c:611
#15 load_thunk_from_path (filename=filename@entry=0x2911a60, 
    source_file_name=source_file_name@entry=0x2911a40, 
    source_stat_buf=source_stat_buf@entry=0x7f283d52bbe0, 
    found_stale_file=found_stale_file@entry=0x7f283d52bbdc) at load.c:754
#16 0x00007f285c81b65b in scm_primitive_load_path (args=<optimized out>)
    at load.c:1199
#17 0x00007f285c87963b in vm_regular_engine (thread=0x2925000, vp=0x2626f30, 
    registers=0x0, resume=43139776) at vm-engine.c:784
#18 0x00007f285c87c6b2 in scm_call_n (proc=0x28bfba0, argv=argv@entry=0x0, 
    nargs=nargs@entry=0) at vm.c:1257
#19 0x00007f285c7fc619 in scm_call_0 (proc=<optimized out>) at eval.c:481
#20 0x00007f285c81b680 in scm_primitive_load_path (args=<optimized out>)
    at load.c:1251
#21 0x00007f285c87963b in vm_regular_engine (thread=0x2925000, vp=0x2626f30, 
    registers=0x0, resume=43139776) at vm-engine.c:784
#22 0x00007f285c87c6b2 in scm_call_n (proc=0x28b9c60, argv=argv@entry=0x0, 
    nargs=nargs@entry=0) at vm.c:1257
#23 0x00007f285c7fc619 in scm_call_0 (proc=<optimized out>) at eval.c:481
#24 0x00007f285c81b680 in scm_primitive_load_path (args=<optimized out>)
    at load.c:1251
#25 0x00007f285c87963b in vm_regular_engine (thread=0x2925000, vp=0x2626f30, 
    registers=0x0, resume=43139776) at vm-engine.c:784
#26 0x00007f285c87c6b2 in scm_call_n (proc=0x288e270, argv=argv@entry=0x0, 
    nargs=nargs@entry=0) at vm.c:1257
#27 0x00007f285c7fc619 in scm_call_0 (proc=<optimized out>) at eval.c:481
#28 0x00007f285c81b680 in scm_primitive_load_path (args=<optimized out>)
    at load.c:1251
#29 0x00007f285c87963b in vm_regular_engine (thread=0x2925000, vp=0x2626f30, 
    registers=0x0, resume=43139776) at vm-engine.c:784
#30 0x00007f285c87c6b2 in scm_call_n (proc=0x26511e0, 
[... 20 lines cut ...]
    cproc=cproc@entry=0x7f285cfa4520 <init_libfive_vec(void*)>, 
    cdata=cdata@entry=0x0) at fluids.c:546
#38 0x00007f285c81fda5 in scm_c_call_with_current_module (
    module=module@entry=0x2896f00, 
    func=func@entry=0x7f285cfa4520 <init_libfive_vec(void*)>, 
    data=data@entry=0x0) at modules.c:129
#39 0x00007f285c81fe38 in scm_c_define_module (
    name=name@entry=0x7f285cfb281f "libfive vec", 
    init=init@entry=0x7f285cfa4520 <init_libfive_vec(void*)>, 
    data=data@entry=0x0) at modules.c:186
#40 0x00007f285cfa5979 in scm_init_libfive_modules ()
    at /home/doug/res/src/libfive/build/libfive/bind/bundle.cpp:1879
#41 0x000000000044ac33 in _Interpreter::init (this=0x24a4a80)
    at /home/doug/res/src/libfive/studio/src/interpreter.cpp:43
#42 0x00007f285b33ab96 in QMetaObject::activate(QObject*, int, int, void**) ()
   from /usr/local/Qt5.9.2/5.9.2/gcc_64/lib/libQt5Core.so.5
#43 0x00007f285b13bc77 in QThread::started(QThread::QPrivateSignal) ()
   from /usr/local/Qt5.9.2/5.9.2/gcc_64/lib/libQt5Core.so.5
#44 0x00007f285b1416f1 in ?? ()
   from /usr/local/Qt5.9.2/5.9.2/gcc_64/lib/libQt5Core.so.5
#45 0x00007f285a2166ba in start_thread (arg=0x7f283d52d700)
    at pthread_create.c:333
#46 0x00007f285a53341d in clone ()
    at ../sysdeps/unix/sysv/linux/x86_64/clone.S:109
```

mkeeter, 2018-02-24:
Interesting, that agrees with my theory about the Guile interpreter somehow interfering with the graphics system on startup.  @doug-moen , can you try the change I suggested [here](https://github.com/libfive/libfive/issues/139#issuecomment-367514213) and see if that helps?