libfive/libfive #141: Compiling libfive on Centos 7.4 gives compiler error.

sj-muelmont, 2018-02-22:
### Description

Compiling libfive on Centos 7.4 gives compiler error.  Is the version of gcc too old?

### Versions

- Operating system: CentOS Linux release 7.4.1708
- `libfive` commit hash: 7d069a6
  - Is this the latest commit to `master`?  yes
- Qt version (if applicable): N/A
- eigen3, version 3.2.5
- Boost version: 1.66.0
- gcc version 4.8.5 20150623 (Red Hat 4.8.5-16) (GCC)

### Build output

```
[  5%] Building CXX object libfive/src/CMakeFiles/five.dir/eval/eval_interval.cpp.o
In file included from ./libfive/libfive/src/../include/libfive/eval/tape.hpp:30:0,
                 from ./libfive/libfive/src/../include/libfive/eval/base.hpp:20,
                 from ./libfive/libfive/src/eval/base.cpp:21:
./libfive/libfive/src/../include/libfive/render/brep/region.hpp: In instantiation of ‘Kernel::Region<N>::Region() [with unsigned int N = 3u]’:
./libfive/libfive/src/../include/libfive/eval/tape.hpp:120:45:   required from here
./libfive/libfive/src/../include/libfive/render/brep/region.hpp:59:33: error: call of overloaded ‘Array(int)’ is ambiguous
     Region() : lower(0), upper(0) {}
```

mkeeter, 2018-02-22:
Does the commit above fix this issue?

sj-muelmont, 2018-02-22:
It fixes that particular error, however, there are about ~600 lines of error messages remaining.  I still think it is a problem with the gcc version.  Do you recommend a minimum gcc to try?

sj-muelmont, 2018-02-22:
Hmm...I am able to compile it on Ubuntu 14.04 with gcc 4.8.4.  There goes that idea.

sj-muelmont, 2018-02-22:
Compiling is currently broken on Ubuntu 14.04 too.

mkeeter, 2018-02-23:
Okay, fixed those compilation errors too.

sj-muelmont, 2018-02-23:
Ubuntu 14.04 is now compiling.

However, looks like there are even more errors on CentOS 7.4:

[CentOS-7.4-broken-2.txt](https://github.com/libfive/libfive/files/1752315/CentOS-7.4-broken-2.txt)

(In the past, I have been able to compile libfive on CentOS 7.4.)

mkeeter, 2018-02-23:
Is it possible that you've regressed in terms of what Eigen version you're building against?

Those errors are all in parts of code that haven't changed recently, and all of the errors are issues in Eigen, which makes me suspicious.

For reference, I'm building against Eigen 3.3.4.

sj-muelmont, 2018-02-23:
Upgrading from Eigen 3.2.5 to 3.3.4 on CentOS allowed it compile--your suspicions were correct.

Maybe there ought to be a warning about old versions of Eigen in the README.me?

Thanks!

closed 2018-02-23 by sj-muelmont

mkeeter, 2018-02-24:
Yup, I'm adding version checking at the CMake level [here](https://github.com/libfive/libfive/commit/5050cc0b32eab1ba962af14a1dee88aefc16064f).

I'm not sure exactly what version of Eigen to bound against, but will use 3.3.0 as a reasonable guess.