libfive/libfive #142: Request for Documentation

sj-muelmont, 2018-02-22:
### Description

The purpose of the abstraction "tape" in the C++ code is unclear.  Could you please document it?

sj-muelmont, 2018-02-23:
A tape of op codes...don't tell me there is a Turing machine in there too.

mkeeter, 2018-02-23:
Nope, it's not Turing-complete 😃 

In brief, the tape contains a bunch of commands of the form
"Evaluate `opcode` on memory locations `a` (and perhaps `b`), storing the result in `id`".

I don't have time to write a detailed doc right now; do you have a more specific question about what it's doing?

sj-muelmont, 2018-02-23:
So the motivation for using the tape class is efficiency, or grouping everything together?  I didn't understand why you wouldn't just do a post-order traversal on the op tree to evaluate.  If you don't have time for documenting, that's cool.  Thanks.

closed 2018-02-26 by sj-muelmont