libfive/libfive #144: 'Eigen/Eigen' file not found

Chrismomo, 2018-02-24:
### Description

While `make` in the build folder, the it shows `'Eigen/Eigen' file not found`. I'm sure that the Eigen has been installed in my computer under `/usr/local/include` shown as pic below: 

[image]

mkeeter, 2018-02-24:
What is the exact log message?  What happens if you run with `VERBOSE=1 make` to see the actual commands that are being run?

I installed Eigen through homebrew, and it was installed in `/usr/local/include/eigen3/`, rather than `/usr/local/include/Eigen`; are you sure that you're using Eigen 3.3.0 or later?

Chrismomo, 2018-02-25:
The log message can seen in the following screenshots. And I've also installed Eigen through homebrew, but because of the log message mentioning `Eigen/Eigen`, I tried to rename the folder under `/usr/local/include/`. It worked the same when its name was eigen3.

👇After running `make`
[image]

👇After running `make VERBOSE=1`
[image]

mkeeter, 2018-02-25:
Oh, it looks like you've run CMake against the wrong directory.  Your build directory should be at 
`/Users/xinyuehou/Desktop/fab/libfive/build`, rather than 
`/Users/xinyuehou/Desktop/fab/libfive/libfive/build`

Chrismomo, 2018-02-26:
Yep! It has successfully built on my computer!!!
Thanks!!!!!  I'm gonna to play with libfive now~~~

closed 2018-02-26 by Chrismomo