libfive/libfive #149: extend libfive API: convert distance function to mesh

doug-moen, 2018-03-01:
I would like to use libfive to create a mesh using a signed distance function defined by Curv. But I can't use the existing libfive API, because I can't construct a tree. The available tree opcodes are too limited to support Curv.

I'd like to extend the existing libfive API so that I can use an external signed distance function with the signature
     float distance(float x,float y,float z)
to construct a mesh. Since we are using C++, a more flexible interface is an object of type
     std::function(float(float,float,float))

Maybe I can do this by adding a new opcode, or a new type of tree node, that contains a signed distance function. This is the approach I am currently investigating, but do you have a better idea? Or suggestions on how to implement this?

omgitsraven, 2018-03-01:
Maybe I'm misunderstanding, but is that not what `lambda-shape` does?

doug-moen, 2018-03-01:
I'm talking about the C++ API to libfive. I'm not using Guile. I want to use an arbitrary C++ function as a signed distance function. As I understand it, lambda-shape converts its argument to a libfive tree, but only a small subset of the Guile language is supported, and that subset is too small for my application.

omgitsraven, 2018-03-01:
Ahh, whoops, sorry.

sj-muelmont, 2018-03-01:
What ops are you missing?  What shapes are you trying to represent?

Are you primarily trying to overcome these limitations (conditionals and comparisons missing):

Issue #44

> lambda-shape isn't powerful enough to evaluate arbitrary Scheme code;
> it's intended to JIT a (basically) arithmetic expression. In this
> case, the issue is that lambda-shape doesn't accept conditionals,
> since the branches are evaluated lazily rather than eagerly.

Issue #138

> You're correct that there's a compilation step going on, and the full
> language isn't supported. Like you found out, conditionals aren't
> supported, and neither are comparison operators. This isn't going to
> change in the near future, because I'm compiling everything to
> two-argument functions in a flat tree.

doug-moen, 2018-03-01:
I am working on Curv (github.com/doug-moen/curv), which is another 3D modelling program that uses signed distance fields. I am currently adding STL export. Matt has put multiple years of work into his meshing algorithm, so I want to try using the libfive C++ library to construct meshes for Curv. I also intend to try dual contouring on the GPU, and see how well that works for me.

Curv has a larger set of operations that can be used in distance functions. It has conditionals, which I use extensively, but it also has `for` loops and `while` loops, which I use for fractals (represented as IFS or iterated function systems), fractal noise, and Voronoi patterns.

The libfive C++ API requires you to construct a tree of arithmetic operations to represent the distance function. I can't compile Curv programs directly into a libfive tree because of the missing operations that exist in Curv but not libfive. So I'm looking for a way around that.

At this point, I've discovered that there is a tree node type called an "oracle", but I don't understand how I would use it to interface to a signed distance function.

sj-muelmont, 2018-03-01:
@doug-moen thanks for the info.  I have seen Curv, another interesting project!  I worked on a port of libfive (I've been using it since it was Ao) to Common Lisp, primarily for mechanical CAD.  I mostly know how it works from a high level, but I have been looking into doing extensions like generalized sweeps--so I am curious how about how others want to extend it.

mkeeter, 2018-03-01:
@doug-moen You're looking in the right place!  The Oracle API lets you plug arbitrary black-box functions into the math tree – in this case, something that evaluates the Curv program.

To use it, you'll need to make a derived class that implements all of the `virtual` functions [here](https://github.com/libfive/libfive/blob/master/libfive/include/libfive/eval/oracle.hpp).  You could also start with the [`OracleStorage` helper class](https://github.com/libfive/libfive/blob/master/libfive/include/libfive/eval/oracle_storage.hpp), which handles data storage.

It's not quite as simple as `std::function<float(float,float,float)>` (which is the `evalPoint` function in the `Oracle` class).  There are a few other types of evaluation that are required:
- For efficient spatial pruning, the oracle must be able to perform interval arithmetic.
- To preserve sharp edges, the oracle must handle a few functions related to normals.

@sj-muelmont Generalized sweeps are a very interesting application of oracles, which I'm just now starting to think about.  I suspect that if you have a parameterized curve `x(t), y(t), z(t)` and an oracles that solves for the closest point (i.e. `oracle(x,y,z) -> t`), you'd be most of the way there...

doug-moen, 2018-03-01:
@sj-muelmont Generalized sweeps are one of the things I want to add to Curv. This appears to be expensive to implement using F-Rep. Looks like the sweep function needs to perform iterative root finding (using Newton's method, etc) on the shape being swept. I haven't tried coding this yet. My current plan is to generalize Curv to a hybrid system that supports both discrete signed distance fields (as voxel arrays) and continuous signed distance fields (as functions). It looks like generalized sweeps should be cheaper to implement on voxel arrays.

doug-moen, 2018-03-01:
@mkeeter said "For efficient spatial pruning, the oracle must be able to perform interval arithmetic."

Curv doesn't do interval arithmetic. I really just have a signed distance to work with. I guess I can just create an infinite interval as output. Actually, I also have a bounding box for the shape, maybe I can make use of this somehow. Actually, the distance function is assumed to be Lipschitz(1), maybe that can also be used to constrain the interval.

"To preserve sharp edges, the oracle must handle a few functions related to normals."

I can use the usual numeric technique to compute a normal from a signed distance function.

If oracles turn out to be infeasible, I guess I can also try to construct an octree directly, then feed that into the rest of the meshing pipeline.

sj-muelmont, 2018-03-01:
I have been reading a book called "Generative Modeling" by Snyder, who created a language called GENMOD to model sweeps using interval analysis.  Maybe not directly applicable, but I thought I would start there.

doug-moen, 2018-03-01:
Snyder's thesis is here: https://thesis.library.caltech.edu/2865/1/Snyder_jm_1991.pdf

It's a B-rep system. Booleans are complex and expensive, but sweeps are easier than with signed distance functions.

doug-moen, 2018-03-02:
@sj-muelmont
I use conditionals in my logarithmic spiral, and also in the implementations of square, cube, cone, and probably other shapes with sharp edges.

For the latter cases, I try to provide two implementations of each shape with sharp edges: one implementation has a "mitred" distance field (which doesn't require conditionals), and the other implementation has an "exact" or Euclidean distance field (which does require conditionals). Libfive apparently only supports the mitred versions. Both variants have identical boundaries, but the distance field has a different structure. This is important for operations like offset, blend, morph and loft, which use the structure of the distance field to add or subtract material from a shape. If you use the mitred versions, you get creases and ridges in the output. If you use the exact versions, these creases and ridges are smoothed out, which is often exactly what you want when blending. An offset of an 'exact' cube has rounded corners (a rounded offset), while an offset of a 'mitred' cube has sharp corners (a mitred offset).

omgitsraven, 2018-03-02:
Ahh, I've been wanting a word to describe the difference between "keeps its creases when expanded" and "gets blobby when expanded", thank you! That was one of the first things I noticed when working with libfive; the box function I originally ported over was exact, but the box function that libfive comes with is mitred (I didn't realise it came with shapes at first, heh...)

doug-moen, 2018-03-02:
@omgitsraven If you have an exact box function that works in libfive, can you share the code?

omgitsraven, 2018-03-02:
Was just getting it ready :) I'm sure this could be written more succinctly, but for now: [mitre.zip](https://github.com/libfive/libfive/files/1773370/mitre.zip)

Got the function from here: http://iquilezles.org/www/articles/distfunctions/distfunctions.htm

sj-muelmont, 2018-03-02:
@doug-moen: Snyder's book is a follow up to his PhD work.  I might be wrong, but so far I wouldn't characterize GENMOD as just "brep", it is a high-level generative language for model representation.  Synder makes use of breps to render his models--but so does libfive. :)  I think he still works at Microsoft research, I guess we could ask him...

I thought it would be interesting to study what does support sweeps before trying to implement it in a system that doesn't so far.  The book to me is also and interesting study of graphical language design and modeling systems.

doug-moen, 2018-03-02:
There are two kinds of geometric representation, B-Rep and Volumetric.

I define a B-Rep (boundary representation) as one that defines a shape by specifying its boundary. This includes triangle meshes, splines, and parametric equations that map one or more parameters onto points on the boundary of the shape. For B-Rep, function representation uses parametric equations, like Snyder's system.

I define a volumetric representation as one that maps an (x,y,z) point onto information specifying whether the point is outside, on the boundary, or inside the shape. Signed Distance Fields are an interesting subset of volumetric representations. For V-Rep, function representations are often called implicit functions (as opposed to parametric). There are also discrete voxel-based V-Reps.

B-Rep and V-Rep have different strengths and weaknesses. In V-Rep, boolean operations are cheap and simple. Eg, you can use `max` and `min` for `intersection` and `union` of signed distance functions. In B-Rep, boolean operations are expensive and complicated. Finding the boundary of a shape is cheap and easy in B-Rep, but is expensive in V-Rep (you have to search through space to find the boundary, eg using Sphere Tracing in Curv, or using recursive subdivision and interval arithmetic in Libfive. Converting between B-Rep and V-Rep, in either direction, is complicated and expensive.

Sweeping a shape appears to be easier in B-Rep, based on the fact that the only readily available systems I know with this operation are B-Rep systems, but I'm still studying this.

Here's a paper from 2005 that discusses how to implement sweeps for V-Rep systems. I've only skimmed this, I haven't extracted a practical implementation from it:
http://unknownroad.com/publications/SweepsTRApril2005.pdf

doug-moen, 2018-03-06:
@mkeeter Here's what I've got so far. It's an attempt to write an Oracle that queries a signed distance function. This code worked for the first example I tried, got an assertion in Eigen for the second example. If the code is wrong, can you point out why it is wrong, and suggest things I could do to fix it?
```
struct CurvOracle : public OracleStorage<>
{
    // The signed distance function.
    float distance(Eigen::Vector3f p);

    // Evaluate signed distance function at a specified (x,y,z) point.
    void evalPoint(float& out, size_t index) override
    {
        Eigen::Vector3f p = {
            points(0, index),
            points(1, index),
            points(2, index),
        };
        out = distance(p);
    }

    // Evaluate distance function over an interval using interval arithmetic.
    void evalInterval(Interval::I& out) override
    {
        float infinity = 1.0f / 0.0f;
        out = Interval::I(-infinity, infinity);
    }

    // Given a set of points, return true or false for each point
    // indicating whether it is 'ambiguous', whatever that means.
    // If a point is ambiguous, then a safer/slower path is used during
    // mesh generation.
    void checkAmbiguous(
            Eigen::Block<Eigen::Array<bool, 1, LIBFIVE_EVAL_ARRAY_SIZE>,
                         1, Eigen::Dynamic> out) override
    {
        // All points are ambiguous. What else can I do?
        out = true;
    }

    // Return a list of gradients at the specified point.
    // For ambiguous points, only one gradient is returned.
    void evalDerivs(
            Eigen::Block<Eigen::Array<float, 3, Eigen::Dynamic>,
                         3, 1, true> out, size_t index=0) override
    {
        // The specified point.
        Eigen::Vector3f p = points.col(index);

        // The gradient at p.
        float d = distance(p);
        float dx = distance({p[0]+0.001f, p[1], p[2]});
        float dy = distance({p[0], p[1]+0.001f, p[2]});
        float dz = distance({p[0], p[1], p[2]+0.001f});
        Eigen::Vector3f gradient {
            (dx - d) / 0.001f,
            (dy - d) / 0.001f,
            (dz - d) / 0.001f
        };

        out = gradient;
    }

    // Return the set of 'Features' at a specified point.
    //
    // Currently I return no features.
    // It is technically legal to return exactly one Feature which consists
    // of simply the gradient at the specified point. Is this better?
    void evalFeatures(
            boost::container::small_vector<Feature, 4>& out) override
    {
    }
};
```

mkeeter, 2018-03-07:
Okay, a few things that need fixing before I debug more deeply:
- `isAmbiguous` should do nothing, since Curv doesn't have special-cases for C1-discontinues (which is what "ambiguous" means in this context).
- `evalFeatures` should return a single feature with the gradient and no epsilons, as you suggested.