libfive/libfive #154: Providing mitred vs exact shape functions

omgitsraven, 2018-03-07:
As discussed here https://github.com/libfive/libfive/issues/149#issuecomment-369778457 , there are two ways to represent a shape with sharp edges in a signed distance function, which look the same by themselves but give different results when used in operations like `offset` and `blend`. 

I think it's worth providing both versions in the shape reference, for people to use as appropriate; the question is, which one should be the *default*? Do we want `box` and `box-mitred`, or `box` and `box-exact`? (Or `box-mitred` and `box-exact`, with no generic `box` at all, to remove ambiguity?)

Personally I'm leaning towards having the exact version be the default -- I'd usually expect things like `offset` and `blend` to make things blobbier, which is what happens with exact functions.

doug-moen, 2018-03-09:
In Curv, I chose to make `box` without qualifications exact, and `box.mitred` is the mitred version, with the same rationale that @omgitsraven gave.

Other operations that are sensitive to the structure of the distance field: `loft`, `morph` and `shell`.

mkeeter, 2018-03-09:
API-wise, I'd suggest having `box-exact` and `box-mitered`, with `box` as an alias for `box-mitered`. The more common use for boxes is just basic CSG, and it makes sense to make that as fast as possible (since the mitered box has fewer clauses).

closed 2018-05-23 by omgitsraven