libfive/libfive #156: Opcodes clash with Windows macros CONST, EXP, MIN, MAX, MOD

igfraga, 2018-03-10:
Opcodes clash with Windows macros CONST, EXP, MIN, MAX, MOD

Unfortunately Windows.h defines these and many more ugly macros.
So including Windows.h before opcode.hpp will prevent the code from compiling.
Because of this we are forced to always include libfive at the very top of our source files.

In our codebase at nTopology we prefix all enums with "e_" to avoid this. This is just one possible solution of course.

mkeeter, 2018-03-13:
Can you see if the `rename-opcodes` branch works?

I couldn't find a full list of the `Windows.h` macros, but renamed all of the opcodes that looked suspicious.

speps, 2018-03-13:
Did you check if defining NOMINMAX and WIN32_LEAN_AND_MEAN also fix the problem?

igfraga, 2018-03-14:
The rename-opcodes branch works!

NOMINMAX and WIN32_LEAN_AND_MEAN are not enough (CONST for example seems to get in there with Windows.h)

mkeeter, 2018-03-14:
Great, merged.

closed 2018-03-14 by mkeeter