libfive/libfive #166: "pointer being freed was not allocated" when Mesh gets destructed

Ahbee, 2018-04-19:
### Description

Hello I am trying to mesh a sdf that I have using libfive, the meshing worked , but I get a nasty crash when the Mesh object goes out of scope,  saying "pointer being freed was not allocated" 
here is the sample code to reproduce 

```c++
#include <libfive/eval/oracle_storage.hpp>
#include <libfive/tree/oracle_clause.hpp>
#include <libfive/render/brep/mesh.hpp>

class SDFOracle:public Kernel::OracleStorage<>
{
    
public:
    SDFOracle(std::function<float(float,float,float)> sdf):mSdf(sdf){};
    
    void evalPoint(float& out, size_t index)override
    {
        out = mSdf(points(0, index),points(1, index),points(2, index));
    };
    
    void evalInterval(Kernel::Interval::I& out) override
    {
        
        auto size = upper - lower;
        auto center = (upper + lower)/2.f;
        auto val = mSdf(center[0],center[1],center[2]);
        auto dig = (std::sqrt(size[0] * size[0] + size[1]*size[1] * size[2] * size[2]))/2.f;
        out = {val - dig, val+dig};
    }
    
    void evalFeatures(boost::container::small_vector<Kernel::Feature, 4>& out) override
    {
        const float delta = .001f;
        auto val = mSdf(points(0, 0), points(1, 0),points(2, 0));
        
        Eigen::Vector3f dx{points(0, 0) - delta, points(1, 0),points(2, 0)};
        Eigen::Vector3f dy{points(0, 0), points(1, 0) - delta, points(2, 0)};
        Eigen::Vector3f dz{points(0, 0), points(1, 0), points(2, 0) - delta};

        
        Eigen::Vector3f n {val - mSdf(dx[0],dx[1],dx[2]),
                           val - mSdf(dy[0],dy[1],dy[2]),
                           val - mSdf(dz[0],dz[1],dz[2])};
        out.push_back(n);
    }
    void checkAmbiguous(Eigen::Block<Eigen::Array<bool, 1, LIBFIVE_EVAL_ARRAY_SIZE>,
                        1, Eigen::Dynamic> out) override{}
private:
    std::function<float(float,float,float)> mSdf;
};

class SDFOracleClause : public Kernel::OracleClause
{
public:
    SDFOracleClause(std::function<float(float,float,float)> sdf):mSdf(sdf){};
    std::unique_ptr<Kernel::Oracle> getOracle() const override
    {
        return std::make_unique<SDFOracle>(mSdf);
    };
    std::string name() const override
    {
        return "SDFOracle";
    };
    
private:
   std::function<float(float,float,float)> mSdf;
};

static void writeMesh(std::string path,
                      std::function<float(float,float,float)> sdf)
{
    auto tree = Kernel::Tree(std::make_unique<SDFOracleClause>(sdf));
    
    auto lower = Eigen::Array<double, 3, 1>{-10,-10,-10};
    auto upper = Eigen::Array<double, 3, 1>{10,10,10};
    
    auto box = Kernel::Region<3>{lower,upper};
    
    auto mesh = Kernel::Mesh::render(tree,box,.5,
                                     1.0e-8,
                                     true);
    
    mesh->saveSTL(path);
    
    
}

int main(int argc, const char * argv[]){
    
    auto sphereFunc = [](float x,float y,float z)
    {
        return std::sqrt(x*x + y*y + z*z) - 5;
    };
    
    writeMesh("sphereMesh.stl", sphereFunc);
    
    return 0;
}

``` 

### Versions

i am on mac os 13.3 , I tried all different c++ compliers in xcode and all crashed.
Thanks for any help!

### Additional Information
here is the call stack 
[image]

Ahbee, 2018-04-20:
I have 
```
#define EIGEN_WORLD_VERSION 3
#define EIGEN_MAJOR_VERSION 3
#define EIGEN_MINOR_VERSION 90
```

Ahbee, 2018-04-20:
I'll also try to see if i can reproduce the error without libfive

Ahbee, 2018-05-12:
If I build lib5 as a static lib, and use it I do not get this error? It only happens if I build lib5 as dylib. so it might mean that the dylib version is using a different `free` or `malloc` implementation than my app. But i dont know how to change or specify that within xCode.

mkeeter, 2018-05-12:
I haven't ever tried building through XCode; I'm just using the command-line tools as described [here](https://github.com/libfive/libfive#mac).

FunMiles, 2020-05-26:
I have tried this and can reproduce a similar thing with the simplest C++ code.
I am interested in using this library for my C++ project and created the simplest project on MacOS using CLion. 
Here is the code:

```Cpp

#include <libfive.h>
#include <libfive/render/brep/mesh.hpp>
#include <libfive/render/brep/settings.hpp>

using namespace libfive;
namespace Kernel = libfive;
int main()
{
	auto x = Kernel::Tree::X();
	auto y = Kernel::Tree::Y();
	auto z = Kernel::Tree::Z();

// Arithemetic is overloaded for the Kernel::Tree type
	auto out = (x * x) + (y * y) + (z * z) - 1;

// Pick the target region to render
	auto bounds = Region<3>({-2, -2, -2}, {2, 2, 2});

// Mesh::render returns a unique_ptr, so it cleans up automatically
	Mesh::render(out, bounds,BRepSettings{})->saveSTL("sphere.stl");
	return 0;
}

```

And a CMakeLists.txt:
```
cmake_minimum_required(VERSION 3.16)
project(LibFiveTest)

set(CMAKE_CXX_STANDARD 17)

find_path(LIBFIVE_INCLUDE libfive.h PATH_SUFFIXES include HINTS $ENV{LIBFIVE_INSTALL_PATH})
find_library(LIBFIVE_LIB five PATH_SUFFIXES lib HINTS $ENV{LIBFIVE_INSTALL_PATH})

message("Libfive: ${LIBFIVE_INCLUDE} and ${LIBFIVE_LIB}")
find_package(Eigen3 3.3 REQUIRED NO_MODULE)
find_package(Boost)

add_executable(LibFiveTest main.cpp)
target_include_directories(LibFiveTest PRIVATE ${LIBFIVE_INCLUDE} ${Boost_INCLUDE_DIRS})
target_link_libraries(LibFiveTest ${LIBFIVE_LIB} Eigen3::Eigen)
```

As a side note: I wish the documentation were better for use as a C++ library. Is there a reason not to use Doxygen?
I can also help add the capability of creating the cmake-importable `find_package(libfive)` if that is of interest to others.