libfive/libfive #171: kernel::findBounds for oracle does not work

Ahbee, 2018-05-11:
calling `kernel::findBounds`  runs forever when i try to call it with a tree that has an oracle.
This same tree meshes fine if i pass an overestimated bounds.

also when the oracles `evalInterval(Kernel::Interval::I& out)`  gets called, the  
`out` interval has infinity range, what do I return in that case?

mkeeter, 2018-05-11:
`evalInterval` should populate `out` with a [interval](https://en.wikipedia.org/wiki/Interval_arithmetic) that contains the range of output values that are possible, given the range of input values stored by [`Oracle::set`](https://github.com/libfive/libfive/blob/master/libfive/include/libfive/eval/oracle.hpp#L50-L51).

Slightly rephrasing, `Oracle::set` says "We'll be evaluating this spatial region", and `evalInterval` should return a range that includes every possible output value that you can get in that region.

`findBounds` uses `evalInterval`'s results to give a best-guess estimate; it's not guaranteed to work, and definitely *won't* work if `evalInterval` isn't giving proper results.  Still, it probably shouldn't run forever.

Ahbee, 2018-05-12:
I have a true signed distance function, in order  to get the interval Im finding the sdf at the center of the box then adding and subtracting the distance to the corner from the center. `findBounds` does not run forever but it takes a while then returns in infinite box, here is sample code, thanks!

```c++
#include <libfive/eval/oracle_storage.hpp>
#include <libfive/tree/oracle_clause.hpp>
#include <libfive/render/brep/mesh.hpp>
#include <libfive/solve/bounds.hpp>

class SDFOracle:public Kernel::OracleStorage<>
{
    
public:
    SDFOracle(std::function<double(double,double,double)> sdf):mSdf(sdf){};
    
    void evalPoint(float& out, size_t index)override
    {
        out = mSdf(points(0, index),points(1, index),points(2, index));
    };
    
    void evalInterval(Kernel::Interval::I& out) override
    {
        
        auto size = upper - lower;
        auto center = (upper + lower)/2.0;
        auto val = mSdf(center[0],center[1],center[2]);
        auto dig = (std::sqrt((size[0] * size[0]) + (size[1]*size[1]) + (size[2]*size[2])))/2.0;
        out = {val - dig, val+dig};
    }
    
    void evalFeatures(boost::container::small_vector<Kernel::Feature, 4>& out) override
    {
        const double delta = .001f;
        auto val = mSdf(points(0, 0), points(1, 0),points(2, 0));
        
        Eigen::Vector3f dx{points(0, 0) - delta, points(1, 0),points(2, 0)};
        Eigen::Vector3f dy{points(0, 0), points(1, 0) - delta, points(2, 0)};
        Eigen::Vector3f dz{points(0, 0), points(1, 0), points(2, 0) - delta};

        
        Eigen::Vector3f n {val - mSdf(dx[0],dx[1],dx[2]),
                           val - mSdf(dy[0],dy[1],dy[2]),
                           val - mSdf(dz[0],dz[1],dz[2])};
        out.push_back(n);
    }
    void checkAmbiguous(Eigen::Block<Eigen::Array<bool, 1, LIBFIVE_EVAL_ARRAY_SIZE>,
                        1, Eigen::Dynamic> out) override{}
private:
    std::function<double(double,double,double)> mSdf;
};

class SDFOracleClause : public Kernel::OracleClause
{
public:
    SDFOracleClause(std::function<double(double,double,double)> sdf):mSdf(sdf){};
    std::unique_ptr<Kernel::Oracle> getOracle() const override
    {
        return std::make_unique<SDFOracle>(mSdf);
    };
    
    std::string name() const override
    {
        return "SDFOracle";
    };
    
private:
   std::function<double(double,double,double)> mSdf;
};

static void writeMesh(std::string path,
                      std::function<double(double,double,double)> sdf)
{
    auto clasue = std::make_unique<SDFOracleClause>(sdf);
    auto tree = Kernel::Tree(std::move(clasue));
    
    Eigen::Array<double, 3, 1> lower{-10,-10,-10};
    Eigen::Array<double, 3, 1> upper{10,10,10};
    
   // auto box = Kernel::Region<3>{lower,upper};
    
    auto box = Kernel::findBounds(tree);

    auto mesh = Kernel::Mesh::render(tree,box,.5,1e-8,false);
    
    mesh->saveSTL(path);
}

int main(int argc, const char * argv[]){
    
    auto sphereFunc = [](double x,double y,double z)
    {
        return std::sqrt(x*x + y*y + z*z) - 5;
    };

    writeMesh("sdfMesh.stl", sphereFunc);
    
    return 0;
}

```

mkeeter, 2018-05-12:
Because `findBounds` is so unpredictable, I've removed it from the public API for now.

You can still use it by defining `ENABLE_FIND_BOUNDS_EXPERIMENTAL` before including `bounds.hpp`, but hopefully this gives the right impression about how much you should trust it 😃

closed 2018-05-12 by mkeeter