libfive/libfive #172: Dragging studio viewport under kde causes whole window to be dragged

traverseda, 2018-05-15:
### Description

Under kde's plasma window manager, trying to click and drag the viewport causes the whole window to be dragged, as if you were clicking on the tiitle bar. This makes it impossible to look at your model from different angles.

[Steps to reproduce]
[What did you expect to happen?  What happened?  All the time?]
[Are there any error messages printed to the console?]

### Versions

- Operating system: Archlinux
- `libfive` commit hash: c2b3b59
  - Is this the latest commit to `master`?  yes
- Qt version: 5.10.1-8, plasma-desktop 5.12.5

mkeeter, 2018-05-16:
Thanks for the report!

A few minutes of Googling hints at this being a window manager setting, e.g. [this thread](https://forum.kde.org/viewtopic.php?f=288&t=127294).

Can you change that setting and see if the problem goes away?

traverseda, 2018-05-16:
I can confirm that fixes it, although the menu path is a bit different. It's "Application style > Widget style [configure] > windows drag mode".

I will note that this doesn't seem to be a problem in openscad. It seems like qt/kde thinks the 3D viewport is "an empty area". It would be better if kde didn't think it was an empty area, and thus a suitable target for those drag action.

mkeeter, 2018-05-16:
If you'd like to fix this, I'd be happy to accept a PR – perhaps adding `event->accept()` to all of the mouse handlers in `view.cpp` would be sufficient?

For more details, there's a long discussion about this on the [KDE bug tracker and mailing list](https://bugs.kde.org/show_bug.cgi?id=245807), which was marked as fixed when they [disabled](https://cgit.kde.org/kde-workspace.git/diff/kstyles/oxygen/oxygenwindowmanager.cpp?id=38295c4e12a794fa248280d5f8d554db1b381a84) this "feature" for `QGraphicsView` completely.

mkeeter, 2019-04-02:
This is theoretically closed by #258, though I don't have a Linux system to test it on – please re-open if it reoccurs.

PR #258 (2019-04-02): Accept mouse(wheel) events
This fixes #172.
commit 67bbbcdd (2019-04-02): Accept mouse(wheel) events, fixes #172
closed 2019-04-02 by mkeeter