libfive/libfive #174: Requires exactly version 7.6.4 of bdw-gc

omgitsraven, 2018-05-20:
When I try building libfive, I get an error that says `ld: warning: directory not found for option '-L/usr/local/Cellar/bdw-gc/7.6.4/lib'`. When I visit `/usr/local/Cellar/bdw-gc/`, it contains a "7.6.0" folder and a "7.6.6" folder, but nothing for "7.6.4".

I'm not sure exactly when this broke -- I recently ran `brew cleanup`, which might have caused it, but I think I also updated Xcode recently, which could also be related...

mkeeter, 2018-05-21:
Hmmm, that's not a library that I use anywhere in libfive itself – maybe it's getting linked by dependencies?  I would look skeptically in Guile's general direction, and as always, try deleting + recreating the build folder from scratch.

omgitsraven, 2018-05-21:
I've deleted & recreated already; I'm afraid I don't even know enough about Guile (or compiling in general) to even know what to start looking into. I've already brought all the dependencies listed on the main page up to date as far as brew is concerned.

mkeeter, 2018-05-21:
What happens if you just try to build the core library (`ninja libfive/src/libfive.dylib`) vs. the library that's linked against Guile (`ninja libfive/bind/libfive-guile.dylib`)?

mkeeter, 2018-05-21:
Here's what I see in Homebrew:
```➜  brew info guile
guile: stable 2.2.3 (bottled), HEAD
GNU Ubiquitous Intelligent Language for Extensions
https://www.gnu.org/software/guile/
/usr/local/Cellar/guile/2.2.2 (784 files, 51.5MB)
  Built from source on 2017-05-24 at 16:12:12
/usr/local/Cellar/guile/2.2.3_1 (785 files, 48.7MB) *
  Poured from bottle on 2018-02-09 at 13:21:46
From: https://github.com/Homebrew/homebrew-core/blob/master/Formula/guile.rb
==> Dependencies
Required: pkg-config ✔, libtool ✔, libffi ✔, libunistring ✔, bdw-gc ✘, gmp ✔, readline ✔
==> Options
--HEAD
        Install HEAD version
➜   brew info bdw-gc
bdw-gc: stable 7.6.6 (bottled), HEAD
Garbage collector for C and C++
http://www.hboehm.info/gc/
/usr/local/Cellar/bdw-gc/7.6.0 (69 files, 1.4MB)
  Poured from bottle on 2017-05-24 at 15:54:00
/usr/local/Cellar/bdw-gc/7.6.4 (70 files, 1.4MB) *
  Poured from bottle on 2018-02-09 at 13:18:04
From: https://github.com/Homebrew/homebrew-core/blob/master/Formula/bdw-gc.rb
==> Dependencies
Build: pkg-config ✔, libatomic_ops ✘
==> Options
--HEAD
        Install HEAD version
```

My `libfive-guile.dylib` is linked against `bdw-gc` in `/usr/local/opt`:
```➜  otool -L libfive/bind/libfive-guile.dylib
libfive/bind/libfive-guile.dylib:
        @rpath/libfive-guile.dylib (compatibility version 0.0.0, current version 0.0.0)
        @rpath/libfive.dylib (compatibility version 0.0.0, current version 0.0.0)
        /usr/local/opt/guile/lib/libguile-2.2.1.dylib (compatibility version 5.0.0, current version 5.0.0)
        /usr/local/opt/bdw-gc/lib/libgc.1.dylib (compatibility version 5.0.0, current version 5.2.0)
        /usr/local/opt/libpng/lib/libpng16.16.dylib (compatibility version 51.0.0, current version 51.0.0)
        /usr/lib/libz.1.dylib (compatibility version 1.0.0, current version 1.2.11)
        /usr/lib/libc++.1.dylib (compatibility version 1.0.0, current version 400.9.0)
        /usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1252.50.4)
```
The build system is using `pkg-config` to find Guile and figure out what libraries it links against; is it possible that your system has an incorrect version of `guile.pc`?

Here's what mine finds:
```➜   pkg-config --debug guile-2.2
no output option set, defaulting to --exists
Error printing disabled by default due to use of output options --exists, --atleast/exact/max-version, --list-all or no output option at all. Value of --print-errors: 0
Error printing disabled
Adding virtual 'pkg-config' package to list of known packages
Looking for package 'guile-2.2'
Looking for package 'guile-2.2-uninstalled'
Reading 'guile-2.2' from file '/usr/local/lib/pkgconfig/guile-2.2.pc'
Parsing package file '/usr/local/lib/pkgconfig/guile-2.2.pc'
  line>prefix=/usr/local/Cellar/guile/2.2.3_1
 Variable declaration, 'prefix' has value '/usr/local/Cellar/guile/2.2.3_1'
  line>exec_prefix=/usr/local/Cellar/guile/2.2.3_1
 Variable declaration, 'exec_prefix' has value '/usr/local/Cellar/guile/2.2.3_1'
  line>bindir=/usr/local/Cellar/guile/2.2.3_1/bin
 Variable declaration, 'bindir' has value '/usr/local/Cellar/guile/2.2.3_1/bin'
  line>libdir=/usr/local/Cellar/guile/2.2.3_1/lib
 Variable declaration, 'libdir' has value '/usr/local/Cellar/guile/2.2.3_1/lib'
  line>includedir=/usr/local/Cellar/guile/2.2.3_1/include
 Variable declaration, 'includedir' has value '/usr/local/Cellar/guile/2.2.3_1/include'
  line>datarootdir=/usr/local/Cellar/guile/2.2.3_1/share
 Variable declaration, 'datarootdir' has value '/usr/local/Cellar/guile/2.2.3_1/share'
  line>datadir=/usr/local/Cellar/guile/2.2.3_1/share
 Variable declaration, 'datadir' has value '/usr/local/Cellar/guile/2.2.3_1/share'
  line>pkgdatadir=/usr/local/Cellar/guile/2.2.3_1/share/guile
 Variable declaration, 'pkgdatadir' has value '/usr/local/Cellar/guile/2.2.3_1/share/guile'
  line>pkgincludedir=/usr/local/Cellar/guile/2.2.3_1/include/guile
 Variable declaration, 'pkgincludedir' has value '/usr/local/Cellar/guile/2.2.3_1/include/guile'
  line>
  line>sitedir=/usr/local/Cellar/guile/2.2.3_1/share/guile/site/2.2
 Variable declaration, 'sitedir' has value '/usr/local/Cellar/guile/2.2.3_1/share/guile/site/2.2'
  line>extensiondir=/usr/local/Cellar/guile/2.2.3_1/lib/guile/2.2/extensions
 Variable declaration, 'extensiondir' has value '/usr/local/Cellar/guile/2.2.3_1/lib/guile/2.2/extensions'
  line>siteccachedir=/usr/local/Cellar/guile/2.2.3_1/lib/guile/2.2/site-ccache
 Variable declaration, 'siteccachedir' has value '/usr/local/Cellar/guile/2.2.3_1/lib/guile/2.2/site-ccache'
  line>libguileinterface=4:0:3
 Variable declaration, 'libguileinterface' has value '4:0:3'
  line>
  line>
  line>
  line>
  line>guild=${bindir}/guild
 Variable declaration, 'guild' has value '/usr/local/Cellar/guile/2.2.3_1/bin/guild'
  line>guile=${bindir}/guile
 Variable declaration, 'guile' has value '/usr/local/Cellar/guile/2.2.3_1/bin/guile'
  line>
  line>Name: GNU Guile
  line>Description: GNU's Ubiquitous Intelligent Language for Extension
  line>Version: 2.2.3
  line>Libs: -L${libdir} -lguile-2.2 -L/usr/local/Cellar/bdw-gc/7.6.4/lib -lgc
  line>Libs.private:  -L/usr/local/opt/gmp/lib -lgmp -lltdl -L/usr/local/Cellar/libffi/3.2.1/lib -lffi    -lunistring  -liconv
Unknown keyword 'Libs.private' in '/usr/local/lib/pkgconfig/guile-2.2.pc'
  line>Cflags: -I${pkgincludedir}/2.2 -I/usr/local/opt/gmp/include -I/usr/local/opt/readline/include -D_THREAD_SAFE  -I/usr/local/Cellar/bdw-gc/7.6.4/include
Path position of 'guile-2.2' is 1
Adding 'guile-2.2' to list of known packages
```

omgitsraven, 2018-05-21:
Doing `ninja libfive/src/libfive.dylib` just gives me `ninja: error: loading 'build.ninja': No such file or directory`

It's interesting that your `brew info guile` gives an ✘ for its bdw-gc dependency; mine gives it a ✔!

Also I'm not sure what it means that we've both got bdw-gcc 7.6.6 apparently, but your Cellar only has 7.6.0 and 7.6.4 where mine has 7.6.0 and 7.6.6?

```
➜    brew info guile
guile: stable 2.2.3 (bottled), HEAD
GNU Ubiquitous Intelligent Language for Extensions
https://www.gnu.org/software/guile/
/usr/local/Cellar/guile/2.2.2 (783 files, 51.2MB)
  Poured from bottle on 2017-10-01 at 18:41:16
/usr/local/Cellar/guile/2.2.3_1 (784 files, 48.7MB) *
  Poured from bottle on 2018-05-20 at 18:01:14
From: https://github.com/Homebrew/homebrew-core/blob/master/Formula/guile.rb
==> Dependencies
Required: pkg-config ✔, libtool ✔, libffi ✔, libunistring ✔, bdw-gc ✔, gmp ✔, readline ✔
==> Options
--HEAD
	Install HEAD version

➜    brew info bdw-gc
bdw-gc: stable 7.6.6 (bottled), HEAD
Garbage collector for C and C++
http://www.hboehm.info/gc/
/usr/local/Cellar/bdw-gc/7.6.0 (69 files, 1.4MB)
  Poured from bottle on 2017-10-01 at 18:41:06
/usr/local/Cellar/bdw-gc/7.6.6 (70 files, 1.4MB) *
  Poured from bottle on 2018-05-20 at 18:27:04
From: https://github.com/Homebrew/homebrew-core/blob/master/Formula/bdw-gc.rb
==> Dependencies
Build: pkg-config ✔, libatomic_ops ✘
==> Options
--HEAD
	Install HEAD version
```

I don't seem to have a `libfive-guile.dylib`:

```
➜    cd Documents/GitHub/libfive/libfive/bind/

➜    otool -L libfive-guile.dylib
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/objdump: 'libfive-guile.dylib': No such file or directory

➜    ls
CMakeLists.txt		libfive-guile.h		transforms.scm
bundle.cmake		sandbox.scm		util.scm
csg.scm			shapes.scm		vec.scm
libfive-guile.cpp	text.scm
```

My output of `pkg-config --debug guile-2.2` is identical to yours, other than one bit of whitespace which I'm chalking up to Github's formatting.

omgitsraven, 2018-05-21:
Also given that you expected my installation to have .dylib files, I might as well show exactly where my compilation is going wrong:

```
➜    ./build_ao.txt
-- The C compiler identification is AppleClang 9.1.0.9020039
-- The CXX compiler identification is AppleClang 9.1.0.9020039
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Boost version: 1.67.0
-- Found PkgConfig: /usr/local/bin/pkg-config (found version "0.29.2") 
-- Checking for module 'guile-2.2>=2.2.1'
--   Found guile-2.2, version 2.2.3
-- Checking for module 'eigen3>=3.2.92'
--   Found eigen3, version 3.3.4
-- Found ZLIB: /usr/lib/libz.dylib (found version "1.2.11") 
-- Found PNG: /usr/local/lib/libpng.dylib (found version "1.6.34") 
Checking dependencies:
  libfive: ✓
  libfive-guile:     ✓
  Studio:       ✓
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/andrew/Documents/GitHub/libfive/build
[  1%] Generating version.c, _version.c
Scanning dependencies of target five
[  2%] Building CXX object libfive/src/CMakeFiles/five.dir/eval/base.cpp.o
[  3%] Building CXX object libfive/src/CMakeFiles/five.dir/eval/eval_interval.cpp.o
[  4%] Building CXX object libfive/src/CMakeFiles/five.dir/eval/eval_jacobian.cpp.o
[  5%] Building CXX object libfive/src/CMakeFiles/five.dir/eval/eval_array.cpp.o
[  6%] Building CXX object libfive/src/CMakeFiles/five.dir/eval/eval_deriv_array.cpp.o
[  7%] Building CXX object libfive/src/CMakeFiles/five.dir/eval/eval_deriv.cpp.o
[  8%] Building CXX object libfive/src/CMakeFiles/five.dir/eval/eval_feature.cpp.o
[ 10%] Building CXX object libfive/src/CMakeFiles/five.dir/eval/eval_point.cpp.o
[ 11%] Building CXX object libfive/src/CMakeFiles/five.dir/eval/tape.cpp.o
[ 12%] Building CXX object libfive/src/CMakeFiles/five.dir/eval/feature.cpp.o
[ 13%] Building CXX object libfive/src/CMakeFiles/five.dir/eval/transformed_oracle.cpp.o
[ 14%] Building CXX object libfive/src/CMakeFiles/five.dir/render/discrete/heightmap.cpp.o
[ 15%] Building CXX object libfive/src/CMakeFiles/five.dir/render/discrete/voxels.cpp.o
[ 16%] Building CXX object libfive/src/CMakeFiles/five.dir/render/brep/xtree2.cpp.o
[ 17%] Building CXX object libfive/src/CMakeFiles/five.dir/render/brep/xtree3.cpp.o
[ 19%] Building CXX object libfive/src/CMakeFiles/five.dir/render/brep/contours.cpp.o
[ 20%] Building CXX object libfive/src/CMakeFiles/five.dir/render/brep/mesh.cpp.o
[ 21%] Building CXX object libfive/src/CMakeFiles/five.dir/render/brep/marching.cpp.o
[ 22%] Building CXX object libfive/src/CMakeFiles/five.dir/render/brep/neighbors.cpp.o
[ 23%] Building CXX object libfive/src/CMakeFiles/five.dir/solve/bounds.cpp.o
[ 24%] Building CXX object libfive/src/CMakeFiles/five.dir/solve/solver.cpp.o
[ 25%] Building CXX object libfive/src/CMakeFiles/five.dir/tree/cache.cpp.o
[ 26%] Building CXX object libfive/src/CMakeFiles/five.dir/tree/opcode.cpp.o
[ 28%] Building CXX object libfive/src/CMakeFiles/five.dir/tree/archive.cpp.o
[ 29%] Building CXX object libfive/src/CMakeFiles/five.dir/tree/tree.cpp.o
[ 30%] Building CXX object libfive/src/CMakeFiles/five.dir/tree/oracle_clause.cpp.o
[ 31%] Building CXX object libfive/src/CMakeFiles/five.dir/tree/transformed_oracle_clause.cpp.o
[ 32%] Building CXX object libfive/src/CMakeFiles/five.dir/libfive.cpp.o
[ 32%] Generating version.c, _version.c
[ 33%] Building C object libfive/src/CMakeFiles/five.dir/version.c.o
[ 34%] Linking CXX shared library libfive.dylib
ld: warning: directory not found for option '-L/usr/local/Cellar/bdw-gc/7.6.4/lib'
[ 34%] Built target five
[ 35%] Generating bundle.cpp
Scanning dependencies of target five-guile
[ 37%] Building CXX object libfive/bind/CMakeFiles/five-guile.dir/bundle.cpp.o
[ 38%] Linking CXX shared library libfive-guile.dylib
ld: warning: directory not found for option '-L/usr/local/Cellar/bdw-gc/7.6.4/lib'
ld: library not found for -lgc
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [libfive/bind/libfive-guile.dylib] Error 1
make[1]: *** [libfive/bind/CMakeFiles/five-guile.dir/all] Error 2
make: *** [all] Error 2
```

the contents of build_ao.txt is just:
```
cd Documents/GitHub/libfive
mkdir build
cd build
cmake -DCMAKE_PREFIX_PATH=/usr/local/Cellar/qt/5.10.1  ..
make
```

mkeeter, 2018-05-21:
Okay, it looks like the problem is that `guile.pc` is pointing at a version of `bdw-gc` that don't exist on your system.  You could try uninstalling + reinstalling Guile, or manually edit `/usr/local/lib/pkgconfig/guile-2.2.pc`, replacing `7.6.4` with `7.6.6` in the `bdw-gc` paths.

omgitsraven, 2018-05-23:
Phew -- un/reinstalling did nothing, but manually editing guile-2.2.pc fixed it. Thanks so much :) I guess I'll take this issue to homebrew now.

closed 2018-05-23 by omgitsraven