libfive/libfive #176: Cannot build libfive-guile/studio under Debian stretch, even after recompiling guile

galex-713, 2018-05-21:
### Description

I can’t build libfive-guile and studio, as guile 2.2 is not detected by cmake, even though I recompiled guile manually myself and checked guile-2.2 was available through pkg-config

```
-- Boost version: 1.62.0
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29") 
-- Checking for module 'guile-2.2>=2.2.1'
--   
-- Checking for module 'eigen3>=3.2.92'
--   Found eigen3, version 3.3.2
-- Found ZLIB: /usr/lib/x86_64-linux-gnu/libz.so (found version "1.2.8") 
-- Found PNG: /usr/lib/x86_64-linux-gnu/libpng.so (found version "1.6.28") 
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
Checking dependencies:
  libfive: ✓
  libfive-guile:     ✘   (needs Guile 2.2 or later)
  Studio:       ✘   (needs Guile 2.2 or later)
-- Configuring done
-- Generating done
-- Build files have been written to: /home/galex-713/src/libfive
```

### Versions

- Operating system: Debian GNU/Linux stretch amd64
- `libfive` commit hash: aa2018ce8de97beaa579ca6c83a001c5c3aec9db
  - Is this the latest commit to `master`?  [yes]
- Qt version (if applicable): 5.7.1+dfsg-3+b1

### Common questions
[Delete this section if not relevant]

If this is a behavior issue, does the test suite (`libfive/test/libfive-test`) pass?

Yes it does.

mkeeter, 2018-05-22:
That's odd – what does `pkg-config --debug guile-2.2` print?

galex-713, 2018-05-22:
I finally solved the problem, sorry for inconvenience: the first time before yesterday I thought I just needed a more recent version of guile so I recompiled guile trunk (3.0, from git, so it took one entire night) and libfive cmake didn’t recognize it, then I focused on “guile-2.2” and I followed the link on guile website and recompiled 2.2.0 from the tarball (faster) but cmake still wasn’t recognizing it, then I finally realized it was written “2.2.1” in the cmake output, so I searched the link I used for 2.2.0, I went up on the ftp and observed the list of tarball, then downloaded 2.2.3, recompiled, uninstalled 2.2.0 and installed 2.2.3 and now it works! For the first time I got libao/ao/libfive working here \o/ :D

Here the output of `pkg-config --debug guile-2.2` before 2.2.0 uninstall and 2.2.3 install, then after:
[2.2.0.cmd.txt](https://github.com/libfive/libfive/files/2025666/2.2.0.cmd.txt)
[2.2.3.cmd.txt](https://github.com/libfive/libfive/files/2025669/2.2.3.cmd.txt)

closed 2018-05-22 by galex-713