libfive/libfive #177: Unexpected result from (loft S1 S2)

klpeders, 2018-05-26:
### Description

Simple loft expression return unexpected result.

The following snippet reproduce the issue

  (loft 
    (ring 11 10)
    (ring  9  8)
    0 10
  )

I think this used to work, but on latest git it doesn't work. 

No errors messages in the console

### Versions

- Operating system: Linux bleeding edge (Fedora 28)
- `libfive` commit hash: 7a46e316ecc755bb3597f4f3ca84e7ea9a8e75b8
  - Is this the latest commit to `master`?  [yes]
- Qt version (if applicable): [here]

### Additional Information
[image]

mkeeter, 2018-05-26:
Hi @klpeders !

This isn't working as you'd expect because of how `loft` works under the hood.

The `loft` operation does a weighted blend between the two distance fields, with the weight controlled by `z`.  You're lofting between two shapes that have no shared area (i.e. the rings are concentric and non-overlapping).  As one ring goes to 0 and the other goes to 1, there's a region with neither of the rings being solid.

You can get an intuition with the following code:
```
#! RENDER -15 -15 -15 / 15 15 15 / 10 / 8 !#
(define m 0)
(extrude-z
  (morph
    (ring 11 10)
    (ring 9 8) m)
0 1)
```
As you sweep `m` from 0 to 1, notice that one ring disappears, then there's a gap with no rings visible, then the second ring appears.

klpeders, 2018-06-03:
I am trying to understand the implications of what you say, but this behaviour is extremely non-intuitive, IMHO.

I would naively expect that 

```scheme
(define p_ri 0.95)
(difference
  (loft (circle 1) (circle 1)     0 1)
  (loft (circle 0) (circle p_ri)  0 1)
)

(loft
  (difference (circle 1) (circle 0))
  (difference (circle 1) (circle p_ri))
  0 1
)
```
would generate the same output. 
But, this is what happen:

[image]
Beautiful effect, but rather surprising, no? Especially the discontinuity ...

Even more interesting, if `p_ri < inner_radius/2` things are all as expected:

[image]

doug-moen, 2018-06-04:
@klpeders Re: your second example:
```
(loft
  (difference (circle 1) (circle 0))
  (difference (circle 1) (circle p_ri))
  0 1
)
```
I'm the author of Curv, and I'm going to use Curv to explain this behaviour, since Curv has tools for visualizing signed distance fields. The loft operation behaves the same way in both libfive and curv.

Curv represents shapes as signed distance fields (SDFs). A shape is represented by a function that maps each point (x,y,z) in 3D space onto an estimate of the shortest distance from (x,y,z) to the nearest surface of the shape. It's a signed distance, because the distance value is positive outside the shape, zero on the boundary, and negative inside the shape. It's an estimate because the distance estimate is permitted to be less than the actual distance.

Because distance estimates are used, instead of exact distances, this means that there are an infinite number of signed distance fields that can represent any given shape. As a special case, an *exact* distance field is one where all of the distance estimates are exact values, not underestimates. Normally, you don't know or care about the structure of this distance field, you only care about the boundary surface, where the SDF is 0.

However, there are certain operations in Curv, such as `loft`, `morph` and `offset`, that depend on the structure of the distance field. To understand what `loft` is doing, we need to examine the structure of the SDFs that are passed as arguments.

Let's first consider a simplified version of your example:
```
(loft
  (circle 1)
  (difference (circle 1) (circle p_ri))
  0 1
)
```
In Curv, this is
```
loft 1 (circle 1, difference(circle 1, circle p_ri))
```
The result is:
[image]

The cross section is:
[image]

Loft performs a linear interpolation between the SDFs of two 2-dimensional shapes. To understand why we get this particular cross section, we need to look at the distance fields for each shape argument.

Here's a visualization of the SDF for `circle 1`. I'm displaying it as surface, where the Z height is the SDF value at that (x,y) point, with red for distance values <=0, and yellow for >0.
[image]

Here is a visualization of the SDF for `difference(circle 1, circle p_ri)`.
[image]

To create each Z-slice of the loft output, we linearly interpolate between the first SDF and the second SDF. If you consider the point at (0,0), it has distance -0.5 in the `circle 1` SDF: that is the red point at the apex of the cone. In the second SDF, (0,0) has distance +0.475: that's the apex of the smaller cone rising above the red ring. By linear interpolation, the distance at (0,0) goes from negative to positive roughly half way up the loft, which corresponds to the bottom of the cup shape that appears to be carved out of the cylinder in the loft output. (This would be a lot clearer if I could post an animation of the circle SDF morphing into the ring SDF: then the reason for the profile of the loft output would be more obvious.)

Now let's consider the original version of your example. Instead of beginning with `circle 1`, your loft begins with `difference (circle 1, circle 0)` (in Curv syntax). This subtracts a zero-diameter circle, which has no area, from a circle with diameter 1. Geometrically, this should make no difference, and the resulting circle looks the same as `circle 1`, but the distance field is different. It looks like this:
[image]
[image]

It's the same distance field as `circle`, except for the small inverted cone at the centre of the distance field. That inverted cone accounts for the difference in the loft output. In particular, note that the distance value at (0,0) is 0. What has happened is that the point (0,0) has been included in the boundary of the shape, as a result of the difference operation, and this SDF is the logical consequence of this.

I created the `loft` operation as an experiment. It was named after the loft operation in Autocad. It works well if you loft between two roughly symmetrical shapes that cover the origin and are centred on the origin. When you deviate from that, the behaviour becomes unintuitive. Producing "intuitive" behaviour in the general case, unfortunately seems to be a research project.