libfive/libfive #178: Problems with Pyramid in libfive studio

Paul-Wray, 2018-06-16:
### Description
Pyramid with vertex at origin causes immediate crash
eg (pyramid #[-1 -1] #[1 1] -1 1) crashes the app

### Versions
Precompiled OSX version at https://libfive.com/download/#studio

mkeeter, 2018-06-16:
Thanks, I'll check it out!

mkeeter, 2018-06-17:
Oh, that build is from January!

There have been a bunch of changes since then, so I just updated the build from the `thread-pool` branch (which is a bit less stable than `master`, but has all of my latest updates).

This pyramid now works fine; feel free to re-open if you run into other issues.

closed 2018-06-17 by mkeeter

Paul-Wray, 2018-06-18:
Thanks very much for updating the Studio build. I stumbled across this after trying openSCAD. Already I modelled a simple tool and printed it successfully, and I think it may be easier to use in some ways, and more powerful.

Do you have a roadmap anywhere? Is the intention to continue to develop studio to be more full-featured? 

Your project deserves to be known more widely, I could perhaps contribute eg a simple screencast.

mkeeter, 2018-06-19:
I'm certainly planning to continue developing the kernel and UI.

Right now, I have a commercial partner that's sponsoring kernel development, so that's where I've been focusing my efforts; the Studio UI is meant as a basic demo, and I'm hoping that as a community forms, folks will begin building more powerful GUIs on top of the kernel / infrastructure layers.