libfive/libfive #182: Blend causes segfault

Paul-Wray, 2018-06-22:
(blend 
  (sphere 1)
  (box [0 0 0] [ 1 1 1])
  0.1
)

Seems to work OK for fractions above 0.2, but crashes with 0.1

mkeeter, 2018-06-22:
Thanks!  This has the same traceback as the cone issue (#179), so hopefully we can fix them both simultaneously.

Paul-Wray, 2018-06-25:
This may be another example. Open the program below in Studio, then reduce the resolution from 10 to 9. Gives me a segfault.

`#! RENDER -32 -32 -32 / 32 32 37 / 10 / 8 !#/

(define Ri 22)
(define t 4)
(define r (/ t 2))
(define h 15)
(define (smooth-tube R r h)
  (union
    (torus (+ R r) r [0 0 0]) 
    (torus (+ R r) r [0 0 h]) 
    (extrude
      (ring (+ R r r) R)
      0
      h
    )
  )
)

(blend
  (move 
    (smooth-tube Ri r h)
    [0 0 15]
  )
  (smooth-tube (+ Ri t) r h)
  1.5
)`

mkeeter, 2018-06-26:
The `blend` bug is fixed by [this commit](https://github.com/libfive/libfive/commit/ca5f9aea81b4a4d9033e29f288517554f6859a20); I'm still looking at the `smooth-tube` issue.

mkeeter, 2018-06-28:
Fixed in the latest `thread-pool` commit (1eadc7582e94c67b37abe882af23a54426a71f80), with a few changes:
- Interval arithmetic now detects cases where intermediate values are `inf` or `NaN`, and flags them for further subdivision even if the top-level result is nominally inside or outside.
- I'm now always doing point/derivative evaluation in SIMD-sized bundles, because Eigen's array functions behaves differently in corner cases (i.e. giving one result when evaluation 1-7 of an operation, then a different result when evaluating a full 8 with SIMD).

I've also uploaded a new Studio build to [the downloads page](libfive.com/download).

closed 2018-06-28 by mkeeter