libfive/libfive #183: No auto-union within functions.

Paul-Wray, 2018-06-25:
Seems you need an explicit union within functions to union a sequence of shapes. 
Is that by design?

doug-moen, 2018-06-25:
It has to work that way, due to the nature of function definitions in the
host language (Guile).

The only reason that OpenSCAD manages to have implicit unions inside of
module definitions, is that OpenSCAD has a distinction between module
definitions and function definitions. In OpenSCAD, a function has an
expression as its body: the body is evaluated, yielding a value that is the
result of the function. This is just like a Guile function. The body of an
OpenSCAD module has a special syntax which allows for a sequence of
"statements", where an OpenSCAD statement can construct a shape which is
added to an implicit union.

Guile is a general purpose language, it doesn't have those weird OpenSCAD
semantics, but you could define a macro called "module", such that
   (module (M a b)
      (cube a)
      (sphere b))
is equivalent to
   (define (M a b)
     (union
        (cube a)
        (sphere b)))

OpenSCAD also has 2 kinds of conditionals: there is an "if statement" which
supports implicit unions, and there is a conditional expression a?b:c which
is equivalent to "if" in Guile. So, you could define another macro that is
equivalent to an OpenSCAD "if statement", and then you would have implicit
unions inside of conditionals.

I think that the cost OpenSCAD pays to have implicit unions is too high.
There are two kinds of function definitions, two kinds of conditionals,
there are essentially two different languages joined together: the
expression language for manipulating values, and the statement language for
manipulating shapes.

Paul-Wray, 2018-06-25:
Thanks for the explanation! I did notice openscad had the custom syntax you describe.

>I think that the cost OpenSCAD pays to have implicit unions is too high.

Yes I tend to agree. Also it makes semantic sense to have no magic in functions. I just don't remember reading about it in the help docs.

I'm just getting interested in 3D printing, and dabbled first with freecad and found the learning curve for the UI (not the concepts, just the UI) to be very high. Then dabbled with openscad and liked it a lot. Defining constraints procedurally just seems like a lot less futzing and clearer. Then stumbled across libfive and became interested in the elegant theory behind f-rep, sphere tracing etc. From my beginners perspective, with a few more features and a bit more stability, seems libfive studio could compete with openscad, although I wonder whether postfix expressions (eg Forth) might be better fit for the application. Without a dedicated lisp editor, brackets will be a problem.

doug-moen, 2018-06-25:
If you want fewer parentheses, or would like to try sphere tracing, you could look at my project, called Curv, at: https://github.com/doug-moen/curv

The Curv language has fewer parentheses than either OpenSCAD or libfive. It uses prefix notation, same as the functional languages Haskell, F#, Elm, etc. So you would write `cube 5` in Curv, instead of `cube(5)` in OpenSCAD or `(cube 5)` in libfive. No implicit union, though. It uses sphere tracing on the GPU (unlike libfive), and a more powerful version of F-rep that supports iterated fractals like the Mandelbulb, and also full colour models and animation.

Paul-Wray, 2018-06-25:
I did have a glance at it, but saw it described as a platform for doing art. So I assumed that it would not be suitable for CAD work. Ill have a closer look tomorrow.