libfive/libfive #193: Does not currently compile on Windows

lf-, 2018-07-20:
### Description

[Description of the issue]
The project fails to build on Windows

[Steps to reproduce]
Follow build instructions in README.md. I used Visual Studio Community 2017 Preview.

[What did you expect to happen?  What happened?  All the time?]
Project should compile cleanly.

[Are there any error messages printed to the console?]
```
1>------ Build started: Project: ZERO_CHECK, Configuration: Debug Win32 ------
1>Checking Build System
1>CMake does not need to re-run because C:/Users/lf/Dev/libfive/build/CMakeFiles/generate.stamp is up-to-date.
1>CMake does not need to re-run because C:/Users/lf/Dev/libfive/build/libfive/CMakeFiles/generate.stamp is up-to-date.
1>CMake does not need to re-run because C:/Users/lf/Dev/libfive/build/libfive/src/CMakeFiles/generate.stamp is up-to-date.
1>CMake does not need to re-run because C:/Users/lf/Dev/libfive/build/libfive/test/CMakeFiles/generate.stamp is up-to-date.
2>------ Build started: Project: five, Configuration: Debug Win32 ------
2>Generating version.c, _version.c
2>Building Custom Rule C:/Users/lf/Dev/libfive/libfive/src/CMakeLists.txt
2>CMake does not need to re-run because C:/Users/lf/Dev/libfive/build/libfive/src/CMakeFiles/generate.stamp is up-to-date.
2>base.cpp
2>deck.cpp
2>eval_interval.cpp
2>eval_jacobian.cpp
2>eval_array.cpp
2>eval_deriv_array.cpp
2>eval_deriv.cpp
2>eval_feature.cpp
2>eval_point.cpp
2>tape.cpp
2>feature.cpp
2>transformed_oracle.cpp
2>heightmap.cpp
2>voxels.cpp
2>xtree2.cpp
2>xtree3.cpp
2>xtree_pool2.cpp
2>xtree_pool3.cpp
2>contours.cpp
2>mesh.cpp
2>Generating Code...
2>Compiling...
2>marching.cpp
2>neighbors.cpp
2>bounds.cpp
2>solver.cpp
2>cache.cpp
2>opcode.cpp
2>archive.cpp
2>deserializer.cpp
2>serializer.cpp
2>c:\users\lf\dev\libfive\libfive\src\tree\serializer.cpp(49): error C3861: 'assert': identifier not found
2>c:\users\lf\dev\libfive\libfive\src\tree\serializer.cpp(65): error C3861: 'assert': identifier not found
2>tree.cpp
2>oracle_clause.cpp
2>transformed_oracle_clause.cpp
2>c:\users\lf\dev\libfive\libfive\src\tree\transformed_oracle_clause.cpp(64): error C2065: 'ids': undeclared identifier
2>c:\users\lf\dev\libfive\libfive\src\tree\transformed_oracle_clause.cpp(64): error C2228: left of '.end' must have class/struct/union
2>c:\users\lf\dev\libfive\libfive\src\tree\transformed_oracle_clause.cpp(64): note: type is 'unknown-type'
2>c:\users\lf\dev\libfive\libfive\src\tree\transformed_oracle_clause.cpp(81): error C2065: 'ts': undeclared identifier
2>c:\users\lf\dev\libfive\libfive\src\tree\transformed_oracle_clause.cpp(81): error C2228: left of '.end' must have class/struct/union
2>c:\users\lf\dev\libfive\libfive\src\tree\transformed_oracle_clause.cpp(81): note: type is 'unknown-type'
2>libfive.cpp
2>Generating Code...
2>Done building project "five.vcxproj" -- FAILED.
========== Build: 1 succeeded, 1 failed, 0 up-to-date, 0 skipped ==========
```

The test suite also seems to have build failures:

```
2>------ Build started: Project: libfive-test, Configuration: Debug Win32 ------
2>Building Custom Rule C:/Users/lf/Dev/libfive/libfive/test/CMakeLists.txt
2>CMake does not need to re-run because C:/Users/lf/Dev/libfive/build/libfive/test/CMakeFiles/generate.stamp is up-to-date.
2>main.cpp
2>api.cpp
2>archive.cpp
2>bounds.cpp
2>cache.cpp
2>c:\users\lf\dev\libfive\libfive\test\cache.cpp(56): error C2124: divide or mod by zero
2>contours.cpp
2>deck.cpp
2>dual.cpp
2>eval_interval.cpp
2>eval_jacobian.cpp
2>eval_array.cpp
2>eval_deriv.cpp
2>eval_deriv_array.cpp
2>eval_feature.cpp
2>c:\users\lf\dev\libfive\libfive\test\eval_feature.cpp(88): error C2398: Element '1': conversion from 'double' to 'const float &' requires a narrowing conversion
2>c:\users\lf\dev\libfive\libfive\test\eval_feature.cpp(88): error C2398: Element '2': conversion from 'double' to 'const float &' requires a narrowing conversion
2>eval_point.cpp
2>feature.cpp
2>heightmap.cpp
2>marching.cpp
2>mesh.cpp
2>neighbors.cpp
2>Generating Code...
2>Compiling...
2>oracle.cpp
2>region.cpp
2>solver.cpp
2>transformed_oracle.cpp
2>tree.cpp
2>voxels.cpp
2>xtree.cpp
2>shapes.cpp
2>Generating Code...
2>Done building project "libfive-test.vcxproj" -- FAILED.
========== Build: 0 succeeded, 2 failed, 1 up-to-date, 0 skipped ==========
```

### Versions

- Operating system: Windows 10 1803
- `libfive` commit hash: 17416ad1812d7ff11d043e14352c249329f7dbec
  - Is this the latest commit to `master`?  yes

mkeeter, 2018-07-21:
Okay, I pushed a few changes that should fix these specific errors – can you pull to bf0e1e1648667d2b47d8d4211e017e304d97465d and try rebuilding?

lf-, 2018-07-21:
Fixes the build errors, so I ran the test suite and found some more issues. I didn't get my prompt back several minutes after the last line here so I CTRL-C'd it; if there is supposed to be more output, the test suite is getting stuck.

```
PS C:\Users\lf\Dev\libfive\build\Output\Debug> .\libfive-test.exe
OracleClause::serialize: no installed "ST"
  You may need to call OracleClause::install.

~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
libfive-test.exe is a Catch v2.2.3 host application.
Run with -? for options

-------------------------------------------------------------------------------
Archive::serialize
  With an oracle
-------------------------------------------------------------------------------
c:\users\lf\dev\libfive\libfive\test\archive.cpp(117)
...............................................................................

c:\users\lf\dev\libfive\libfive\test\archive.cpp(127): FAILED:
  REQUIRE( out.str() == expected )
with expansion:
  "T"""" "ST"  " == "T"""" "ST""hi"  "

OracleClause::deserialize: no installed "ST"
  You may need to call OracleClause::install.
Deserializer: failed to deserialize Oracle "ST"
Deserializer: expected tag == 'T' || tag == 't' at deserializer.cpp:60
-------------------------------------------------------------------------------
Archive::deserialize
  Oracle
-------------------------------------------------------------------------------
c:\users\lf\dev\libfive\libfive\test\archive.cpp(133)
...............................................................................

c:\users\lf\dev\libfive\libfive\test\archive.cpp(133): FAILED:
due to unexpected exception with message:
  60

-------------------------------------------------------------------------------
Heightmap::render: Performance
-------------------------------------------------------------------------------
c:\users\lf\dev\libfive\libfive\test\heightmap.cpp(268)
...............................................................................

benchmark name                                  iters   elapsed ns      average
-------------------------------------------------------------------------------
sphere                                              1  11645621600    11.6456 s
Menger sponge                                       1  25421254900    25.4213 s
```

mkeeter, 2018-07-21:
I've removed the exception, which was debug code that I left in by accident, but I'm not sure about the rest of the issues.  I'm doing a bit of initialization-time magic to install serializers (detailed [here](https://github.com/libfive/libfive/blob/master/libfive/include/libfive/tree/oracle_clause.hpp#L132-L143)), but I'm not sure why that wouldn't be working on MSVC.

mkeeter, 2018-07-21:
The performance is also surprisingly bad.  Running the same benchmark on my laptop gives the following:
```benchmark name                                  iters   elapsed ns      average
-------------------------------------------------------------------------------
sphere                                              1     64298537   64.2985 ms
Menger sponge                                       1    917754755   917.755 ms
```
Is it possible that we're not setting the right compiler switches to build in "release" mode?

southpawflo, 2018-07-21:
I tried to compile from source on ubuntu 18 yesterday, and had the exact same problems with serializer.cpp. I pulled this latest commit and compilation was successful, studio runs, but I'm at the extent of my abilities here. I figured I should speak up since (that part at least) wasn't just a windows thing.

mkeeter, 2018-07-21:
The compiler errors in `serializer` were my mistake; I had left-over `assert()` calls that used invalid names in the body of the assertion.  On my machine, the body of the assertion is ignored when compiling on release mode; on other compilers, it gets name/type-checked before being dropped, which is what you were seeing.

@southpawflo are you also seeing issues when you try to run `libfive-test`?

lf-, 2018-07-21:
In your defense I think it was compiling in debug mode (and those times
should not be 10x slower than Linux; my machine has an i5-6600k). It's sort
of a moot point since the whole system isn't entirely too helpful on
Windows anyway as Studio isn't yet set up to build on the platform and
might have other compatibility issues as well.

southpawflo, 2018-07-21:
unfortunately my experience with all of this is quite limited. if you tell me how to run the test (either ubuntu or mac, I have both) I will, or better yet if you can point me in the direction of a resource to learn from I would appreciate it and then be able to get back to you with an answer.

mkeeter, 2018-07-21:
Okay, that's a different failed test than @lf- was seeing; it's annoying, but unrelated.

I'm surprised that the serializer is failing on Windows, because there are definitely folks using the core library (including custom serialization) on that platform.

Studio and the Scheme buildings definitely won't build on Windows, because of Guile.  My efforts to get Guile working are [here](https://github.com/mkeeter/guile-mingw), but I've abandoned that work for now.

lf-, 2018-07-21:
(!)

LilyPond uses Guile and is on Windows. I wonder what dark incantations they did to get it to work there.

mkeeter, 2018-07-22:
I think it's because they're using a much older version of Guile – something in the 1.7-1.8 range, rather than the current stable release series (e.g. after 2.2).  Looking at the Mac app bundle, they're also using Python 2.6.  I guess if it ain't broke...

lf-, 2018-07-23:
I also saw an experimental option to use 2.x but I didn't look into it any further to see platform support.

ronasuti, 2018-08-01:
It's also not compiling on 18.04, updated to latest.
Seems I'm getting an error where isnan(inp) doesn't exist.
`/home/roman/libfive/libfive/src/eval/eval_interval.cpp: In function ‘Kernel::{anonymous}::Pr Kernel::{anonymous}::toPair(float)’:
/home/roman/libfive/libfive/src/eval/eval_interval.cpp:30:10: error: ‘isnan’ was not declared in this scope
   return isnan(inp)
/home/roman/libfive/libfive/src/eval/eval_interval.cpp:280:34: error: ‘isnan’ was not declared in this scope
                (a.lower() < 0 && isnan(std::pow(-1.f, bPt))));
`

ronasuti, 2018-08-01:
I found the bug. If you specify the std namespace for each call of isnan then it doesn't error out.
Basically, change the two instances of naked isnan() into std::isnan()

mkeeter, 2018-08-01:
@ronasuti I brought in your PR, which should fix the Lnux issues (#196).

DeveloperPaul123, 2018-10-14:
Just wanted to add the the [comment](https://github.com/libfive/libfive/issues/193#issuecomment-406773534) above referencing the test issues on windows. In debug, the test fail immediately and `abort` is called causing further tests to not be executed. In release, I'm able to run all the tests but some do fail. See output below:

````
OracleClause::serialize: no installed "ST"
  You may need to call OracleClause::install.

~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
libfive-test.exe is a Catch v2.2.3 host application.
Run with -? for options

-------------------------------------------------------------------------------
Archive::serialize
  With an oracle
-------------------------------------------------------------------------------
d:\repositories\libfive\libfive-src\libfive\test\archive.cpp(108)
...............................................................................

d:\repositories\libfive\libfive-src\libfive\test\archive.cpp(118): FAILED:
  REQUIRE( out.str() == expected )
with expansion:
  "T"""" "ST"  " == "T"""" "ST""hi"  "

OracleClause::deserialize: no installed "ST"
  You may need to call OracleClause::install.
Deserializer: failed to deserialize Oracle "ST"
Deserializer: expected tag == 'T' || tag == 't' at deserializer.cpp:49
Deserializer::deserializeString: expected opening "
Deserializer::deserializeString: expected opening "
Deserializer: expected op < Opcode::LAST_OP at deserializer.cpp:82
-------------------------------------------------------------------------------
Archive::deserialize
  Oracle
-------------------------------------------------------------------------------
d:\repositories\libfive\libfive-src\libfive\test\archive.cpp(124)
...............................................................................

d:\repositories\libfive\libfive-src\libfive\test\archive.cpp(131): FAILED:
  REQUIRE( t.tree.id() != nullptr )
with expansion:
  nullptr != nullptr

-------------------------------------------------------------------------------
Heightmap::render: Performance
-------------------------------------------------------------------------------
d:\repositories\libfive\libfive-src\libfive\test\heightmap.cpp(259)
...............................................................................

benchmark name                                  iters   elapsed ns      average
-------------------------------------------------------------------------------
sphere                                              1    116162844   116.163 ms
Menger sponge                                       1    976401668   976.402 ms

-------------------------------------------------------------------------------
Solver::findRoot
  Sum-of-squares performance
-------------------------------------------------------------------------------
d:\repositories\libfive\libfive-src\libfive\test\solver.cpp(85)
...............................................................................

benchmark name                                  iters   elapsed ns      average
-------------------------------------------------------------------------------
SOS root                                            1       146915   146.915 ┬╡s

d:\repositories\libfive\libfive-src\libfive\test\solver.cpp(116): FAILED:
  REQUIRE( res < 1e-3 )
with expansion:
  0.00206f < 0.001

===============================================================================
test cases:   116 |   113 passed | 3 failed
assertions: 17182 | 17179 passed | 3 failed
````

siberia3, 2021-07-11:
A true story. 

My installing Libfive via Command Line on Windows 10 (following the Compiling steps page) begins with a procedural block calling for the English version of Visual Studio -- which anyway I prefer. So, after checking the required VS packages (see) and setting the English interface through the VS Install manager, the following command could never work:   

`.\vcpkg\vcpkg.exe install --triplet x64-windows eigen3 boost-container boost-bimap boost-interval boost-lockfree boost-functional boost-algorithm boost-math libpng qt5-base python3`

Actually this command kept on giving errors such as:

```
Error: Building package brotli:x64-windows failed with: BUILD_FAILED
Error: Building package zlib:x64-windows failed with: BUILD_FAILED
...
```

As solving this kind of problems is light years away, I idled on the way middle dropping off this software, and looking for solutions -- eg trying to figure out could I download & install the dependencies one by one, but the logistics looked too messy :P

Then I had this idea of installing the packages in the command line the reverse order and one by one, this way: 

```
.\vcpkg\vcpkg.exe install --triplet x64-windows python3
.\vcpkg\vcpkg.exe install --triplet x64-windows qt5-base
.\vcpkg\vcpkg.exe install --triplet x64-windows ... 
```

(you got it) and I made to the build part!

The following build command however gave my first fatal error with exit BUILD FAILED

`"C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\Common7\IDE\CommonExtensions\Microsoft\CMake\CMake\bin\cmake.exe" --build . --config Release --target Studio -- -v:n -m:8`

Now this build is a heavy consuming process, and considering Windows10 background services too, I guessed I had to restart the machine to get rid of memory issues. 

Kept on having compiling errors and build failures, THEREFORE I kept on launching the build command on and on ... 

This is the result :)
 
[image]

mkeeter, 2021-07-11:
Thanks for bumping this ticket – it's intended to work on Windows now, and it got it compiled eventually, so I'll close this.

Any idea what was causing the build failures?  The original commands worked for me – was it simply lack of RAM that caused the issue?

closed 2021-07-11 by mkeeter

siberia3, 2021-07-12:
Hi Matt - package errors are always a nightmare since it means a day spent in digging forums and messing up with the system without solving the problem at all.

Reversing the order of vcpkg packages, I think is not because of a issue of memory --but, since the last package (Python) looks "important" then I had this idea to vcpkg-install them one by one in the reverse order -- and this worked.

During the VSC Cmake Build part, I had these OOM alerts resulting in build failure -- that suggested me to restart the machine and perform the build again and again until all the packages were ok -- I never thought I had to help my pc with carrots, but you never know!

Thanks!