libfive/libfive #194: Studio: STL export from command line

lf-, 2018-07-27:
### Description

Currently it seems to be required (at the very least it isn't in the help message) to use the libfive-studio GUI to export STLs given Scheme source code. A helpful addition would be a command line option to export STLs without using the GUI for scripting or using an alternative editor.

mkeeter, 2018-07-27:
Interesting idea!

I think the right way to do this would be as a Guile script, so it can be completely headless (and not depend on Qt) – I'll look into making this possible.

feklee, 2018-08-24:
Indeed this would be very nice!

Personally, I would like to play with `libfive` writing code in Emacs/Termux on my phone. If it was possible to export geometry directly from Guile, then I could visualize that using my phone’s web browser and WebGL. This would be a nice light weight solution.

mkeeter, 2018-11-11:
This is done now!

There's now a utility in [`bin/export-meshes`](https://github.com/libfive/libfive/tree/master/bin/export-meshes) which reads in a Studio file and saves all of its meshes to an STL.

Error handling in this script is non-existent, so this would be a good first issue if you're interested in contributing 😃

lf-, 2018-11-11:
Next issue regarding alternative editor is some way to get the studio to reload the preview via some IPC :)

closed 2018-11-22 by mkeeter