libfive/libfive #197: Emacs/Reload on file change support?

ronasuti, 2018-08-07:
### Using Emacs as the code editor?

is it possible to add an auto-refreshing option to Studio, such that when Emacs saves the file, it updates automatically in Studio? This sounds like the easiest way to get Emacs support into libfive, although actual integrations would be awesome if possible (not sure where to start with that though)

### Possible ways to do this, easiest to hardest
- auto-reload of files, save from Emacs to propagate changes
- REST api that code can be POST-ed to, which is very easy to write an Emacs plugin for
- "full" integration
  - Emacs plugin that asks for Studio location, turns off text input, and forces studio to use Emacs's buffer as it's file so changes are propagated in real-tim
  - ???

Right now, the only serious drag with Studio is the text editor, and if there's a way to get Emacs's legendary LISP support with studio functionality it'd honestly be a compelling replacement to the more mature but limited OpenSCAD.

sjm-tl-gh, 2018-08-07:
If you just need a simple cycle of editing the model, rendering, and viewing, you should be able to drop down to the Guile layer of libfive.  Guile has Emacs support with Geiser, and you can load libfive as an extension in Guile:

https://github.com/libfive/libfive/blob/master/doc/guide.md#working-from-the-guile-command-line

Once you had an STL file rendered, you could use any external STL viewer (e.g. meshlab) to view it.  Your Scheme program can call the STL viewer with an external system call with the STL file as an argument:

`(system* "meshlab" "file.stl")`

This is similar to what I did to bootstrap a Common Lisp binding to libfive, called Tovero:

https://common-lisp.net/viewvc/tovero/trunk/examples/sphere.lisp?view=markup

In this model, the `view-mesh` function does a system call in CL, as above.

My company's package Tovero combines an Open Inventor CL binding (Clive), which eventually will have bi-directional interaction with the model from the Emacs REPL/viewer.  If there were a Guile binding to Open Inventor or similar scene graph, libfive could theoretically do the same.  Libfive Studio is much easier to use out of the box, but Clive/Tovero is more programmable (and complicated).

sjm-tl-gh, 2018-08-07:
One method of actually implementing a REST based viewer for libfive outside of Studio would be to have a program that ran as a HTTP server which accepted libfive Scheme models.  I think there is even a web server that runs Guile.  The server could render the model to X3DOM or WebGL with the Guile library and serve the 3D model back to your browser--I have seen this done in another modeling system.

ronasuti, 2018-08-08:
Y'know what, I'm a student on summer break, I've got time. I might just try to add the auto-reload function myself. I'll work from my fork and let @mkeeter know if I get a working prototype.

mkeeter, 2018-08-08:
@ronasuti Awesome, let me know how it goes!

I did something similar in fstl in [this commit](https://github.com/mkeeter/fstl/commit/c7302534e7ec48d1b46ca761d9ed6c6533b03407).

sjm-tl-gh, 2018-08-08:
So fstl would reload after calling `shape->mesh` in Guile?  That would be useful for using libfive with Emacs.

mkeeter, 2018-08-08:
I wasn't thinking about that, just giving hints on how to live-reloading within Studio itself.  Still, fstl could be a decent viewer for a Studio-free system.

sjm-tl-gh, 2018-08-08:
Reloading in Studio on a model file change would be a nice feature and is a quick way to get it working.

Keep in mind that using the file system as "IPC" would be less flexible than using something like a socket.

Eventually it would be nice to have a finer grained, dynamic control of remote updates to the Guile model--like redefining a single function without loading the whole model.  I have been thinking about using zeromq for this.

sjm-tl-gh, 2018-08-08:
@mkeeter Using fstl would have the overhead of using STL, but my point was that it might be able to be used today, without coding. :)

sjm-tl-gh, 2018-08-08:
By the way, I wasn't recommending that the OP use Tovero over libfive Studio either.  Clive/Tovero isn't targeted at end users--it is really meant to be integrated into design/manufacturing systems.  I mentioned it to try to show that it is possible to have close integration with Emacs and a Lisp REPL in a system like libfive Studio, and how we did it.

As I said, libfive Studio really is much better for end users without a lot of extra coding.

doug-moen, 2018-08-08:
@sjm-tl-gh said:
For future reference, here are two ways of integrating Libfive Studio with Emacs using Geiser as a REPL:

1. Same process (threading), everything in Scheme
   This is currently the way Clive/Tovero works

   * Write the viewer in Scheme
   * Run the viewer and REPL under the same process in different threads
   * Use threading primitives for coordinating the viewer thread/REPL thread
   * Use Scheme s-exps as a protocol

-----
Curv works the same way: the viewer and the REPL run in different threads within a single process.

The limitation I encountered was on macOS. Due to the way that the cocoa window system works, the viewer must run in the main thread. I don't use Qt, but I noticed from the documentation that it imposes a similar restriction. (I did not have this problem on Linux: there, it is okay for the REPL to run in the main thread, and the viewer in a different thread.)

So if the plan is to put these two threads into another application, like Emacs, then you might run into a problem if you want to run the application on macOS. You will have to find some way to convince the application to let your viewer code run in the application's main thread. Since the Studio viewer is written in Qt, and Qt imposes the restriction on all platforms that the GUI must run in the main thread, then this is not a macOS specific problem for Libfive Studio.

sjm-tl-gh, 2018-08-08:
@doug-moen I thought Matt would be more likely to balk at:

> * Write the viewer in Scheme

:) but thanks for the info.

Our software currently only runs on Linux using Xt/glut so I didn't know that about Qt.  #2 might be the better option for Libfive Studio on Qt.  I wonder if a separate light-weight viewer running on GLFW or glut might be better for a REPL (I don't know how Matt is rendering models in Libfive Studio, but it could be done with meshes or voxels like Clive/Tovero).

sjm-tl-gh, 2018-08-10:
Update:

I created a tiny project on Github which will be a prototype [Guile viewer](https://github.com/sjm-tl-gh/inspekt3d) for Libfive.  You can track my progress there or submit issues.

Not much to see yet, but I got Guile-OpenGL to work on Ubuntu 18.04.1 with Guile 2.2.3, and started with their VBO particle system example (which I thought would be most appropriate), but there are also versions for client arrays and legacy (glBegin/glEnd).  Guile-OpenGL wraps OpenGL 2.1 and Glut.

mkeeter, 2018-08-12:
I'm going to close this issue since we merged in #199.

The viewer looks cool!  Ping me when it's developed enough to include in the README.

closed 2018-08-12 by mkeeter