libfive/libfive #201: C API function missing from Guile module

sjm-tl-gh, 2018-08-15:
### Description

Currently, the libfive Guile module does not define a Scheme wrapper function for the function `libfive_tree_render_mesh`.  

To get my Guile viewer working using in-memory meshes, I was trying to use the Guile dynamic FFI to define this function as:

    (define %libfive-tree-render-mesh
      (pointer->procedure '*
                          (dynamic-func "libfive_tree_render_mesh" libfive-lib)
                          (list '* libfive-region3-struct-type float)))

The first element of the list in the last line is a pointer to a `libfive_tree` structure.  Currently, given a shape (class `<shape>`), I haven't figured out how to access the underlying `libfive_tree` pointer from Guile.  In C++, for example, in `scm_shape_to_mesh`, you are accessing it with `scm_get_tree`--which is missing from the Guile module as well.  Am I missing something obvious?

### Versions

- Operating system: N/A
- `libfive` commit hash:  d837499
  - Is this the latest commit to `master`?  yes
- Qt version (if applicable): N/A

mkeeter, 2018-08-16:
You're not missing anything obvious – right now, I don't expose a way to get the raw `libfive_tree` objects (which are just `Kernel::Tree*`) from the `<shape>` wrapped pointer class in Guile.

However, the following will probably work:
```
(use-modules (system foreign))
(dereference-pointer (unwrap-shape s))
```
This relies on the the fact that the first member variable in the `scm_shape` is itself a pointer (namely, the `libfive_tree` that you're looking for).

You can either use this or add bindings to `libfive_tree_render_mesh` in `libfive-guile.cpp`, though I'm not sure what you'd want it to return – I normally have the bindings return Guile-friendly types, but `libfive_mesh` is a big bag of data, and it would be expensive to convert into lists of vertices / indices.

sjm-tl-gh, 2018-08-16:
For reference, Guile-OpenGL uses bytevectors to create [packed binary structures](https://github.com/guildhall/guile-opengl/blob/master/gl/contrib/packed-struct.scm) for direct use in Guile with OpenGL.

mkeeter, 2018-08-16:
Interesting.  I'm going to close this for now, since you've solved your original issue – if you get BLOBs working and are interested in contributing it to the repo, let me know.

closed 2018-08-16 by mkeeter