libfive/libfive #202: How do you set a variable in a tree ?

dilevin, 2018-08-21:
Hi,

I really like LibFive but I cannot find an answer to a silly question. How do I setup a tree with a variable I can set using the C++ API. I have tried the following, which does not work. Any help would be much appreciated. 

Kernel::Tree angle = Kernel::Tree(0.f); 
Kernel::Tree shape = rotate2d(box({-0.25f, -1.f, -.25f},{0.25f, 0.0f, .25f}), angle); //This is the standard rotate 2D function but taking in a Tree, not a float, as the second parameter. 
angle = 0.5;
Region<3> r({ -5, -5, -5 }, { 5, 5, 5 });
auto mesh = Mesh::render(shape, r, 0.25);

It seems the mechanism for doing this is a bit more complicated but I'm having lots of difficulty figuring it out from the raw source code.

mkeeter, 2018-08-22:
Good question!  Can you explain a little more what you're trying to do, so I can understand if this is the right question to be answering?

In general, `Kernel::Tree` objects are immutable, so when you do this assignment, you're replacing `angle` with a new tree, rather than changing its value.  Like you've deduced, variables are the exception: they're placeholders that take their value when trees are converted into an evaluator.

Because they're an advanced feature, there's a bit more boilerplate in the API.  Here's your example, with detailed comments:

```cpp
Kernel::Tree angle = Kernel::var(); // No value here, it's just a placeholder
Kernel::Tree shape = rotate2d(box({-0.25f, -1.f, -.25f},{0.25f, 0.0f, .25f}), angle);

// Here's the variable bindings (still not bound):
std::map<Kernel::Tree::Id, float> vars;
vars[angle] = 0.5f;

// We have to manually construct the Evaluators used in meshing.  Since this is the
// advanced API, we manually select the number of threads, and pass in a flag
// for asynchronous cancellation.
unsigned workers = 8;
std::atomic_bool cancel(false);
std::vector<XTreeEvaluator, Eigen::aligned_allocator<XTreeEvaluator>> es;
es.reserve(workers);
// Calling the XTreeEvaluator constructor with vars as the second argument
// binds their values.  You can also call XTreeEvaluator::updateVars to change
// their values after the evaluator is constructed.
for (unsigned i=0; i < workers; ++i)
    es.emplace_back(XTreeEvaluator(t, vars));

Region<3> r({ -5, -5, -5 }, { 5, 5, 5 });

render(es.data(), r, 0.5, 1e-8, workers, cancel);
```

dilevin, 2018-08-23:
Ah, ok. I didn't understand the relationship between trees and evaluators. This clears everything up. Thanks!  

Just so I'm sure I understand:  different evaluators take in parameters (as a map) and evaluate a particular quantity (i.e JacobianEvaluator gives me the Jacobian of the tree evaluated at a point in space wrt to the current parameters stored in the map) ?

mkeeter, 2018-08-23:
Yup, that's right.  `Kernel::Tree` is a lightweight data structure that handles deduplication of math trees, arithmetic identities, etc.

The Evaluator nomenclature is admittedly a bit confusing right now.  There are a handful of `Evaluator` classes that do just one thing (e.g. `JacobianEvaluator`, `PointEvaluator`); there are also classes that bundle every `Evaluator` that you need for a particular operation (e.g. `XTreeEvaluator` holds multiple evaluators used for building meshes).

closed 2018-08-23 by mkeeter