libfive/libfive #205: Support GLSL 120 instead of 150?

rekado, 2018-08-25:
Currently, Studio crashes when it is used on a machine with a graphics chip that does not support GLSL 150 (such as the dated Thinkpad T60).  Would it be at all possible to port libfive to version 120 of the shader language or does libfive use features that absolutely require version 150?

mkeeter, 2018-08-25:
Does it at least print a helpful error message ([here](https://github.com/libfive/libfive/blob/master/studio/src/view.cpp#L238) in the code)?

It's may be possible to use older OpenGL versions, but I'm not interested in doing that work – OpenGL 3.2 has been released almost a decade, and I don't know offhand which features I'd have to work around if I tried to use 2.1 instead.

If you're interested in maintaining an OpenGL 2.1 port, let me know and I'll add a link in the README.

closed 2018-08-25 by mkeeter

sjm-tl-gh, 2018-08-30:
FWIW, the rudimentary Libfive [viewer](https://github.com/sjm-tl-gh/inspekt3d) I wrote in Guile Scheme uses OpenGL 2.1 and GLSL 120.