libfive/libfive #208: Studio: resize shape reference window?

omgitsraven, 2018-09-10:
I just launched Studio for the first time in a while, and it seems like it isn't possible to resize the Shape Reference window any more? I'm sure I used to be able to make it taller, to glance through more entries at once...

sjm-tl-gh, 2018-09-11:
Resizing of the shape reference window is still working for me on the following system:

Ubuntu 18.04
Libfive Revision: 01ace4b (latest commit as of today)
Qt package: 5.9.5 (latest Ubuntu 18.04 version)

omgitsraven, 2018-09-11:
Still not working for me on

macOS 10.13.6
01ace4b
Qt 5.11.1

omgitsraven, 2018-09-11:
Also: I reverted back to Qt 5.9.3 and rebuilt Studio, and the bug is still happening, so that's not it.

mkeeter, 2018-09-11:
Yup, I'm seeing it as well @omgitsraven – I'll see what I can find.