libfive/libfive #209: Generalized Sweeps Feature

sjm-tl-gh, 2018-09-11:
#### Description
This is a separate issue for implementing generalized sweeps as a feature in libfive.  As a starting goal, the idea would be to create solids in libfive by sweeping a closed 2D shape along a continuous 3D curve, such as Beziers and splines.  Sweeps are commonly used in 3D design, and would be an important feature to have in libfive.

#### Previous Work
There has already been some discussion of sweeps on this issue: #149.
 
There is also a libfive git branch **bezier-sweeps**, which uses an oracle to implement sweeps along a Bezier curve.  I'm not sure what the status of this branch is.

#### Papers:
- "Implicit Modeling of Swept Surfaces and Volumes" http://marchingcubes.org/images/8/8d/ImplicitModelingOfSweptSurfacesAndVolumes.pdf

- "Implicit Sweep Surfaces" http://unknownroad.com/publications/SweepsTRApril2005.pdf

sjm-tl-gh, 2018-09-18:
I spent some time looking at the **bezier-oracles** branch of Libfive.  The sweep is currently limited to quadratic Bezier curves.  @mkeeter I thought about trying to add a cubic Bezier next, similar to what you have done with cubics--are you aware of any issues with this?  Having a cubic Bezier could allow an acceptable approximation of arcs and NURBS curve paths (by curve subdivision).

This is a sweep of a sphere, which looks good:

    (shape-sweep (sphere 0.25) #[0 0 1] #[0 1 0] #[0 0 -1])

[image]

and a box:

    (shape-sweep (box #[-0.25 -0.25 -0.25] #[0.25 0.25 0.25])
                 #[0 0 1] #[0 1 0] #[0 0 -1])

[image]

Sweeping a 2D profile doesn't produce a 3D solid (without clamping, e.g. using a half-space), and it seems to flatten the aspect ratio of the profile as it moves towards the end-points:

    (shape-sweep (circle 0.25) #[0 0 1] #[0 1 0] #[0 0 -1])

[image]

Clamped swept circle:

    (define sweep-circle-hs
      (let* ((p0 #[0 0 1])
             (p1 #[0 1 0])
             (p2 #[0 0 -1])
             (v0 (- p0 p1))
             (vn0 (/ v0 (norm v0)))
             (v1 (- p2 p1))
             (vn1 (/ v1 (norm v1))))
        (intersection
         (half-space vn0 p0)
         (shape-sweep (circle 0.25) p0 p1 p2)
         (half-space vn1 p2))))
    sweep-circle-hs

[image]

Clamped rectangle sweep:

    (define sweep-rect-hs
      (let* ((p0 #[0 0 1])
             (p1 #[0 1 0])
             (p2 #[0 0 -1])
             (v0 (- p0 p1))
             (vn0 (/ v0 (norm v0)))
             (v1 (- p2 p1))
             (vn1 (/ v1 (norm v1))))
        (intersection
         (half-space vn0 p0)
         (shape-sweep (rectangle #[-0.25 -0.25] #[0.25 0.25]) p0 p1 p2)
         (half-space vn1 p2))))
    sweep-rect-hs

[image]

mkeeter, 2018-09-19:
Nice writeup, it looks like you've figured most stuff out.

A few implementation notes:

The biggest limitation to this branch is that it doesn't rotate the shape as it moves along the path.  You can see this most clearly in the box example: it's more like taking the Minkowsky sum of the path and the box, rather than a loft.

This is probably fixable in the transform, e.g. by having a way to calculate a rotated coordinate frame that's normal to the path, but I didn't get around to it.  The tricky part is picking the second axis – it's easy to find the direction of travel, but harder to pick an always-reasonable orthonormal basis.

Another issue that I didn't debug: the oracles can't be nested recursively, e.g. to get a 2D surface out.

Finally, I punted on the interval oracle evaluation – it always returns `[0,1]`, indicating that the closest point is somewhere along the path, but never gets more specific.

I don't see any reason why a cubic Bezier implemented along the same lines wouldn't work fine; hopefully my notes also suggest future work that could improve both behavior and performance.

mkeeter, 2018-09-21:
Nice, the Frenet-Serret formulas seem very relevant.  I'm not totally sure what's going on with the circle – I'll have to think about that a little more.

sjm-tl-gh, 2018-09-21:
Re: closest point for ```evalInterval```.  This paper might have some relevant information in the introduction regarding this task:

https://www.researchgate.net/publication/232638188_Point_inversion_and_projection_for_NURBS_curve_Control_polygon_approach

Also, the paper suggests to me the idea of possibly just supporting quadratic Beziers using an oracle, then doing subdivision at a higher level to get cubic Beziers and NURBS:

https://stackoverflow.com/questions/2009160/how-do-i-convert-the-2-control-points-of-a-cubic-curve-to-the-single-control-poi

sjm-tl-gh, 2018-10-03:
> This is probably fixable in the transform, e.g. by having a way to calculate a rotated coordinate frame that's normal to the path, but I didn't get around to it. The tricky part is picking the [...]

I've been thinking  about rotating the swept shape's "natural" normal vector (possibly supplied to the sweep function) to the curve's tangent vector at each ```t``` to get the change in basis.  For example, a circle on the XY plane would have [0 0 1] as it's normal.  It's not obvious to me where in code to apply "the transform", but I have been looking into it.  Any ideas for how to to this would be appreciated.

scheming-pony, 2019-05-28:
I thought I would document a couple of ideas on sweeps under this thread after reading [1].

Conceptually, and related to the "functional" in F-rep, sweeps can be formed by:

1) Applying a bijective mapping function ("twist") to the cross-product of a 2D shape with a 1D segment

2) Projecting the 4D volume swept by a 3D solid (through time) back to 3D
   
The Pasko paper refers to a concept called "R-functions" (to implement the Cartesian product and projections for example).  I know little about R-functions or how they would fit into Libfive.

[1] ["Function Representation in Geometric Modeling" by Pasko, et al.](https://www.researchgate.net/publication/220067662_Function_representation_in_geometric_modeling_concepts_implementation_and_applications)

scheming-pony, 2019-05-29:
Another idea that the paper in the previous post gave me was the use of a computer algebra system (CAS) to work with functions symbolically at a layer above Libfive.  Sussman's Scmutils or Axiom come to mind.

doug-moen, 2019-05-29:
R-Functions are Rvachev functions. Rvachev invented the function representation of geometric shapes in the 1960s. We've known the equation for a circle for hundreds of years, but Rvachev discovered the equation for a rectangle, and a new field was born. R-functions should work fine in libfive. Old papers that talk about R-functions are usually concerned about C-1 continuity, because the old methods for rendering distance functions (originally, R-functions) required C-1 continuity. So they didn't use 'max' to compute the union of two shapes, because that introduces discontinuities.

scheming-pony, 2019-05-29:
@doug-moen thanks for describing R-functions!  I will probably read more about them--even just to understand the "old papers" better.  (I've been thinking about starting a F-rep mailing list for discussions like this but thought it might be too low of traffic to bother with. :)

Again one of the key ideas I got from the paper is that we are dealing with functions here, and that brings a lot of power it itself (e.g. through function composition).

doug-moen, 2019-05-29:
@scheming-pony : There is an F-Rep mailing list at f-rep@googlegroups.com, although for some reason it's private. It's a group of developers (like me) who are all developing their own F-Rep systems. I can ask the owner to send you an invite if you like. It is very low volume (last message was September 2018).

There is somewhat more frequent F-Rep theory discussion on the Curv mailing list, although the main purpose of the group is to discuss my own F-Rep software, [Curv](https://github.com/curv3d/curv). That's a public google group, at https://groups.google.com/forum/#!forum/curv

scheming-pony, 2019-05-29:
@doug-moen Yes, I would be interested in an invite to the F-Rep mailing list-my email address is in my GitHub profile.  Thanks.

I have seen Curv, thanks for the link to its mailing list--that would interest me too.

mkeeter, 2019-05-30:
Good discussion going on here – I haven't thought about sweeps in a little while, but I'm glad other folks are continuing to ponder it.

I'd love to see some CAS integration into libfive / Studio.  Right now, the underlying tree representation does a bit of rearranging, e.g. collapsing things where it can prove equivalents, but it's very primitive.  

At one point in the past, I tried to use Python + Sage to generate [implicit representation of NACA airfoils](https://gist.github.com/mkeeter/355a269fd0c4c5ff82c3f413159b547c) (but didn't quite succeed); it would be really cool to have that power built-in.

Thanks for the historical details, Doug – you're a well of useful papers and context 😃

scheming-pony, 2019-05-30:
I was thinking about dusting off my MIT Scheme binding to Libfive (done back when it was Ao).  I thought it might be interesting to visualize code using Scmutils/Libfive from [SICM](https://mitpress.mit.edu/books/structure-and-interpretation-classical-mechanics) which I have read or [Functional Differential Geometry](https://mitpress.mit.edu/books/functional-differential-geometry) which I own but haven't read yet.

@mkeeter That was an interesting attempt at representing airfoils.  The professor who started Sage left his academic position to commercialize Sage, IIRC.  I have started reading the references from the Pasko paper I mentioned and some older ones on  modeling sweep using implicit equations.  So, hopefully I can add more on this topic later.  One higher level goal of mine is to use Libfive for a threaded fastener library (nuts and bolts).

scheming-pony, 2019-06-01:
I have been talking to various academics who do geometric modeling this week, and in fact over the last couple of years, and it is surprising how low the awareness is of F-rep modeling of those I have encountered.  A professor in computational geometry (who has worked on a CGAL module) kind of understood that it was related to SDF theory.

My own reading about F-rep this week led me to think about this quote from SICP (Section 1.1):

   Every powerful language has three mechanisms for accomplishing this [combining simple ideas into more complex ones]:
      * primitive expressions, which represent the simplest entities the language is concerned with,
      * means of combination, by which compound elements are built from simpler ones, and
      * means of abstraction, by which compound elements can be named and manipulated as units.

It really does seem like the F-rep paradigm does qualify as a powerful language under these critera--and should be more widely known in geometric modeling. /shrug

scheming-pony, 2019-06-01:
I was describing this issue to the CG researcher, and I told him, "One disadvantage [of  F-rep vs. B-rep] is that constructs like sweeps appear to be awkward to formulate as implicit equations".

But I think that illustrates a flaw in thinking about this problem.

The real question may be, "is it true that a **general** sweep of a 2D profile (ignoring the 3D case for now) can be modeled as a bijective function applied to a Cartesian product (as in Pasko's paper).".  If so, the solution should be straightforward.  Correct me if I'm wrong--but it seems like extrusion-z gives us the Cartesian product.  The (likely non-trivial) problem now becomes how to formulate the bijective map given a space curve, then applying this map to the Cartesian product.

You could apply similar reasoning to the 3D case.  So, again I could be wrong, but in line with my previous post--we are not just dealing with just implicit equations in F-rep.  It brings in the power of functional composition.

doug-moen, 2019-06-01:
There are two principle ways to represent a 3D solid using math.

In **boundary representation** (B-Rep), you represent the boundary of the shape. In B-Rep, it is easy to traverse the boundary, but it is difficult/expensive to answer the question: given a point (x,y,z), is the point outside, on the boundary, or inside the shape. A B-Rep only represents a 2D surface embedded in 3D space. It might not represent a solid object at all, it is expensive to make this determination. There might be holes (it isn't watertight), or the surface might be non-orientable. Examples of B-Rep are triangle meshes (OpenSCAD), splines (CAD programs) and function representations where two synthetic parameters (u,v) are mapped onto points on the surface of the shape.

In **volume representation** (V-Rep), you represent the shape as a solid object, with an inside and an outside. It is trivial to answer this question: given a point (x,y,z), is it inside, outside, or on the boundary. Examples of V-Rep are voxel grids, and the particular type of F-Rep used by Libfive and Curv.

A corollary is that there are two main classes of function representations, B-Rep and V-Rep. The V-Rep function representation, used by Curv and Libfive, is a **signed distance field** (SDF), where the result returned by the function is not just a signifier of inside/outside/boundary, but which also gives you the distance of the point to the boundary in some metric space.

If you scour the literature, you will find V-Rep function representation that is not SDF, and you will find B-Rep function representation. The methods used with non-SDF function representation are not generally applicable to SDF systems.

You mentioned the book Functional Differential Geometry. I don't know differential geometry, but based on quickly skimming the material, this is a mathematics of B-Rep function representation, where the functions are required to be higher-order differentiable. In other words, you are (for example) representing a 2D surface embedded in 3D shape using a function. Since the surfaces must be smoothly differentiable, you can't exactly represent shapes that have edges or vertices, such as a cube or cylinder or cone. It is possible that differential geometry may have some application within the context of an SDF modelling system, but at present I don't know what that connection is. I'd be interesting in learning the application, if you are planning to investigate this.

doug-moen, 2019-06-01:
> One disadvantage [of F-rep vs. B-rep] is that constructs like sweeps appear to be awkward to formulate as implicit equations.

In general this is true, but there are special cases that are easy.

> The real question may be, "is it true that a general sweep of a 2D profile (ignoring the 3D case for now) can be modeled as a bijective function applied to a Cartesian product (as in Pasko's [...]

`extrusion-z` is the Cartesian product of a 1D line segment with a 2D shape, giving a 3D shape. That's easy.

The Curv library has an operation that sweeps a 2D shape along a 2D path embedded in the X-Y plane, giving a 3D shape. I call it `perimeter_extrude`. This is simple, but maybe it's not obvious. This operation is my invention, I've never seen it mentioned in any published paper or in any other SDF software system.

It's true that you can define a general sweep of a 2D shape through 3D space as a Cartesian product, but the resulting product exists in higher dimensional space, and must be projected down to 3D space. The problem is this projection: it is very expensive (complex and slow). If you can find a fast method for doing projections, that would be great.

sjkelly, 2019-06-01:
@doug-moen one benefit of differentiability of an FRep is towards mesh-free solutions to BVPs. Vadim Shapiro (who AFAIK, along with Pasko, brought Rvachev's "inverse problem of analytic geometry" to the English academic literature) has quite a lot of good work in this area. His site: https://spatial.engr.wisc.edu/  

The thesis by Michael Freytag at UW is a pretty comprehensive exposition of the mesh-free analysis method using Rvachev Clipping, e.g. (max -> max formulated Ck continuous). From what I can tell from their results the sharpness of a mesh is unaffected by the rvachev clipping. However, I have not been able to reproduce this yet.

sjkelly, 2019-06-01:
More saliently, there is a paper that build upon the ideas by Rvachev, Shapiro, et. al. on using continuity to solve the orientation problem by Ryan Schmidt (of MeshMixer fame) that may help http://www.unknownroad.com/publications/SweepsGraphite05.pdf

doug-moen, 2019-06-01:
@sjkelly Thanks for the link to the paper by Schmidt & Wyvill (Generalized Sweep Templates for Implicit Modeling)

> A limitation of the sweep technique we have described is that numerical root-finding is usually necessary in the case of curved trajectories. Root finding is expensive and requires an analytic expression for the curve tangent to function robustly. A key avenue for future work is a more general implicit sweep formulation that does not rely on root-finding.

Analytic root finding is not general purpose: you are very limited in the set of functions that will work. But it might be worth doing for special cases.

Numerical root finding requires iteration (or recursion) plus conditional expressions. AFAIK this is impossible in Libfive, although it is supported by Curv.

sjkelly, 2019-06-01:
I'm not too familiar with the back end of libfive. FWIW I'm doing most of my experimentation in Julia and can make use of the auto differentiation libraries to allow me to use Newton's method for fast convergence on root finding. 

I really like what everyone here is doing and it is really nice to see different approaches to these problems along with the general infrastructure. I'll join the curv list so we can exchange more ideas there.

doug-moen, 2019-06-02:
@sjkelly
> FWIW I'm doing most of my experimentation in Julia and can make use of the auto differentiation libraries to allow me to use Newton's method for fast convergence on root finding.

That's interesting. I'd like to hear more about these experiments.

scheming-pony, 2019-06-02:
@sjkelley Do you have a direct Julia binding to Libfive?  I have heard it is fairly easy to bind to C in Julia.

scheming-pony, 2019-06-02:
It's also interesting to see that Shapiro has done work on analogs to FEM using mesh-free techniques.  In other words, engineering analysis (like stress calculations) of F-reps.

scheming-pony, 2019-06-02:
Re: http://www.unknownroad.com/publications/SweepsGraphite05.pdf

I read this paper last week, and I guess I'm thick, but I didn't understand what they meant by "sweep templates."  How is it related to the sweeping space curve in generalized sweeps?

scheming-pony, 2019-06-02:
@doug-moen Why not FVR  "functional volumetric representation"--I think mesh free is a decent term too, but why keep reminding people you are not **meshing**? :)

doug-moen, 2019-06-02:
> I read this paper last week, and I guess I'm thick, but I didn't understand what they meant by "sweep templates."

quote (section 2.2): "Solid modeling by sweeping a 2D area (commonly referred to as a sweep template) along a 3D trajectory is a well known technique in computer graphics"

scheming-pony, 2019-06-02:
@doug-moen Thanks ( "cross section"), I thought it was some more formal definition I was missing.

Speaking of quotes, in paragraph 1 of Pasko "Function representation..." (referenced above):

"The properties of these geometric objects are studied in differential geometry."

This was my motivation for investigating the use of Libfive with Scmutils in reading FDG.  (Sussman/Wisdom use FDG to help explain general relativity and quantum field theory) And of course, F-rep has its origins in the modeling of physics--so also kind of a general intuition.

scheming-pony, 2019-06-02:
Back on topic, I want to create a threaded fastener library.  This Libfive Issue 209 needs to be resolved before I can do that.

Towards a resolution of this issue, my question remains:
> "is it true that a general sweep of a 2D profile (ignoring the 3D case for now) can be modeled as a bijective function applied to a Cartesian product (as in Pasko's paper)?"

The Pasko '95 paper has a (simple) example of such a sweep, which I am now trying to reproduce in Libfive with  bijective function applied to a Cartesian product of the the 2D profile and a line segment.

I'll look at sweeping 3D objects as described in Pasko '95 after looking at the 2D case.

scheming-pony, 2019-06-03:
@doug-moen wrote:
> It's true that you can define a general sweep of a 2D shape through 3D space as a Cartesian product, but the resulting product exists in higher dimensional space, and must be projected down to 3D [...]

I would argue with you here--the technique in Pasko '95 is to *compose* a bijective mapping with a Cartestian product of a 2D profile with a line segment (generalized or not TBD).  The Cartesian product generates a right 3D prism of the 2D profile and the bijective mapping "twists" that object.  Both the bijective mapping and Cartesian product are operations that are **closed** on the system described in the paper.  No higher level space.

The other technique Pasko '95 describes is the sweep of a 3D solid through time, yielding a 4D object that would have to be projected back to 3D .  Since Libfive is a 3D system (Pasko '95 describes an N-D system), a CAS might help do the projection.  Pasko '95 references a paper [Wang 1984] that appears to propose this idea.

scheming-pony, 2019-06-03:
> If you scour the literature, you will find V-Rep function representation that is not SDF, and you will find B-Rep function representation. The methods used with non-SDF function representation are not generally applicable to SDF systems.

I'm trying to go from "scour" to "pick a reasonable starting place in the literature, look backward at what has been done, then move forward catch up to where we are now".  I thought about pitching the development of a textbook to someone like Shapiro. Each one of us having to do a graduate level literature review I think lowers access to volumetric modeling.  Maybe there already is a text book.
 (This [text book](https://www.amazon.com/Geometry-Dummies-Mark-Ryan/dp/0470089466) might most appropriate for my level.) 

Terminology has indeed been an issue with trying to understand the principles of "volumetric rendering".  Libfive is described as using "functional [adjective] representation", whereas the Pasko '95 paper describes itself as "function [noun] representation."  But then the README for Libfive links to the "function representation" Wikipedia page.  That describes the trinary (inside/surface/outside) approach of Pasko/Hyperfun.  But the Wikipedia article links to the concept of SDF, which implies some relationship to function representation.  If you look at HyperFun code, it uses similar "implicit-style" expressions (e.g. for a sphere) like Libfive.  [Stack overflow, exiting.]

Is what Pasko '95 describes mathematically equivalent (or a subset/superset thereof) as a representation to Libfive?  Are the techniques presented in Pasko '95 applicable to Libfive?  The answer I have determined so far is yes/maybe/no.

doug-moen, 2019-06-04:
> The technique in Pasko '95 is to _compose_ a bijective mapping with a Cartestian product of a 2D profile with a line segment (generalized or not TBD). The Cartesian product generates a right 3D [...]

This is just the standard technique that everybody uses today, but it has problems.

You mentioned "terminology has been an issue". So, in modern terminology, "Cartestian product of a 2D profile with a line segment" is just called "extrude" (that's AutoCAD terminology, but Curv and Libfive use the same name). "Bijective mapping": I just call this a [transformation](https://github.com/curv3d/curv/blob/master/docs/shapes/Transformations.rst); Wikipedia calls it a [geometric transformation](https://en.wikipedia.org/wiki/Geometric_transformation). I further subdivide transformations into "similarity transformations" and "deformations", but "deformation" is not standard terminology.

If you compose an extrude with a deformation, you get something that looks like a sweep, but I don't consider it a proper sweep: it is distorted. In a proper sweep (of a 2D shape along a 3D path or trajectory), the 2D shape ought to be normal to the path at every point. Take a cylinder oriented along the Z axis; offset it from the Z axis; apply Barr's "twist" deformation to twist it around the Z axis. You get the so-called swept solid in figure 7 on page 439 of the Pasco paper. It's kind of like sweeping a circle along a helix, but the circles are not normal to the helical path, and it looks flattened.

The other problem with this technique: if you are sweeping something that is not a circle, you probably want to control the rotational orientation of the 2D shape at each point on the path so that the sweep "looks right". I guess I don't have the proper terminology to describe this.

I agree with the general approach of composing an extrusion with *something* in order to obtain a sweep. However, the *something* must describe not just the path or trajectory through space, but it must also associate each point on the path with an "up vector" that determines the orientation of the 2D shape at that point. In the simple case, the up vector always has magnitude 1 and is normal to the path. And with a proper mathematical understanding, maybe there is a way to automatically assign an appropriate up vector to each point in the path so that you get something that looks good.

doug-moen, 2019-06-04:
Instead of reading Pasco, which I don't think has aged well, I would instead suggest John C. Hart's classic paper "Sphere Tracing". The techniques he describes for constructing and rendering signed distance fields are still close to the state-of-the-art. Hart's methods work extremely well on the GPU, so Curv and a lot of other modern tech is based on his work. Also, Hart created a breakthrough: his sphere-tracing technique is still the fastest and most accurate way to render hypertextures and fractals.

http://mathinfo.univ-reims.fr/IMG/pdf/hart94sphere.pdf

doug-moen, 2019-06-04:
Another limitation of Pasko's approach of composing an extrusion with a geometric transformation is: this doesn't work if the sweep path is self-intersecting, because now you are mapping two points in the extrusion to a single point in the result, and that is not a bijective mapping.

doug-moen, 2019-06-04:
The question I asked about automatically computing the orientation of the 2D sweep template at each point in the trajectory were addressed earlier in this thread, last year. See [Frenet–Serret formulas](https://en.wikipedia.org/wiki/Frenet%E2%80%93Serret_formulas) and other links. Also, this involves differential geometry, and I had previous asked how D-G is relevant to 3D geometric modelling.

scheming-pony, 2019-06-04:
@doug-moen Re: self intersection...good observation--bijective is 1-1, so Pasko '95 sweep most likely isn't general.  By the way, like Pasko's paper--I am old and not aging well..but I like to think I'm still useful.

scheming-pony, 2019-06-05:
Interestingly, and in keeping with the overall circularity of this issue--my thinking is back to higher dimensions.  A little outside of the treatment of 2D sweeps along a space curve in Pasko '95, the idea of composing the bijective with the extruded profile might be accomplished in 4D then projected back--an additional dimension might disambiguate the self-intersection.

BUT, as I learned again, anytime I see the need for an inverse function (again Pasko '95), I should be suspicious that there might be discontinuities in some fashion.  The HyperFun system allows kD, whereas Libfive has 3D so the idea of adding a CAS looks better and better.

To save a round of discussion @doug-moen -- this might have been what you were saying earlier, but I was back then only considering the direct method in Pasko '95 as the general case.  Thanks a lot for your contributions here!

scheming-pony, 2019-06-05:
@mkeeter By the way, one of the researchers I talked to last week asked me if Libfive had a convolution function.  Another one, V. Shapiro, is working on an interesting grant:

> The objective of this research project is to establish a universal formal computational model for the information that flows from design into additive manufacturing.  This model plays a role similar to that of the Church-Turing model that underlies general computation.

https://www.nsf.gov/awardsearch/showAward?AWD_ID=1344205

which is related to a fuzzy idea I had  many years ago about extending CSG to lambda calculus that led me to Libfive and F-rep.  He apparently does have some answers the the questions this issue poses.

scheming-pony, 2019-06-05:
@doug-moen By the way I don't consider "Cartesian product" and "bijective map" as dated terminology--they are formalisms of set theory.  In some ways, despite the (perhaps) dated rendering ideas, Pasko's work is appealing due to its formality.  "Extrusion" seems to have come from mechanical design--and is in fact a physical technique for forming material (although physically a 3D -> 3D deformation).

scheming-pony, 2019-06-07:
The use of Frenet–Serret (or not) would be incorporated when specifying the space curve as a bijective mapping.  In FreeCAD you can optionally use Frenet-Serret (again see https://www.freecadweb.org/wiki/Part_Sweep#Frenet).

scheming-pony, 2019-06-11:
```
( Convolution function? )
 -----------------------
    o
     o
      o

  (\     /)
  % ^---^ %
  %{ @ @ }%%
   ( *** ) \%%
   / ** /   \%%
  / ** /|    \---
  \oo/ sjm
```

scheming-pony, 2019-06-11:
Humor aside, the idea of using convolution on functions was suggested to me as a potential for doing sweeps--and is an interesting idea.

Outside of sweeps, what power would adding a convolution operator to Libfive give us for modeling in other ways?

doug-moen, 2019-06-11:
I think it would be useful to talk about concrete examples. The original problem statement in this issue was "to create solids in libfive by sweeping a closed 2D shape along a continuous 3D curve, such as Beziers and splines".

So here are some concrete examples:
1. Sweep a 2D shape along a [helix](https://en.wikipedia.org/wiki/Helix).
2. Sweep a 2D shape along a [trefoil knot](https://en.wikipedia.org/wiki/Trefoil_knot), which has the parametric equation:
   ```
   x = sin t + 2 sin 2t
   y = cos t + 2 cos 2t
   z = - sin 3t
   ```
3. Sweep a 2D shape along an arbitrary cubic [Bezier curve](https://en.wikipedia.org/wiki/B%C3%A9zier_curve) in 3-space.

The helix is the simplest problem, because it is not too difficult to derive a distance function for a helix, whereas there is no practical way to analytically derive a formula for the distance function of a trefoil knot from the parameteric equation that I gave. I figure that if you have a distance function for the continuous 3D curve that you want to sweep along, then maybe it is easier to derive a distance function for the corresponding swept solid.

The Bezier curve is the hardest problem, because unlike the trefoil knot, a Bezier can have cusps and self intersections.

In all three cases, we want the 2D shape to be orthogonal to the 3D curve at all points in the curve. The solution should also take into account the orientation of the 2D shape as it is being swept. Either the orientation is automatically computed to give aesthetically pleasing results, or there is some way to control the orientation.

A simplified version of these three problems is to sweep a circle along the curve. In this case, the orientation doesn't matter.

@scheming-pony, you have mentioned the ideas of using a bijective map, and using convolution, but there are a lot of missing details in how to use these ideas to derive the distance function of a swept solid.

scheming-pony, 2019-06-11:
> you have mentioned the ideas of using a bijective map, and using convolution, but there are a lot of missing details in how to use these ideas to derive the distance function of a swept solid.

Yes.  So?  How is that worse than Monte-Carlo or empirical reasoning by citing papers and concrete examples to lead us to a solution?  We probably need both.

Unlike in proofs you read in books the real process of deductive reasoning ain't pretty.

[Don't flame me for this being about surface representation, its only an idea.]
https://repositories.lib.utexas.edu/handle/2152/3619

scheming-pony, 2019-06-12:
From the Descartes problem (or rather the inverse), I had to go back to [Euclid](https://en.wikipedia.org/wiki/Pons_asinorum):

"The name of this statement is also used metaphorically for a problem or challenge which will separate the sure of mind from the simple, the fleet thinker from the slow, the determined from the dallier, to represent a critical test of ability or understanding."

```
 ------------------------------
(Are sweeps my "pons asinorum"?)
 ------------------------------
    o
     o
      o

  (\     /)
  % ^---^ %
  %{ @ @ }%%
   ( *** ) \%%
   / ** /   \%%
  / ** /|    \---
  \oo/ sjm
```

scheming-pony, 2019-06-14:
On to Isaac Newton...what does this remind of us of in classical mechanics?

```
 ------------------------------------------------------
( Integration?  So we sort of need a, uhh,...calculus? )
 ------------------------------------------------------
    o
     o
      o

  (\     /)
  % ^---^ %
  %{ @ @ }%%
   ( *** ) \%%
   / ** /   \%%
  / ** /|    \---
  \oo/ sjm
```

scheming-pony, 2019-06-14:
My ideas so far:

(0) Try to pose problem in terms of higher level language (e.g. CAS, functional composition, calculus).

(1) bijective-function o cartesian-product [possibly promoted to 4D to disambiguate self-intersection, then projected back to 3D]
    a) bijective-function = [some combination of] path (P) and orientation
    b) cartesian-product = 2D-profile x ( ||P|| unit-line-segment)

(2) Convolve a 2D profile (convolution kernel) with an infinitesimal tube.  Unknown how to control orientation.

doug-moen, 2019-06-14:
@scheming-pony Thanks for sharing your ideas. Here are my (preliminary) ideas.

The sweep function is `sweep(profile, path, alignment)` where:
* `profile` is the signed distance function for a 2D shape;
* `path` is a parametric function that maps some parameter `t` onto a point in 3-space, for some range of values for `t`;
* `orientation` controls the orientation of the profile as it is swept along the path. By default, the profile is kept orthogonal to the path at all points along the path, and the rotation of the profile at each path point is automatically chosen to create an esthetically pleasing result. In the general case, the user could provide an orientation function that maps `t` onto a 3D rotation (quaternion). Details of this parameter are still to be decided.

And the result is a signed distance field for a 3D shape.

In the simplest case, where the profile is a 2D point with zero area (eg, a circle with zero radius), then the result is the signed distance from an arbitrary 3D point to the closest point on the path. This is root finding. In most cases, there is no way to determine a closed form (symbolic) solution for the resulting signed distance function. For example, there is no practical closed form solution for the trefoil knot. This means the result is an approximation.

There are a range of methods for computing an approximate solution. At one end of the range, we'll be using a pure function that incorporates some approximate root finding algorithm, like Newton's method. These solutions will be slow, but very memory efficient. At the other end of the range, we'll be using a 3D volumetric data structure to cache the results of computations at various points in 3D space, and the resulting signed distance function will look up its argument in this data structure to retrieve some cached values, before computing the signed distance. These methods will be faster, at the expense of using more memory.

The most extreme data-structure solution is to compute a voxel grid containing a discretization of the final signed distance field. The signed distance function of the swept volume looks up its (x,y,z) argument in the voxel grid and uses trilinear interpolation to compute a signed distance value.
* You could start by creating a voxel grid containing density values at each voxel (a density grid), where the density is 1 if the voxel is entirely inside the swept solid, or 0 if the voxel is entirely outside the swept solid, or a fraction > 0 and < 1 if the voxel straddles the boundary.
* Then the density grid is converted to the voxel form of a signed distance field using a "distance transform".
* To construct the density grid, you could iteratively stamp copies of the profile shape into the density grid while slowly increasing the value of `t`. It's a bit more complicated than that, details TBD.

Self-intersection can occur for both the helix and the trefoil-knot examples in my previous post, if the profile is large enough, even though the path itself is not self intersecting. So we need some story about how self intersection is handled.

Ideally, when N branches of the sweep overlap at the same point, then the N branches are unioned together. The voxel grid solution that I outlined would work that way.

However, I think that unioning would be difficult to achieve for a pure functional solution using Newton's method. For that case, I think it would be easier if, instead of unioning all the branches, we choose one branch as the winner. We'll pick the branch corresponding to the point in the path that is closest to the current point, and we'll make an arbitrary choice if two branches are equidistant.

scheming-pony, 2019-06-14:
@doug-moen Wow, thanks for all the info--I'll read it more closely later (my wife is starting to call herself an "F-rep widow").

Anyway, Dr. Shapiro kindly referred me to one of his papers when I inquired about his work and the significance of Rvachev and R-functions:

https://www.researchgate.net/publication/228386894_Semi-analytic_geometry_with_R-functions

Section 1 has an interesting discussion of the so-called "inverse Descartes problem" where we try to yield a single function given a shape.  It also has some of the history of R-functions.

mkeeter, 2021-06-02:
Here's an interesting relevant paper that just came out:
http://dgp.toronto.edu/~sgsellan/pdf/swept-volumes-low-res.pdf

sene-je, 2022-06-21:
Interesting, oui.  Potentially patent encumbered, also oui.  See their project's Github project, it's been a while since I looked, being off the 'net for over a year (but man, the work you can accomplish).  I think they are trying to write their own Libfive too, to replace yours, but I'm just conjecturing.

I've made some progress myself.  My AVC (artificial visual cortex) isn't patented (by me, I'm a Knuth-ist WRT patents):

https://github.com/SanderMertens/flecs/discussions/750

Wild, but it is kind of fun to conjecture.  I forgot to add my idea about using quantum computing tech for superimposition of the 3D object swept for all time [0 .. 1] ([-inf .. inf]?), but that previous post is so long I forget.  I should trademark "0xtrees" (pun)--Meager patented octrees and had a company called Octree, Inc (not joking).  I dusted off his 40 year old idea (somebody on a game dev site said, "Of what use is a 40 year old algorithm?"  It's unpatented, well thought-out and like Euclid and Descartes and Pythogoras (still unpatented, despite the efforts of my old schoolmates).  I use all of these every day, slow but steady).  Read link for details.

Regards.

sene-je, 2022-06-22:
I also worked through "Solid modeling for mold design and manufacture" by Weiping Wang, 1984 using the Julia package Symbolics.jl to produce envelopes--knowing the limitations (see the paper).  It was an interesting exercise.

> quantum computing tech for superimposition 
You can't go out and buy a Dell Quantum 704, but maybe it could be simulated (albeit slowly) for when you can.

My best current way is octrees and CSG, but as the patented? paper points out there is aliasing and higher time complexity.

sene-je, 2022-06-22:
And finally some "code poetry" and why we should (or shouldn't) have developers study literature:

https://www.mail-archive.com/mit-scheme-devel@gnu.org/msg01683.html

"How we treat the least of those in our society reflects it moral compass.  How the rich treat themselves is irrelevant, as it is approximately a constant.""  -- Victor Hugo, approximately

sene-je, 2022-06-22:
https://github.com/orgs/github-community/discussions/19179

sene-je, 2022-06-22:
^^^ Bump -- update.

ramcdona, 2025-05-08:
Has there been any progress on swept volume generation in libfive or fidget?

I am interested in sweeping a volume along a trajectory -- not the process of sweeping a 2d profile to create a volume (generalized extrusion).

For bonus points, I am interested in not just simple trajectories, but with some dispersion added.  For example, sweep the given volume along a trajectory, but the trajectory is uncertain -- represented by a 15deg cone around the trajectory.