libfive/libfive #210: Minimum boost version, how to link newere boost?

kasbah, 2018-09-20:
### Description

Having some trouble building on Ubuntu 16.04, I do have g++-7 which seemed to help. I am not sure what the minimum boost version is though and how I would link the newer one I donwloaded from boost.org. 

```
/home/kaspar/projects/libfive/libfive/libfive/src/render/brep/xtree_pool.cpp:210:22    required from ‘static typename Kernel::XTree<N>::Root Kernel::XTreePool<N>::build(Kernel::XTreeEvaluator*, Kernel::Region<N>, double, double, unsigned int, std::atomic_bool&, Kernel::ProgressCallback) [with unsigned int N = 2; typename Kernel::XTree<N>::Root = Kernel::XTree<2>::Root; std::atomic_bool = std::atomic<bool>; Kernel::ProgressCallback = std::function<void(float)>]’
/home/kaspar/projects/libfive/libfive/libfive/src/render/brep/xtree_pool2.cpp:15:17    required from here
/usr/include/boost/lockfree/stack.hpp:73:5: error: static assertion failed: boost::has_trivial_assign<T>::value
     BOOST_STATIC_ASSERT(boost::has_trivial_assign<T>::value);
     ^
/usr/include/boost/lockfree/stack.hpp:74:5: error: static assertion failed: boost::has_trivial_destructor<T>::value
     BOOST_STATIC_ASSERT(boost::has_trivial_destructor<T>::value);
```
### Versions

- Operating system: Ubuntu 16.04
- `libfive` commit hash:  d9033df 
  - Is this the latest commit to `master`?  yes

mkeeter, 2018-09-20:
This [StackOverflow discussion](https://stackoverflow.com/questions/3016448/how-can-i-get-cmake-to-find-my-alternative-boost-installation) seems relevant.

kasbah, 2018-09-20:
Thanks for your answers. libboost on 16.04 is at 1.58.0. Armed with that Stackoverflow link I'll have another go at compiling in the the downloaded one. I'll let you know how I get on.

mkeeter, 2018-09-21:
@sjm-tl-gh I agree, it makes sense to check for a minimum Boost version to the build system, though it's hard to tell which one it is – between 1.58.0 and 1.65, there's no mention of lockfree in any of the release notes!

I'm going to pin it to 1.65.0, since my rule of thumb is to be compatible with the latest Ubuntu LTS.

mkeeter, 2018-10-07:
@kasbah Any luck?

I'm closing this issue because we've found the root cause, but feel free to comment if you're having other problems.

closed 2018-10-07 by mkeeter