libfive/libfive #224: Move "Other Projects" section of README.md to it's own file

sjm-tl-gh, 2018-10-16:
I have written 2 new viewers for Python/C, and there is a Python binding started.  For better organization  I thought I could move the "Other Projects" section in `README.md` to a new document `OtherProjects.md` in the `./doc` subdirectory and put a link in the README.  Should I do a PR for this?

[OtherProjects.md5](https://github.com/libfive/libfive/files/2485130/OtherProjects.txt)

mkeeter, 2018-10-16:
I'd prefer to keep it in the README, since it's immediate evidence for the project's robustness / community (albeit a small one).

closed 2018-10-16 by mkeeter

sjm-tl-gh, 2018-10-17:
That makes sense, and hopefully bindings for more mainstream languages like Python will increase the size of the Libfive community.  The OtherProjects.md attachment contains links to my new viewers, if you want to add those.  The C5H12 library has the potential to add viewing to any language with an FFI.  Thanks!

mkeeter, 2018-10-17:
Thanks, I added those to the README.