libfive/libfive #238: Error compiling on Mac OS 10.13.6

omgitsraven, 2018-12-06:
### Description

Tried compiling the latest commit today, and it's giving me a "file not found" error on `solver.hpp`. To be safe, I deleted everything local and cloned it totally fresh, and it's still happening. (This is as of commit 350860fb7de300b8eae02efec24101d4ec2880d8 .)

Here's my build command:
```
cd Documents/GitHub/libfive
mkdir build
cd build
cmake -DCMAKE_PREFIX_PATH=/usr/local/Cellar/qt/5.11.1  ..
make
```
and here's the result:
(note: the warnings about unused parameters aren't anything new, they've been happening for a while but it's always compiled fine regardless)
```
-- The C compiler identification is AppleClang 10.0.0.10001145
-- The CXX compiler identification is AppleClang 10.0.0.10001145
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc
-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Boost version: 1.67.0
-- Found PkgConfig: /usr/local/bin/pkg-config (found version "0.29.2") 
-- Checking for module 'guile-2.2>=2.2.1'
--   Found guile-2.2, version 2.2.3
-- Checking for module 'eigen3>=3.2.92'
--   Found eigen3, version 3.3.4
-- Found ZLIB: /usr/lib/libz.dylib (found version "1.2.11") 
-- Found PNG: /usr/local/lib/libpng.dylib (found version "1.6.34") 
Checking dependencies:
  libfive:		✓
  libfive-guile:	✓
  Studio:		✓
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/andrew/Documents/GitHub/libfive/build
[  1%] Generating version.c, _version.c
Scanning dependencies of target five
[  2%] Building CXX object libfive/src/CMakeFiles/five.dir/eval/base.cpp.o
[  3%] Building CXX object libfive/src/CMakeFiles/five.dir/eval/deck.cpp.o
[  4%] Building CXX object libfive/src/CMakeFiles/five.dir/eval/eval_interval.cpp.o
[  5%] Building CXX object libfive/src/CMakeFiles/five.dir/eval/eval_jacobian.cpp.o
[  6%] Building CXX object libfive/src/CMakeFiles/five.dir/eval/eval_array.cpp.o
[  7%] Building CXX object libfive/src/CMakeFiles/five.dir/eval/eval_deriv_array.cpp.o
[  8%] Building CXX object libfive/src/CMakeFiles/five.dir/eval/eval_deriv.cpp.o
[  9%] Building CXX object libfive/src/CMakeFiles/five.dir/eval/eval_feature.cpp.o
[  9%] Building CXX object libfive/src/CMakeFiles/five.dir/eval/eval_point.cpp.o
[ 10%] Building CXX object libfive/src/CMakeFiles/five.dir/eval/tape.cpp.o
[ 11%] Building CXX object libfive/src/CMakeFiles/five.dir/eval/feature.cpp.o
[ 12%] Building CXX object libfive/src/CMakeFiles/five.dir/oracle/transformed_oracle.cpp.o
[ 13%] Building CXX object libfive/src/CMakeFiles/five.dir/render/discrete/heightmap.cpp.o
[ 14%] Building CXX object libfive/src/CMakeFiles/five.dir/render/discrete/voxels.cpp.o
[ 15%] Building CXX object libfive/src/CMakeFiles/five.dir/render/brep/contours.cpp.o
In file included from /Users/andrew/Documents/GitHub/libfive/libfive/src/render/brep/contours.cpp:17:
/Users/andrew/Documents/GitHub/libfive/libfive/src/../include/libfive/render/brep/dual.hpp:88:33: warning: 
      unused parameter 't' [-Wunused-parameter]
void Dual<2>::handleTopEdges(T* t, Mesher& m)
                                ^
/Users/andrew/Documents/GitHub/libfive/libfive/src/../include/libfive/render/brep/dual.hpp:88:44: warning: 
      unused parameter 'm' [-Wunused-parameter]
void Dual<2>::handleTopEdges(T* t, Mesher& m)
                                           ^
2 warnings generated.
[ 16%] Building CXX object libfive/src/CMakeFiles/five.dir/render/brep/progress.cpp.o
[ 17%] Building CXX object libfive/src/CMakeFiles/five.dir/render/brep/mesh.cpp.o
In file included from /Users/andrew/Documents/GitHub/libfive/libfive/src/render/brep/mesh.cpp:17:
/Users/andrew/Documents/GitHub/libfive/libfive/src/../include/libfive/render/brep/dual.hpp:88:33: warning: 
      unused parameter 't' [-Wunused-parameter]
void Dual<2>::handleTopEdges(T* t, Mesher& m)
                                ^
/Users/andrew/Documents/GitHub/libfive/libfive/src/../include/libfive/render/brep/dual.hpp:88:44: warning: 
      unused parameter 'm' [-Wunused-parameter]
void Dual<2>::handleTopEdges(T* t, Mesher& m)
                                           ^
2 warnings generated.
[ 18%] Building CXX object libfive/src/CMakeFiles/five.dir/render/brep/neighbor_tables.cpp.o
[ 19%] Building CXX object libfive/src/CMakeFiles/five.dir/render/brep/dc/intersection_aligner.cpp.o
[ 20%] Building CXX object libfive/src/CMakeFiles/five.dir/render/brep/dc/marching.cpp.o
[ 21%] Building CXX object libfive/src/CMakeFiles/five.dir/render/brep/dc/dc_contourer.cpp.o
[ 22%] Building CXX object libfive/src/CMakeFiles/five.dir/render/brep/dc/dc_mesher.cpp.o
[ 23%] Building CXX object libfive/src/CMakeFiles/five.dir/render/brep/dc/dc_neighbors.cpp.o
[ 24%] Building CXX object libfive/src/CMakeFiles/five.dir/render/brep/dc/dc_tree2.cpp.o
[ 25%] Building CXX object libfive/src/CMakeFiles/five.dir/render/brep/dc/dc_tree3.cpp.o
[ 26%] Building CXX object libfive/src/CMakeFiles/five.dir/render/brep/simplex/simplextree.cpp.o
/Users/andrew/Documents/GitHub/libfive/libfive/src/render/brep/simplex/simplextree.cpp:22:10: fatal error: 
      'libfive/render/brep/simplex/solver.hpp' file not found
#include "libfive/render/brep/simplex/solver.hpp"
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
1 error generated.
make[2]: *** [libfive/src/CMakeFiles/five.dir/render/brep/simplex/simplextree.cpp.o] Error 1
make[1]: *** [libfive/src/CMakeFiles/five.dir/all] Error 2
make: *** [all] Error 2
```

### Versions

- Operating system: Mac OS 10.13.6
- `libfive` commit hash: 350860fb7de300b8eae02efec24101d4ec2880d8
  - Is this the latest commit to `master`?  yes
- Qt version (if applicable): 5.11.1

mkeeter, 2018-12-06:
Oh, that was dumb – I renamed a bunch of files, but had copies locally in `/usr/local/include/libfive` which were still being included.  The commit above should fix things.

closed 2018-12-06 by mkeeter

omgitsraven, 2018-12-07:
That fixed it, thanks :)