libfive/libfive #239: A file that crashes Studio on launch

omgitsraven, 2018-12-07:
### Description

I was just opening an old Studio file that I hadn't touched in a while, and I found that it consistently crashes Studio on launch.

I narrowed it down to an interaction between two things:

This crashes:
```
(define (other)
  (remap-shape ((sphere 1) x y z)
    x y (- z  (/ 0.1 y))
  )
)

(blend-rough (sphere 1) (other) 0.3)
```

Even though the identity remap is safe:
```
(define (other)
  (remap-shape ((sphere 1) x y z)
    x y z
  )
)

(blend-rough (sphere 1) (other) 0.3)
```

and so is the 'dangerous' remap, as long as it isn't being blended:
```
(remap-shape ((sphere 1) x y z)
  x y (- z  (/ 0.1 y))
)
```

Here are all three examples, for testing:
[crashable.zip](https://github.com/libfive/libfive/files/2658035/crashable.zip)

### Versions

- Operating system: Mac OS 10.13.6
- `libfive` commit hash: c0846280b017eb637beffc1674a19b0bed8d9aaf
  - Is this the latest commit to `master`?  yes
- Qt version (if applicable): 5.11.1

mkeeter, 2018-12-07:
Thanks, I'll check it out!

tachiniererin, 2019-04-02:
This still happens in https://github.com/libfive/libfive/blob/460717e1d16f6b1c3af25325e119878d25ce94ff/libfive/src/render/brep/dc/dc_tree.cpp#L858 when it gets passed a -1 from somewhere. For me it only crashes sometimes when loading the file (but when it does not crash it also does not render at all). I can reproduce it every time though when switching the meshing algorithm to iso-simplex and back again.

omgitsraven, 2019-04-02:
I'm pretty sure the "sometimes renders nothing" is unrelated—that happens to me all the time, even with files that work. It feels like a 50% chance that I'll have to quit the program and relaunch it when opening a file (I'll try switching meshing, thanks!)

I hadn't bothered reporting it because it was easy to work around and I figured it was happening to everyone, but maybe I should make a post anyway :)

klpeders, 2019-06-09:
Any news on this one?
I am hitting this bug consistently without using any advanced features:

```scheme
(set-bounds! [-10 -10 -10] [10 10 10])
(set-quality! 8)
(set-resolution! 6)

(define h 4)
(define r 2.5)
(define w 8)

(difference
  (box-centered #[w w h])
  (circle r)
  (reflect-z (cone-z r h) (/ h 2))
)
```

The core point to `dc_tree.cpp:861`:

```
Thread 183 "Thread (pooled)" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fff3effd700 (LWP 24164)]
Kernel::DCTree<3u>::vert (this=0x7fff34001e20, i=<optimized out>)
    at github/libfive/libfive/src/render/brep/dc/dc_tree.cpp:861
861	    return this->leaf->verts.col(i);
```

mkeeter, 2019-06-10:
Interestingly, I can't reproduce that crash on my machine.

My theory for this crash is that there are `NaN` or `infinity` values in here (in @omgitsraven 's example, injected by the `(/ 0.1 y)` operation), which aren't correctly handled in meshing.

In your example, it's probably the `cone` that's causing the problem – making a cone with a taper-to-zero (as the standard library does now) creates a nasty distance field right at the point.

Can you try this alternate definition of `cone`?

```
(define (cone r height base)
  (define angle (atan height r))
  (define half 
    (sequence (half-space [0 1 0])
      (rotate-z angle)
      (move [(- r) 0 0])))
  (define c (revolve-y
    (intersection
      half
      (reflect-x half)
      (half-space [0 -1 0]))))
  (sequence c
    (reflect-yz)
    (move base)))
```

If this fixes your crash, we can bring it into the standard library (without fixing the underlying kernel bug, which will remain until I have time to track it down).

klpeders, 2019-06-11:
Eh??  I see what you mean!  I have to change the `set-bounds!` to this:

```scheme 
(set-bounds! [-10 -10 -10] [40 20 10])
```

to get the crash now.

klpeders, 2019-06-11:
The alternative `cone` implementation works - Now I can make screws again :+1: 

[image]

klpeders, 2019-06-11:
:man_facepalming: The solution doesn't work when I integrate back into my gadget:

```scheme
(set-bounds! [-10 -10 -10] [40 20 10])
(set-quality! 8)
(set-resolution! 6)

(define (cone r height base)
  (define angle (atan height r))
  (define half 
    (sequence (half-space [0 1 0])
      (rotate-z angle)
      (move [(- r) 0 0])))
  (define c (revolve-y
    (intersection
      half
      (reflect-x half)
      (half-space [0 -1 0]))))
  (sequence c
    (reflect-yz)
    (move base)))

(define (vscrew  hw hh d)
  (define r  (/ d 2))
  (define hr (/ hw 2))
  (union
    (circle r)
    (reflect-z (cone-z hr hh)  (/ hh 2))
  )
)

;(vscrew 5 4 2) ;; Crash!
```

There also seem to be some interaction between `set-bounds!` and `set-resolution!`

For example if resolution is 10, then un-commenting the `vscrew` doesn't cause a crash.
It also seems that only when the bounds is exactly `[-10 -10 -10] [40 20 10]` there is crash.

This seem to be a *stable* workaround:

```scheme
(reflect-z (cone-z hr hh)  (/ (+ hh 0.0001) 2))
```

mkeeter, 2019-06-11:
Yup, that agrees with the theory that it's instability at the tip of the cone – the crash probably happens only if a sample happens to line up exactly with that point.  I'm glad you found a workaround, but we'll leave this issue open for the kernel robustness fixing.

klpeders, 2019-07-06:
Just rebuild from a fresh pull. With this version i get crashes left and right. Until now I have seen two different crashes :
The usual one:

```
Thread 81 "Thread (pooled)" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fff3e7fc700 (LWP 27536)]
Kernel::DCTree<3u>::vert (this=0x7fff10002ee0, i=<optimized out>)
    at github/libfive/libfive/src/render/brep/dc/dc_tree.cpp:840
840	    return this->leaf->verts.col(i);
```

and the new one:

```
Thread 1026 "Thread (pooled)" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fff50ff9700 (LWP 2201)]
Kernel::Feature::Feature (this=0x7fff50ff62f0, d=..., a=..., b=...)
    at github/libfive/libfive/src/eval/feature.cpp:36
31	    for (auto& e : b.epsilons)
32	    {
33	        bool found = false;
34	        for (auto& e_ : epsilons)
35	        {
36	            if (e == e_)
37	            {
38	                found = true;
39	                break;
40	            }
```
and this one:

```
Thread 307 "Thread (pooled)" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fff3dbdf700 (LWP 2663)]
0x00007ffff7c08ec0 in std::__shared_count<(__gnu_cxx::_Lock_policy)2>::__shared_count (__r=..., this=0x7fff3dbdca08)
    at /usr/include/c++/9/bits/shared_ptr_base.h:1167
1162		__shared_ptr(const __shared_ptr<_Yp, _Lp>& __r,
1163			     element_type* __p) noexcept
1164		: _M_ptr(__p), _M_refcount(__r._M_refcount) // never throws
1165		{ }
1166	
1167	      __shared_ptr(const __shared_ptr&) noexcept = default;
1168	      __shared_ptr& operator=(const __shared_ptr&) noexcept = default;
1169	      ~__shared_ptr() = default;
1170	
1171	      template<typename _Yp, typename = _Compatible<_Yp>>

(gdb) bt
#0  0x00007ffff7c08ec0 in std::__shared_count<(__gnu_cxx::_Lock_policy)2>::__shared_count(std::__shared_count<(__gnu_cxx::_Lock_policy)2> const&)
    (__r=..., this=0x7fff3dbdca08)
    at /usr/include/c++/9/bits/shared_ptr_base.h:1167
#1  0x00007ffff7c08ec0 in std::__shared_ptr<boost::container::small_vector<Kernel::Intersection<3u>, 4ul, boost::container::new_allocator<Kernel::Intersection<3u> > >, (__gnu_cxx::_Lock_policy)2>::__shared_ptr(std::__shared_ptr<boost::container::small_vector<Kernel::Intersection<3u>, 4ul, boost::container::new_allocator<Kernel::Intersection<3u> > >, (__gnu_cxx::_Lock_policy)2> const&)
    (this=0x7fff3dbdca00) at /usr/include/c++/9/bits/shared_ptr_base.h:1167
#2  0x00007ffff7c08ec0 in std::shared_ptr<boost::container::small_vector<Kernel::Intersection<3u>, 4ul, boost::container::new_allocator<Kernel::Intersection<3u> > > >::shared_ptr(std::shared_ptr<boost::container::small_vector<Kernel::Intersection<3u>, 4ul, boost::container::new_allocator<Kernel::Intersection<3u> > > > const&) (this=0x7fff3dbdca00) at /usr/include/c++/9/bits/shared_ptr.h:129
#3  0x00007ffff7c08ec0 in Kernel::DCTree<3u>::intersection(unsigned int) const
    (this=this@entry=0x7fff2c15cd20, edge=<optimized out>)
    at github/libfive/libfive/src/render/brep/dc/dc_tree.cpp:856
#4  0x00007ffff7c08f21 in Kernel::DCTree<3u>::intersection(unsigned int, unsigned int) const (this=0x7fff2c15cd20, a=<optimized out>, b=4)
    at /usr/include/c++/9/array:189
```

It is a little tedious to build a minimal example when the editor crashes at seemingly unimportant changes:

```scheme

```

mkeeter, 2019-07-06:
Interesting – there are two new changes that came in recently.  Can you try reverting `9242e4c6e49968118df149431a499a5311cf212d` and seeing if that fixes it?  If not, it's probably something that came in with PR #286 .

It looks like you meant to paste a minimal example, but I don't see anything at the bottom of your post.

klpeders, 2019-07-06:
I am on Fedora 30, x64. gcc version 9.1.1

Sorry about the example, that is gone. 
When buiding I get a few warnings:

```
In file included from github/libfive/libfive/src/../include/libfive/render/brep/hybrid/hybrid_tree.hpp:19,
                 from github/libfive/libfive/src/render/brep/hybrid/hybrid_tree.cpp:15,
                 from github/libfive/libfive/src/render/brep/hybrid/hybrid_tree3.cpp:10:
github/libfive/libfive/src/../include/libfive/render/brep/simplex/qef.hpp: In member function ‘Kernel::QEF<N>::Solution Kernel::QEF<N>::solveBounded(const Kernel::Region<N>&, double) [with unsigned int N = 2]’:
github/libfive/libfive/src/../include/libfive/render/brep/simplex/qef.hpp:360:18: warning: ‘out’ may be used uninitialized in this function [-Wmaybe-uninitialized]
  360 |         Solution out;
      |                  ^~~
```

After reverting 9242e4c6e49968118df149431a499a5311cf212d, i still see the crashes when loading old designs. 

Running the unittest with the 9242e4c6e49968118df149431a499a5311cf212d patch:

```
-------------------------------------------------------------------------------
Cache: performance of commutative tree balancing
-------------------------------------------------------------------------------
github/libfive/libfive/test/cache.cpp:411
...............................................................................

github/libfive/libfive/test/cache.cpp:439: FAILED:
  REQUIRE( false )
with messages:
  std::chrono::duration_cast<std::chrono::microseconds>(dt) .count() := 1100
  (0x44c)
  std::chrono::duration_cast<std::chrono::microseconds>(dt_) .count() := 16548
  (0x40a4)

-------------------------------------------------------------------------------
Heightmap::render: Performance
-------------------------------------------------------------------------------
github/libfive/libfive/test/heightmap.cpp:259
...............................................................................

benchmark name                                  iters   elapsed ns      average 
-------------------------------------------------------------------------------
sphere                                              1     61103864   61.1039 ms 
Menger sponge                                       1    733564926   733.565 ms 

-------------------------------------------------------------------------------
SimplexTree<3>: assignIndices with cell collapsing
-------------------------------------------------------------------------------
github/libfive/libfive/test/simplex.cpp:322
...............................................................................

github/libfive/libfive/test/simplex.cpp:349: FAILED:
  REQUIRE( (a.first - b.first).norm() < 1e-9 )
with expansion:
  0.3124999997 < 0.000000001
with messages:
  a.first := -0.625
  0.9375
   0.625
  b.first := -0.625
    1.25
   0.625
  k.first.first->region.lower.transpose() := -0.625  0.625      0
  k.first.first->region.upper.transpose() :=     0  1.25 0.625
  (a.first - b.first).norm() := 0.3124999997

-------------------------------------------------------------------------------
Solver::findRoot
  Sum-of-squares performance
-------------------------------------------------------------------------------
github/libfive/libfive/test/solver.cpp:85
...............................................................................

benchmark name                                  iters   elapsed ns      average 
-------------------------------------------------------------------------------
SOS root                                            1       447644   447.644 µs 

===============================================================================
test cases:    179 |    177 passed | 1 failed | 1 failed as expected
assertions: 334713 | 334711 passed | 1 failed | 1 failed as expected
```

This is a little different than without the patch. After revert:, 

```
[...]
github/libfive/libfive/test/cache.cpp:439: FAILED:
  REQUIRE( false )
with messages:
  std::chrono::duration_cast<std::chrono::microseconds>(dt) .count() := 1030
  (0x406)
  std::chrono::duration_cast<std::chrono::microseconds>(dt_) .count() := 15503
  (0x3c8f)
[...]
benchmark name                                  iters   elapsed ns      average 
-------------------------------------------------------------------------------
sphere                                              1     46459960     46.46 ms 
Menger sponge                                       1    748290607   748.291 ms 
[...]
benchmark name                                  iters   elapsed ns      average 
-------------------------------------------------------------------------------
SOS root                                            1       418390    418.39 µs 

-------------------------------------------------------------------------------
Abs and skew applied to Oracle: Render and compare (cube as oracle)
-------------------------------------------------------------------------------
github/libfive/libfive/test/transformed_oracle.cpp:199
...............................................................................

github/libfive/libfive/test/transformed_oracle.cpp:106: FAILED:
  REQUIRE( valid )
with expansion:
  false
with messages:
  point := 0
  0
  0
  oFeatures.size() := 6
  cFeatures.size() := 4

===============================================================================
test cases:    179 |    176 passed | 1 failed | 2 failed as expected
assertions: 334363 | 334360 passed | 1 failed | 2 failed as expected
```

klpeders, 2019-07-06:
git bisect just finished:
Last good commit: 
0abc38789ac25499a555507189d99f1c9ea20b13 : More renaming to prepare for new inheritance
First bad commit:
8edf9c28ed8a773ee944c6d0716b9f39c46346c8 : All of the FeatureEvaluator tests now pass, though others fail.

klpeders, 2019-07-06:
The unittest of this *last good* state (where things don't crash) have this:

```
-------------------------------------------------------------------------------
QEF::solveBounded
  Circle outside of region
-------------------------------------------------------------------------------
github/libfive/libfive/test/qef.cpp:219
...............................................................................

github/libfive/libfive/test/qef.cpp:245: FAILED:
  REQUIRE( sol.error > 0 )
with expansion:
  -0.0 > 0
with messages:
  sol.position := 1.5
  2.7
  sol.value := -0.9999999448
  sol.rank := 2
  sol.error := -0.0
```

mkeeter, 2019-07-06:
Thank for the details!

This implies that the major reorganization in PR #286 is responsible for the crashes that you're seeing.  Still, it's very strange that the unit tests look okay (those failures are unrelated, and aren't segfaults), but Studio is crashing continuously.  

Can you nuke the entire build folder and do a clean rebuild, just to make sure that it's not a badly cached compilation object?

klpeders, 2019-07-07:
For the bisect the script would create independent build directories for each try. (I discovered this currious thing that building from scratch is faster than rebuilding!?)

I finally managed to simplify a test-case. Unfortunately it require manual editing:

```scheme
(difference
  (cylinder 3 .1)
  (rectangle-centered-exact  #[1 2])
)
```

You will notice that output is not exactly what you'd expect:
[image]

Now change any of the constants - for example change the constant '2' to '1'  then '3'. The exact values doesn't seem to matter for this. But changing the value twice when the full shape is visible does...

This reproduce the SIGSEGV in

```
feature.cpp:36            if (e == e_)
```

klpeders, 2019-07-07:
Running ./libfive/test/libfive-test  'last-good' and 'master' commits in DEBUG mode:
Release:
```
sphere                                              1     61103864   61.1039 ms 
Menger sponge                                       1    733564926   733.565 ms
```
DEBUG:
```
sphere                                              1   3965973076    3.96597 s 
Menger sponge                                       1 108562611092    1.80938 m
```
Wooh! A release build is 20.000 times faster than DEBUG?

After this there is a crash using both git versions :
```
-------------------------------------------------------------------------------
Mesh::render (different algorithms)
-------------------------------------------------------------------------------
libfive/libfive/test/mesh.cpp:105
...............................................................................

libfive/libfive/test/mesh.cpp:105: FAILED:
due to a fatal error condition:
  SIGSEGV - Segmentation violation signal

===============================================================================
test cases:   94 |   92 passed | 1 failed | 1 failed as expected
assertions: 8307 | 8305 passed | 1 failed | 1 failed as expected
```

Only difference in RELEASE is the number of test cases:
```
==============================================================
test cases:   90 |   88 passed | 1 failed | 1 failed as expected
assertions: 8241 | 8239 passed | 1 failed | 1 failed as expected
```

This exception seems related to executing some code path that is normally not executed:

```c
gdb --arg ./libfive/test/libfive-test  'Mesh::render (different algorithms)'
[...]
Thread 50 "libfive-test" received signal SIGSEGV, Segmentation fault.
std::__atomic_base<Kernel::SimplexTree<3u>*>::load (__m=std::memory_order_seq_cst, 
    this=0x10) at /usr/include/c++/9/bits/atomic_base.h:740
740		return __atomic_load_n(&_M_p, int(__m));
(gdb) l
735	      {
736		memory_order __b = __m & __memory_order_mask;
737		__glibcxx_assert(__b != memory_order_release);
738		__glibcxx_assert(__b != memory_order_acq_rel);
739	
740		return __atomic_load_n(&_M_p, int(__m));
741	      }
742	
743	      _GLIBCXX_ALWAYS_INLINE __pointer_type
744	      load(memory_order __m = memory_order_seq_cst) const volatile noexcept
(gdb) bt
#0  std::__atomic_base<Kernel::SimplexTree<3u>*>::load (
    __m=std::memory_order_seq_cst, this=0x10)
    at /usr/include/c++/9/bits/atomic_base.h:740
#1  std::atomic<Kernel::SimplexTree<3u>*>::load (this=0x10, 
    __m=std::memory_order_seq_cst) at /usr/include/c++/9/atomic:519
#2  0x00007ffff7825686 in Kernel::assignIndicesWorker<3u> (tasks=..., index=..., 
    done=..., cancel=...)
    at libfive/libfive/src/render/brep/simplex/simplex_tree.cpp:733
#3  0x00007ffff782449f in Kernel::SimplexTree<3u>::assignIndices(Kernel::BRepSettings const&) const::{lambda()#1}::operator()() const (this=0x10ef7b8)
    at libfive/libfive/src/render/brep/simplex/simplex_tree.cpp:877
#4  0x00007ffff78329ec in std::__invoke_impl<void, Kernel::SimplexTree<3u>::assignIndices(Kernel::BRepSettings const&) const::{lambda()#1}>(std::__invoke_other, Kernel::SimplexTree<3u>::assignIndices(Kernel::BRepSettings const&) const::{lambda()#1}&&) (
    __f=...) at /usr/include/c++/9/bits/invoke.h:60
```

klpeders, 2019-07-07:
Test also crash:
1. SimplexPool
1. SimplexTree<2>::assignIndices
1. SimplexTree<3>: assignIndices with cell collapsing (**before** banner?)
1. SimplexMesher<3>: tricky shape (**before** banner?)
1. SimplexMesher<3>: sphere (**before** banner?)

```c
Thread 3 "libfive-test" received signal SIGSEGV, Segmentation fault.
0x0000000000b9e441 in __gnu_cxx::__exchange_and_add (__mem=0x63206c6c65632033, 
    __val=-1) at /usr/include/c++/9/ext/atomicity.h:49
49	  { return __atomic_fetch_add(__mem, __val, __ATOMIC_ACQ_REL); }
(gdb) l
44	  // __exchange_and_add_dispatch
45	  // __atomic_add_dispatch
46	#ifdef _GLIBCXX_ATOMIC_BUILTINS
47	  static inline _Atomic_word 
48	  __exchange_and_add(volatile _Atomic_word* __mem, int __val)
49	  { return __atomic_fetch_add(__mem, __val, __ATOMIC_ACQ_REL); }
50	
51	  static inline void
52	  __atomic_add(volatile _Atomic_word* __mem, int __val)
53	  { __atomic_fetch_add(__mem, __val, __ATOMIC_ACQ_REL); }
(gdb) bt
#0  0x0000000000b9e441 in __gnu_cxx::__exchange_and_add (__mem=0x63206c6c65632033, 
    __val=-1) at /usr/include/c++/9/ext/atomicity.h:49
#1  0x0000000000b9e4dc in __gnu_cxx::__exchange_and_add_dispatch (
    __mem=0x63206c6c65632033, __val=-1) at /usr/include/c++/9/ext/atomicity.h:82
#2  0x0000000000bb6261 in std::_Sp_counted_base<(__gnu_cxx::_Lock_policy)2>::_M_release (this=0x63206c6c6563202b) at /usr/include/c++/9/bits/shared_ptr_base.h:152
#3  0x0000000000ba9e9d in std::__shared_count<(__gnu_cxx::_Lock_policy)2>::~__shared_count (this=0x10e64d8, __in_chrg=<optimized out>)
    at /usr/include/c++/9/bits/shared_ptr_base.h:730
#4  0x00007ffff76b10b2 in std::__shared_ptr<Kernel::AssignIndexTask<2u>, (__gnu_cxx::_Lock_policy)2>::~__shared_ptr (this=0x10e64d0, __in_chrg=<optimized out>)
    at /usr/include/c++/9/bits/shared_ptr_base.h:1169
#5  0x00007ffff76b10ce in std::shared_ptr<Kernel::AssignIndexTask<2u> >::~shared_ptr
    (this=0x10e64d0, __in_chrg=<optimized out>)
    at /usr/include/c++/9/bits/shared_ptr.h:103
#6  0x00007ffff771fca8 in Kernel::AssignIndexTask<2u>::~AssignIndexTask (
    this=0x10e6488, __in_chrg=<optimized out>)
    at libfive/libfive/src/render/brep/hybrid/hybrid_tree.cpp:847
#7  0x00007ffff780a59c in boost::lockfree::stack<Kernel::AssignIndexTask<2u>, boost::lockfree::fixed_sized<true> >::node::~node (this=0x10e6480, __in_chrg=<optimized out>)
    at /usr/include/boost/lockfree/stack.hpp:87
```

Also crash, look like 'Mesh::render (different algorithms)' above.

1. SimplexMesher<3>: box with problematic edges
1. SimplexMesher<3>: tricky shape (**before* banner?)
After commenting these out, all the tests pass
```
  #Mesh::render (different algorithms)
  #SimplexPool
  #SimplexMesher<3>: box with problematic edges
  #SimplexTree<2>::assignIndices
  #SimplexMesher<3>: tricky shape
  #SimplexTree<3>: assignIndices with cell collapsing
  #SimplexMesher<3>: cell collapsing and vertex placement
  #SimplexMesher<3>: sphere
  #SimplexMesher<3>: cylinder meshing
  #SimplexTree<3>: vertex placement in centered cylinder
  #SimplexTree<3>: meshing + cell collapsing
  #SimplexTree<3>::assignIndices
  #SimplexTree<3>::leafLevel
  #SimplexMesher (smoke test)
  #SimplexMesher: sphere-box intersection vertex placement
  #SimplexMesher: edge pairing
  #SimplexMesher: menger sponge
```

klpeders, 2019-07-07:
:woman_facepalming: Yeah all of that was somehow an artefact of '-O0'. Changing the debug flags to recommended switches for debugging - fixed all the crashes:

```patch
-    set(CMAKE_CXX_FLAGS_DEBUG "-O0")
+    set(CMAKE_CXX_FLAGS_DEBUG "-Og -ggdb3")
```

mkeeter, 2019-07-08:
Thanks, with the test case
```
(difference
  (cylinder 3 .1)
  (rectangle-centered-exact  #[1 2])
)
```
I could reliably tgiger a crash!

It seems to be fixed with 791bef96cb35ba8c34acd180341b4f97b14e788f – want to see if that resolves your issue?  (this is definitely unrelated to @omgitsraven original posting, for what it's worth).

klpeders, 2019-07-08:
That commit seems to fix the new crash isues.
It is a bit difficult to see the difference between @omgitsraven 's report and this without loading the core into the debugger.
@omgitsraven issue is this here and now throw this assert:

```c++
Studio: libfive/libfive/src/render/brep/dc/dc_mesher.cpp:126: void Kernel::DCMesher::load(const std::array<const Kernel::DCTree<3>*, 4>&, unsigned int) [with Kernel::Axis::Axis A = Kernel::Axis::Z; bool D = true]: Assertion `vi != -1' failed.
Studio: libfive/libfive/src/render/brep/dc/dc_mesher.cpp:126: void Kernel::DCMesher::load(const std::array<const Kernel::DCTree<3>*, 4>&, unsigned int) [with Kernel::Axis::Axis A = Kernel::Axis::Z; bool D = false]: Assertion `vi != -1' failed.
/home/kp/bin/launch-studio: line 1: 13847 Aborted                 (core dumped) QT_SCALE_FACTOR=1.5 studio/Studio
```

mkeeter, 2019-08-19:
This should be fixed in e66bd75c243341e37be2e4c8908c009f74f34bb4.

The shape is still bad, because there's a division by zero, so I think that some vertices trend off towards infinity or even `NaN`, but it doesn't crash anymore!

I've written a new helper function to check consistency (which is the root cause behind crashes like this), so it should be easier to debug in the future.

closed 2019-08-19 by mkeeter