libfive/libfive #243: Crashes immediately on OS X 10.14.2

InformationOcean, 2019-01-08:
### Description

[Crashes immediately after double clicking on the application icon]
[Steps to reproduce]
[I expected that it will run.  Crashed?  All the time]
[Jan  8 01:25:03 Daddy com.apple.xpc.launchd[1] (com.apple.xpc.launchd.oneshot.0x10000064.Studio[27080]): Service exited due to SIGILL | sent by exc handler[27080]
J]

### Versions

- Operating system: [Mac OS X 10.14.2]
- `libfive` commit hash: [here]
  - Is this the latest commit to `master`?  [yes]
- Qt version (if applicable): [here]

### Additional Information

Process:               Studio [27132]
Path:                  /private/var/folders/*/Studio.app/Contents/MacOS/Studio
Identifier:            com.impraxical.studio
Version:               [thread-pool]
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           Studio [27132]
User ID:               501

Date/Time:             2019-01-08 01:26:17.072 -0500
OS Version:            Mac OS X 10.14.2 (18C54)
Report Version:        12
Anonymous UUID:        31047825-BF74-3F58-C121-F6A3BD334938

Sleep/Wake UUID:       DD86FE53-2A5B-4A88-BF80-0AD3375607E1

Time Awake Since Boot: 1100000 seconds
Time Since Wake:       130000 seconds

System Integrity Protection: enabled

Notes:                 Translocated Process

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_INSTRUCTION (SIGILL)
Exception Codes:       0x0000000000000001, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Illegal instruction: 4
Termination Reason:    Namespace SIGNAL, Code 0x4
Terminating Process:   exc handler [27132]

Application Specific Information:
dyld: launch, running initializers
/private/var/folders/y6/xyrsj9311n93frnrh_3t87gw0000gn/T/AppTranslocation/35A90EE1-3EA2-4DFE-B5C1-A1902C88209A/d/Studio.app/Contents/MacOS/../Frameworks/libfive.dylib

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libfive.dylib                 	0x000000010770ba33 0x107688000 + 539187
1   dyld                          	0x000000011758fcc8 ImageLoaderMachO::doModInitFunctions(ImageLoader::LinkContext const&) + 518
2   dyld                          	0x000000011758fec6 ImageLoaderMachO::doInitialization(ImageLoader::LinkContext const&) + 40
3   dyld                          	0x000000011758b0da ImageLoader::recursiveInitialization(ImageLoader::LinkContext const&, unsigned int, char const*, ImageLoader::InitializerTimingList&, ImageLoader::UninitedUpwards&) + 358
4   dyld                          	0x000000011758b06d ImageLoader::recursiveInitialization(ImageLoader::LinkContext const&, unsigned int, char const*, ImageLoader::InitializerTimingList&, ImageLoader::UninitedUpwards&) + 249
5   dyld                          	0x000000011758b06d ImageLoader::recursiveInitialization(ImageLoader::LinkContext const&, unsigned int, char const*, ImageLoader::InitializerTimingList&, ImageLoader::UninitedUpwards&) + 249
6   dyld                          	0x000000011758a254 ImageLoader::processInitializers(ImageLoader::LinkContext const&, unsigned int, ImageLoader::InitializerTimingList&, ImageLoader::UninitedUpwards&) + 134
7   dyld                          	0x000000011758a2e8 ImageLoader::runInitializers(ImageLoader::LinkContext const&, ImageLoader::InitializerTimingList&) + 74
8   dyld                          	0x0000000117579774 dyld::initializeMainExecutable() + 199
9   dyld                          	0x000000011757e78f dyld::_main(macho_header const*, unsigned long, int, char const**, char const**, char const**, unsigned long*) + 6237
10  dyld                          	0x00000001175784f6 dyldbootstrap::start(macho_header const*, int, char const**, long, macho_header const*, unsigned long*) + 1154
11  dyld                          	0x0000000117578036 _dyld_start + 54

Thread 0 crashed with X86 Thread State (64-bit):
  rax: 0x00000001175f9938  rbx: 0x00007fff8f2ef960  rcx: 0x00007ffee865eb30  rdx: 0x00007ffee865eac0
[... 289 lines cut ...]
TOTAL                              1.0G      528 
TOTAL, minus reserved VM space   633.4M      528 

Model: MacBookPro9,1, BootROM 222.0.0.0.0, 4 processors, Intel Core i7, 2.6 GHz, 16 GB, SMC 2.1f173
Graphics: Intel HD Graphics 4000, Intel HD Graphics 4000, Built-In
Graphics: NVIDIA GeForce GT 650M, NVIDIA GeForce GT 650M, PCIe
Memory Module: BANK 0/DIMM0, 8 GB, DDR3, 1333 MHz, 0x029E, 0x434D534F384758334D314131333333433920
Memory Module: BANK 1/DIMM0, 8 GB, DDR3, 1333 MHz, 0x029E, 0x434D534F384758334D314131333333433920
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0xD6), Broadcom BCM43xx 1.0 (7.21.190.32 AirPortDriverBrcm4360-1315.1)
Bluetooth: Version 6.0.9f2, 3 services, 27 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en1
Serial ATA Device: Crucial_CT1050MX300SSD1, 1.05 TB
Serial ATA Device: MATSHITADVD-R   UJ-8A8
USB Device: USB 2.0 Bus
USB Device: Hub
USB Device: FaceTime HD Camera (Built-in)
USB Device: USB 2.0 Bus
USB Device: Hub
USB Device: Hub
USB Device: Apple Internal Keyboard / Trackpad
USB Device: IR Receiver
USB Device: BRCM2070 Hub
USB Device: Bluetooth USB Host Controller
USB Device: USB 3.0 Bus
Thunderbolt Bus: MacBook Pro, Apple Inc., 25.1

mkeeter, 2019-01-08:
I assume this is the pre-built `Studio.app` bundle?

Can you try building from source to see if that works?

InformationOcean, 2019-01-09:
Hi Matt,

Yes, it is the pre-built.  On the weekend I will try.

mkeeter, 2019-08-02:
I've updated the pre-built `Studio.app` (at libfive.com/download) with many months of improvements or bug-fixes, so closing this issue.

closed 2019-08-02 by mkeeter