libfive/libfive #248: Attract variations: nomenclature?

omgitsraven, 2019-03-04:
I've written variations on the `attract` and `repel` functions that attract (and repel) to a line and a plane rather than just a point, but I'm uncertain about how best to handle their names and arguments, and thought I'd check what you think.

First, the names:
Should a function that attracts along the `x` axis towards a plane on the `yz` axes be `attract-x` or `attract-yz`? I'm tempted to label the axis that it affects (`attract-x`), but people might be thinking about it in terms of the attractor that's being defined (`attract-yz`).

Secondly, the argument:
For the point version of the function, the locus is a vec3 for obvious reasons. But for the line and the plane, there's one or two axes that are meaningless (i.e. the plane is an infinite yz plane, so only moving its X has any meaningful effect)... Should they still take a vec3 for the sake of more easily switching between modes to compare the result, or should they take a vec2 and float respectively, for the sake of not confusing people by asking for data that'll be ignored?

omgitsraven, 2019-03-04:
I did just notice that `reflect` uses the "`reflect-x` reflects the x axis on the yz plane" approach, so maybe it would be good to stick with that for question 1.

mkeeter, 2019-03-05:
Yup, I'd recommend sticking with the same behavior as `reflect-x` for naming.

My intuition for your second question is a bit inconsistent:  it's to take a single value for the plane function, and a full `vec3` (of which one coordinate is ignored) for the line function.

My reasoning is that a `vec2` expects to be `xy`, and has accessor functions to that effect. Expecting users to figure out that in some cases, it's actually `yz` or `xz` is counterintuitive (e.g. should it be `zx` to be right-handed?).  A single coordinate has no ambiguity (for the plane case), and is what we do for `reflect-x` right now.

Does that make sense?

omgitsraven, 2019-03-05:
I see what you mean, but I feel like the inconsistency of only *one* of them being different might be even more confusing—and if the last one took a vec3 too, then at least it would be easy to switch between the three functions to compare results, as an extra bonus. I think I'll just do all three as vec3, if that sounds alright!

closed 2019-03-05 by omgitsraven

mkeeter, 2019-03-05:
Sounds good 👍