libfive/libfive #254: Possible to upgrade to C++14 ?

Lorac, 2019-03-26:
It brings a couple of nice improvements such as std::make_shared/unique<>.

https://en.wikipedia.org/wiki/C%2B%2B14

mkeeter, 2019-03-26:
It's possible, but I'm being conservative – the only applicable feature that I've noticed is `std::make_unique`, and I don't know if that's enough to justify it.  Are there other C++14 features that would improve the `libfive` codebase?

Lorac, 2019-03-26:
The generic lambda is pretty nice, you don't need to specify the types anymore. The `make_unique` is pretty huge, because it prevents memory leaks since you don't have to deal with out-of-memory exception incase a `new` fails to allocate.

I don't know which distro you are using but I think most of the stable distro(even debian:stretch) have a compiler that supports C++14. Since you are using `ubuntu:latest` inside that dockerfile you don't really know which version of the compiler you are getting. I would suggest pinning down the version for that docker image to something stable.

What's the target for this library: embedded devices and/or desktop computers?

Do you know if it can cross-compile on arm?

mkeeter, 2019-03-26:
The library is targeted for the desktop, though that includes things like Raspberry Pi's and Beaglebones.  As far as I know, no one has tried cross-compiling on ARM yet.

Right now, I don't think this is worth the time to implement + validate + answer questions from folks whose ancient Linux distros don't support it – I already have trouble with systems that don't have full C++11 support.

closed 2019-03-26 by mkeeter