libfive/libfive #259: Small chance that Studio will fail to render on launch

omgitsraven, 2019-04-02:
### Description

Sometimes, Studio launches in a state that means that it will never display a mesh, just the spinning circle icon (and the axes) forever, even if you make changes to the code in the left pane. The only things that fix this are quitting and relaunching, or changing the meshing algorithm (at which point the original algorithm will finally work too, if you change back to it.)

This can happen when launching the application directly and typing into a new document, but given that it can take many attempts (feels like a 10% chance to me?) it's easier to just save a file like [spheretest.zip](https://github.com/libfive/libfive/files/3034052/spheretest.zip) and open/quit/open/quit... repeatedly until the bug happens.

### Versions

- Operating system: macOS 10.13.6
- `libfive` commit hash: 460717e1d16f6b1c3af25325e119878d25ce94ff (but it's been happening for I'd say a few months?)
  - Is this the latest commit to `master`?  yes
- Qt version (if applicable): 5.11.1

mkeeter, 2019-04-02:
I've noticed this too, but haven't had time to dig into it – thanks for opening an issue.

lshoravi, 2019-05-20:
This _always_ happends to me, no matter what.

cnasc, 2019-06-02:
Bringing the discussion from #275 into here:

> it's probably some kind of race condition on startup, because the script evaluation, UI, and shape rendering all happen in different threads!

Any ideas on likely places for this issue to manifest? Given that changing the meshing algorithm seems to reliably resolve it, I think that narrows it down to the UI and rendering modules (though I think it's probably more likely in rendering).

I'll try to look through the code today and build up a mental model of what happens when. I think the key will be to see what changes when you change rendering algorithms and tracing out from there.

mkeeter, 2019-07-12:
I've fixed one bug where the spinner can get stuck on if a render operation is cancelled midway through (in 4c480bc6d9b89bba3a6c6c261270ec2f94eb036e).  This doesn't explain why sometimes everything fails to render, though.

mkeeter, 2019-07-12:
Also, I managed to get a crash with this fascinating traceback, which could be related:

```
Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   com.impraxical.studio         	0x000000010e526e54 Shape::done() const + 4 (shape.cpp:257)
1   com.impraxical.studio         	0x000000010e53379c View::setShapes(QList<Shape*>) + 1500
2   com.impraxical.studio         	0x000000010e53f148 QtPrivate::FunctorCall<QtPrivate::IndexesList<0>, QtPrivate::List<QList<Shape*> >, void, void (View::*)(QList<Shape*>)>::call(void (View::*)(QList<Shape*>), View*, void**) + 152 (qlist.h:829)
3   org.qt-project.QtCore         	0x000000010f6f2d7b QMetaObject::activate(QObject*, int, int, void**) + 2219
4   com.impraxical.studio         	0x000000010e542457 Interpreter::gotShapes(QList<Shape*>) + 55
5   com.impraxical.studio         	0x000000010e531b38 QtPrivate::FunctorCall<QtPrivate::IndexesList<0>, QtPrivate::List<QList<Shape*> >, void, void (Interpreter::*)(QList<Shape*>)>::call(void (Interpreter::*)(QList<Shape*>), Interpreter*, void**) + 152 (qlist.h:829)
6   org.qt-project.QtCore         	0x000000010f6eb801 QObject::event(QEvent*) + 753
7   org.qt-project.QtWidgets      	0x000000010e9fef7d QApplicationPrivate::notify_helper(QObject*, QEvent*) + 269
8   org.qt-project.QtWidgets      	0x000000010ea00382 QApplication::notify(QObject*, QEvent*) + 594
9   org.qt-project.QtCore         	0x000000010f6c1fb4 QCoreApplication::notifyInternal2(QObject*, QEvent*) + 212
10  org.qt-project.QtCore         	0x000000010f6c31ee QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) + 878
```

scheming-pony, 2019-07-12:
A Git bisect procedure yielded this commit as (notwithstanding false negatives, since the bug is intermittent) possibly the problem:

8dc8dcbdb2b00c8d4fb1020aabbef0eea3a29689

Note its suspiciously apropos title "Add option to select meshing algorithm in GUI"!

HTH

omgitsraven, 2019-12-27:
For the record, for whatever difference this information makes: I'm now using Studio on Ubuntu, and this has gone from "occasional" to "literally every launch" (as mentioned in #275 ), just in case you needed more people to test this with.

SGSSGene, 2021-01-23:
This should be fixable with PR #370

mkeeter, 2021-01-23:
Yup, I agree that that should fix it, but feel free to reopen if it reoccurs!

closed 2021-01-23 by mkeeter