libfive/libfive #261: Planes aren't visible on their own under dual contouring

omgitsraven, 2019-04-02:
Try the following scene:
```
(half-space #[1 0 0])
(sphere 1)
```
The sphere is visible, but not the plane.

Now wrap both shapes in a union:
```
(union
    (half-space #[1 0 0])
    (sphere 1)
)
```
Now the plane is visible.

This holds for using a simple `(lambda-shape (x y z) x)` in place of `half-space` as well.
If you switch to Iso-simplex, the plane is visible in all cases.

This isn't hard to work around, but it seemed like enough of a strange behavior that it was worth mentioning.

mkeeter, 2019-04-03:
This is "working as intended" from the perspective of the Dual Contouring algorithm, but it's definitely confusing.

The simple fix is to take the intersection of the shape and the bounding box, which would clamp it and produce a cube in the half-plane case; I think the reason I haven't done this already is for fear of numerical issues (e.g. whether we'd have to make the clamping cube _slightly_ smaller than the bounding box for it to work).

omgitsraven, 2019-04-03:
Fair enough! Like I said, it's easy enough to work around (I just use a big cube for whatever I would've used the plane for) but I thought I'd check if it's a sign of a bug. If this is expected, so be it :)

closed 2019-04-03 by omgitsraven