libfive/libfive #262: Feedback on a recursive sequentiable for loop

omgitsraven, 2019-04-15:
I've just finally written out a generic version of a function that I've been using in some form or another for a while, and before I commit it to the binds, I just wanted to check:

1) is there some way of doing this that's already built into scheme, that I've just rewritten without realising it (which is a pretty common theme over the course of my learning it, heh)

2) is this a decent name, or is there something clearer that I could call it

The function basically passes its result to itself until a `for`-style loop completes, making that result by calling a function that takes the previous shape and the loop's current progress, for the sake of stringing a number of similar shapes into a `sequence` easily.

This is the code:
```
(define (call-sequence-ftb shape func from to by)
  (let ((result (func shape from)))
    (if (>= from to)
      result
      (call-sequence-ftb result func (+ from by) to by )
    )
  )
)
```

and here's an example of how it'd be used:
```
(define (slices shape i) (difference shape (box-centered #[0.8 1 0.1] [1 0 i])) )
(define (poles shape i) (union shape (cylinder 0.15 2 #[0 i 0])) )

(sequence
  (sphere 1)
  (call-sequence-ftb slices 0 0.5 0.25)
  (call-sequence-ftb poles -1 1 0.5)
)
```

also, while I think of it, one more question:

3) Is there any kind of "throw" functionality in scheme that I should implement, if someone accidentally passes a negative `by` value, to keep Studio from crashing?

mkeeter, 2019-04-17:
I really enjoy this kind of "find the standard functions that replicates a certain hand-rolled logic" question, and spent a while digging into this.

The `from to by` form is very similar to [`iota`](https://www.gnu.org/software/guile/manual/html_node/SRFI_002d1-Constructors.html), with a slightly different signature compared to `(iota count start step)`.  The repeated application looks a lot like some kind of [fold](https://www.gnu.org/software/guile/manual/html_node/SRFI_002d1-Fold-and-Map.html#SRFI_002d1-Fold-and-Map).

All together, we can replicate your example with a function `fold-shape` like this:
```
(define (fold-shape shape func vs)
  (fold (lambda (i shape) (func shape i))
        shape vs))

(sequence
  (sphere 1)
  (fold-shape slices (iota 3 0 0.25))
  (fold-shape poles (iota 6 -1 0.5))
)
```
In fact, this is a bog-standard `fold`; `fold-shape` and the inner `lambda` function are used to swap argument order, since the `libfive` standard library expects `shape` as the first argument.

This is also maximally flexible – if you wanted to keep the `from to by` form, you could write an `arange` function on top of `iota` (or manually), then call `(fold-shape slices (arange 0 0.5 0.25))`.

To make this work, I added `fold` and `iota` to the sandboxed libraries in f121cf158d8de0e7b7a46e92394970106d58c1f7.

As for your third question – the Scheme sandbox times out after a certain amount of computation, so I'd expect to see something like
```
In function with-resource-limits:
limit-exceeded: Time limit exceeded

Stack trace:
0: limit-exceeded
```
Unfortunately, the geometry kernel hits a stack overflow before that point, which is a separate bug that should be fixed.

Of course, you can add condition checking to your own Scheme functions; `(if condition (error "something went wrong"))` is the minimal example.

omgitsraven, 2019-04-17:
Ahh, nice!! I knew I couldn't be the first person who needed to do something like this, heh :)
I'll look deeper into `fold` and `iota` soon, thanks for the tip!
At this point, would you say that what's left of `fold-shape` is enough to justify adding to the binds, or would it be better put as maybe an example that would show it in context?
(I remember there being examples at one point, though I can't find them now...)

mkeeter, 2019-04-17:
In general, I've added new bindings (e.g. for `iota` and `fold`) into the sandbox whenever I need a particular Guile function (as long as it's seemingly safe) – it's easy to do, and if I need a function, other people probably do too.

`fold-shape` is borderline, but I'd vote for it to go into the libfive stdlib, since it's useful to have a `fold` operation that composes nicely with `sequence`.

Dunno what namespace it should go under.  Right now, `(libfive kernel)` is a bit crowded with non-kernel stuff; `(libfive util)` is a bit lonely, and this seems like a utility, so maybe it should go there.  At some point, we'll have to do a namespace cleanup and think a bit more about organization.

omgitsraven, 2019-04-17:
I'm trying implementing it, but it's telling me `Unbound variable: fold`?
I deleted the whole local project and downloaded it fresh and compiled it from scratch and it's still happening.

mkeeter, 2019-04-17:
You probably need to add
```
(use-modules (srfi srfi-1))
```
to wherever you're trying to add `fold-shape`

omgitsraven, 2019-04-18:
That worked! Thanks :)

closed 2019-04-18 by omgitsraven