libfive/libfive #265: Build fail on Debian

pdu4, 2019-04-21:
### Description

I am trying to install the latest version of studio from master and it fails with this error when I run make

```

[ 13%] Building CXX object libfive/src/CMakeFiles/five.dir/render/brep/contours.cpp.o
In file included from /opt/libfive-4-19/libfive/src/../include/libfive/render/brep/contours.hpp:13:0,
                 from /opt/libfive-4-19/libfive/src/render/brep/contours.cpp:15:
/opt/libfive-4-19/libfive/src/../include/libfive/render/brep/progress.hpp:51:10: error: ‘atomic_uint64_t’ in namespace ‘std’ does not name a type
     std::atomic_uint64_t counter;
          ^~~~~~~~~~~~~~~
In file included from /opt/libfive-4-19/libfive/src/../include/libfive/render/brep/brep.hpp:17:0,
                 from /opt/libfive-4-19/libfive/src/render/brep/contours.cpp:16:
/opt/libfive-4-19/libfive/src/../include/libfive/render/brep/per_thread_brep.hpp:28:39: error: expected ‘)’ before ‘&’ token
     PerThreadBRep(std::atomic_uint32_t& c) : c(c)
                                       ^
/opt/libfive-4-19/libfive/src/../include/libfive/render/brep/per_thread_brep.hpp:58:10: error: ‘atomic_uint32_t’ in namespace ‘std’ does not name a type
     std::atomic_uint32_t& c;
          ^~~~~~~~~~~~~~~
/opt/libfive-4-19/libfive/src/../include/libfive/render/brep/per_thread_brep.hpp: In member function ‘uint32_t Kernel::PerThreadBRep<N>::pushVertex(const Eigen::Matrix<float, N, 1>&)’:
/opt/libfive-4-19/libfive/src/../include/libfive/render/brep/per_thread_brep.hpp:34:26: error: ‘c’ was not declared in this scope
         const auto out = c.fetch_add(1);
                          ^
In file included from /opt/libfive-4-19/libfive/src/render/brep/contours.cpp:19:0:
/opt/libfive-4-19/libfive/src/../include/libfive/render/brep/dual.hpp: In static member function ‘static std::unique_ptr<typename M::Output> Kernel::Dual<N>::walk_(const Kernel::Root<typename M::Input>&, unsigned int, std::atomic_bool&, Kernel::ProgressCallback, std::function<M(Kernel::PerThreadBRep<N>&, int)>)’:
/opt/libfive-4-19/libfive/src/../include/libfive/render/brep/dual.hpp:257:5: error: ‘atomic_uint32_t’ is not a member of ‘std’
     std::atomic_uint32_t global_index(1);
     ^~~
/opt/libfive-4-19/libfive/src/../include/libfive/render/brep/dual.hpp:260:45: error: ‘global_index’ was not declared in this scope
         breps.emplace_back(PerThreadBRep<N>(global_index));
                                             ^~~~~~~~~~~~
libfive/src/CMakeFiles/five.dir/build.make:398: recipe for target 'libfive/src/CMakeFiles/five.dir/render/brep/contours.cpp.o' failed
make[2]: *** [libfive/src/CMakeFiles/five.dir/render/brep/contours.cpp.o] Error 1
CMakeFiles/Makefile2:142: recipe for target 'libfive/src/CMakeFiles/five.dir/all' failed
make[1]: *** [libfive/src/CMakeFiles/five.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
```

Older versions have installed successfully and I think all the dependencies are up to date.

### Versions

- Operating system: Debian
- `libfive` commit hash: 5ee3be3
  - Is this the latest commit to `master`?  [yes]
- Qt version (if applicable): 5

mkeeter, 2019-04-21:
It looks like Debian / GCC 6 don't have the appropriate typedefs, judging by [this Chromium patch](https://sources.debian.org/patches/chromium/73.0.3683.75-1/gcc6/atomic.patch/).

Can you make a similar patch and submit a PR?

pdu4, 2019-04-22:
Thanks
Those changes got it to compile and work

I am not sure exactly what you wanted, so i created a gist with the patch [here](https://gist.github.com/pdu4/51cd4a65524926cfa6ce51a961eaf2ef)

And I created a PR with the changes added

mkeeter, 2019-04-22:
Thanks for the PR!

Closing since this is fixed in #266

closed 2019-04-22 by mkeeter