libfive/libfive #274: libfive-studio automatic reload works only the first time

Alkeryn, 2019-05-29:
After saving the file(in vim) it reloads
after saving a second time it dosen't

Also it would be nice to be able to open it as a visualizer only without the code section since it is open in another editor

Also on a side note, how comes it is so fast compared to openscad ?
also are you using gpu ?
Other than that this project rocks so far

mkeeter, 2019-05-30:
Hi Alkeryn!  Next time, please fill out the issue template for ease of debugging.

The "Open as Viewer" option opens the file and checks the auto-reload button for you, though the script is still visible (because it may be useful to see the errors pane below it).  You can always drag the bar to collapse the script window.

I can't reproduce the autoreload behavior – when I use "Open as Viewer", then edit and save a file in Vim multiple times, it correctly reloads.  This may be an OS difference, and you could help debug by adding a few print statements to `Window::onAutoLoad`.

This is using a different underlying technology than OpenSCAD.  Shapes are represented by functions, which you can see an example of [here](http://libfive.com/examples/#scheme).  The function-to-mesh conversion is still on the CPU, for now.

scheming-pony, 2019-05-30:
Issue #197 has some discussion about stand-alone viewers and automatic reload.

@Alkeryn OpenSCAD uses CGAL as a geometry (meshing) engine with B-rep modeling.  Aside from the complexity of doing CSG operations on boundary triangles, it uses exact arithmetic ("exact computation paradigm") and can be quite slow with the tradeoff of increased accuracy and robustness.  Libfive uses F-rep modeling, which by itself is probably faster, and the mesh is generated with IEEE 754 floating point arithmetic--which is again faster, but can produce degeneracies which have to be addressed (by the library maintainer).

kastaren, 2020-07-31:
I think I may have a lead on this: When writing to an io-file from vis (https://github.com/martanne/vis) with "savemethod" set to "atomic" or "auto" I get precisely this scenario. After the first update I no longer get any updates in Studio whatever I try to do to the file, including trying to write to it from other applications.

Thing is atomic rewrites in vis actually writes to a backup file, then uses rename to replace the original with the backup when done. My guess is Studio will no longer follow when we are not writing to the same file. With "savemethod" set to "inplace" in vis everything works as intended.

In vim setting "backupcopy" to "no" or "auto" would probably lead to the same situation, and according to the manual "auto" is default for vim running on non Unix (http://vimdoc.sourceforge.net/htmldoc/options.html#'backupcopy').

mkeeter, 2020-07-31:
Great catch – I wonder if the file watcher ends up looking at the old file descriptor?

(This still isn't high on my to-do list, but that's a good lead that someone could follow up on!)

bsima, 2020-11-17:
In vim, `set backupcopy=yes` fixed this for me.

I would argue in favor of a `libfive-studio --viewer <file>` option that just showed the object and the error information. You could even overlay the error on top of the object window when there is an error, to save some space in the gui.

I'm not very familiar with the codebase but would definitely consider hacking on this.

bsima, 2020-11-30:
Alternatively, one could use inpekt3d as an independent viewer: https://gitlab.com/kavalogic-inc/inspekt3d

This is the method I'm using at the moment, I think it works just fine.

SGSSGene, 2021-01-23:
I have just discovered libfive.... It is an amazing piece of software.
I have tried to fix this issue: #369

closed 2021-01-25 by mkeeter