libfive/libfive #275: Shapes don't render in Studio on Ubuntu 19.04

cnasc, 2019-06-01:
### Description

#### Description of the issue
On Ubuntu 19.04, after having successfully built and launched Studio, the only output I ever see is the preexisting axes in the render window. A loading spinner shows up when I first edit a file, and never goes away.
#### Steps to reproduce
Build on 19.04, launch Studio, copy in an example from the project homepage, such as
```lisp
(set-bounds! [-10 -10 -10] [10 10 10])
(set-quality! 8)
(set-resolution! 10)

(difference (sphere 1 [0 0 0])
  (sphere 0.6 [0 0 0])
  (cylinder-z 0.6 2 [0 0 -1])
  (reflect-xz (cylinder-z 0.6 2 [0 0 -1]))
  (reflect-yz (cylinder-z 0.6 2 [0 0 -1])))

```
#### What did you expect to happen?  What happened?  All the time?
Expected to see rendered output of the shape. Nothing but the axes and loading spinner appear in the render pane. This happens 100% of the time for me.

[image]

#### Are there any error messages printed to the console?

Nothing is output to the console upon pasting in the above example, but if I interact with the menu items (File, Edit, ...) the following gets logged:

```
~/code/libfive/build/studio$ ./Studio
qt.glx: qglx_findConfig: Failed to finding matching FBConfig (8 8 8 8)
qt.glx: qglx_findConfig: Failed to finding matching FBConfig (1 8 8 8)
qt.glx: qglx_findConfig: Failed to finding matching FBConfig (1 1 8 8)
qt.glx: qglx_findConfig: Failed to finding matching FBConfig (1 1 1 8)
qt.glx: qglx_findConfig: Failed to finding matching FBConfig (1 1 1 8)
qt.glx: qglx_findConfig: Failed to finding matching FBConfig (1 1 1 8)
qt.glx: qglx_findConfig: Failed to finding matching FBConfig (1 1 1 8)
[etc.]
```

### Versions

- Operating system: Kubuntu 19.04
- `libfive` commit hash: 93a90c15520e7ca3ed13c61a4a51a108edaeba15
  - Is this the latest commit to `master`?  yes
- Qt version (if applicable): 5.12.2

### Additional Information

```$ lspci | grep VGA
00:02.0 VGA compatible controller: Intel Corporation HD Graphics 620 (rev 02)
```

### Common questions
#### If this is a behavior issue, does the test suite (`libfive/test/libfive-test`) pass?

3 failing tests, not sure if these are relevant to this issue.

```
$ ./libfive-test 

~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
libfive-test is a Catch v2.2.3 host application.
Run with -? for options                                                                                                           

-------------------------------------------------------------------------------
Cache: performance of commutative tree balancing
-------------------------------------------------------------------------------
/home/cjn/code/libfive/libfive/test/cache.cpp:411
...............................................................................

/home/cjn/code/libfive/libfive/test/cache.cpp:439: FAILED:
  REQUIRE( false )
with messages:
  std::chrono::duration_cast<std::chrono::microseconds>(dt) .count() := 900
  (0x384)
  std::chrono::duration_cast<std::chrono::microseconds>(dt_) .count() := 13529
  (0x34d9)

-------------------------------------------------------------------------------
Heightmap::render: Performance
-------------------------------------------------------------------------------
/home/cjn/code/libfive/libfive/test/heightmap.cpp:259
...............................................................................

benchmark name                                  iters   elapsed ns      average 
-------------------------------------------------------------------------------
sphere                                              1     92006033    92.006 ms 
Menger sponge                                       1   1082338240    1.08234 s 

-------------------------------------------------------------------------------
QEF::solveBounded
  Circle outside of region
-------------------------------------------------------------------------------
/home/cjn/code/libfive/libfive/test/qef.cpp:219
...............................................................................

/home/cjn/code/libfive/libfive/test/qef.cpp:245: FAILED:
  REQUIRE( sol.error > 0 )
with expansion:
  -0.0 > 0
with messages:
  sol.position := 1.5
  2.7
  sol.value := -0.9999999448
  sol.rank := 2
  sol.error := -0.0

-------------------------------------------------------------------------------
SimplexTree<3>: assignIndices with cell collapsing
-------------------------------------------------------------------------------
/home/cjn/code/libfive/libfive/test/simplex.cpp:319
...............................................................................

/home/cjn/code/libfive/libfive/test/simplex.cpp:346: FAILED:
  REQUIRE( (a.first - b.first).norm() < 1e-9 )
with expansion:
  0.3124999997 < 0.000000001
with messages:
  a.first := -0.625
  0.9375
   0.625
  b.first := -0.625
    1.25
   0.625
  k.first.first->region.lower.transpose() := -0.625  0.625      0
  k.first.first->region.upper.transpose() :=     0  1.25 0.625
  (a.first - b.first).norm() := 0.3124999997

-------------------------------------------------------------------------------
Solver::findRoot
  Sum-of-squares performance
-------------------------------------------------------------------------------
/home/cjn/code/libfive/libfive/test/solver.cpp:85
...............................................................................

benchmark name                                  iters   elapsed ns      average 
-------------------------------------------------------------------------------
SOS root                                            1        60101    60.101 µs 

/home/cjn/code/libfive/libfive/test/solver.cpp:116: FAILED:
  REQUIRE( res < 1e-3 )
with expansion:
  0.00206f < 0.001

-------------------------------------------------------------------------------
Abs and skew applied to Oracle: Render and compare (cube as oracle)
-------------------------------------------------------------------------------
/home/cjn/code/libfive/libfive/test/transformed_oracle.cpp:200
...............................................................................

/home/cjn/code/libfive/libfive/test/transformed_oracle.cpp:106: FAILED:
  REQUIRE( valid )
with expansion:
  false
with messages:
  point := 0
  0
  0
  oFeatures.size() := 6
  cFeatures.size() := 4

===============================================================================
test cases:    174 |    169 passed | 3 failed | 2 failed as expected
assertions: 333228 | 333223 passed | 3 failed | 2 failed as expected
```

#### If this is a graphics issue on Linux, report the results of updating your graphics drivers (and switching to binary graphics drivers if applicable).

Intel graphics, official drivers are in mainline kernel

#### If there are error messages in the console, have you searched for them and followed up on any promising leads?

Searching for the error in the console about the `FBConfig`  leads to a Stackoverflow post, the discussion in which makes it appear that this could possibly be related to Qt 5.12 on X in particular. Although since the error only appears when using the menu items, I suspect that may not be entirely relevant.

mkeeter, 2019-06-02:
The fact that the OpenGL window is rendering the axes means that OpenGL is baseline working.  Can you try toggling meshing algorithms (in the debug menu) to Simplex and back?  If that fixes it, then this could be the same as issue #259.

cnasc, 2019-06-02:
[image]

Aha, that did it. Looks like this is a dupe of #259 then. Thanks for the quick help!

mkeeter, 2019-06-02:
Cool, I'll close this as a duplicate.  If you're interested, I'd appreciate debugging help on this – it's probably some kind of race condition on startup, because the script evaluation, UI, and shape rendering all happen in different threads!

closed 2019-06-02 by mkeeter