libfive/libfive #276: Rotate around objects own axis?

lshoravi, 2019-06-02:
Hi! 

Is it possible to (easily) rotate around an objects own axis?

omgitsraven, 2019-06-02:
Performing the rotation before translating the object anywhere should do that (assuming that it's an object that's defined relative to its own center, so if you're using something like `box`, try `box-centered` instead).

If you're using the optional parameter to define the object's position, just don't do that, and move it there with a `move` outside of the `rotation` (which is outside of the shape).

lshoravi, 2019-06-03:
Right, that could work, but what I'd like is to map rotate on a list of objects in such a way that they rotate around their own axis individually, akin to rotating and then moving.

omgitsraven, 2019-06-03:
I'm not sure what you're describing, but I'm also not sure why you would need a different feature to allow doing this to a list vs doing it to a single object?

lshoravi, 2019-06-03:
The use case is that I create a list of items that are evenly spaced out and I want them all slanted in the same way, akin to rotating first and and moving later. It would be nice to have this feature as it would simplify the code, but I'm not skilled enough to do it myself, so I'll hold out for now.

lshoravi, 2019-06-19:
I found great workarounds, so I'm closing this.

closed 2019-06-19 by lshoravi