libfive/libfive #279: Build fails with boost 1.70

erikstrand, 2019-06-05:
### Description

Libfive fails to build against Boost 1.70. It ultimately instantiates a reference to void inside `boost::small_vector`. Using Boost 1.67, everything is fine. So I haven't looked into what's going on. Just thought you might want to know.

I noticed this when trying to build b390d84b8748e8aae6c6c68708022c1f65be957c (latest master) on a fresh EC2 instance (m5.xlarge running Amazon Linux 2). As a minor note, I also had to define the environment variable `PKG_CONFIG_PATH=/usr/local/share/pkgconfig` in order for cmake to find Eigen on this system. Oddly enough if I tell it to use `find_package` (as is currently done on windows) instead of `pkg_check_modules` it finds it fine without the environment variable.

### Additional Information

Here are the errors:

```
[ec2-user@ip-xx-xx-xx-xx build]$ make
[  0%] Built target five-version
[  1%] Building CXX object libfive/src/CMakeFiles/five.dir/render/brep/dc/dc_tree2.cpp.o
In file included from /usr/local/include/eigen3/Eigen/Core:368:0,
                 from /usr/local/include/eigen3/Eigen/StdVector:14,
                 from /home/ec2-user/libfive/libfive/src/render/brep/dc/dc_tree.cpp:17,
                 from /home/ec2-user/libfive/libfive/src/render/brep/dc/dc_tree2.cpp:10:
/usr/local/include/eigen3/Eigen/src/Core/util/Memory.h: In instantiation of ‘class Eigen::aligned_allocator<void>’:
/usr/local/include/boost/container/allocator_traits.hpp:136:48:   required from ‘struct boost::container::allocator_traits<Eigen::aligned_allocator<void> >’
/usr/local/include/boost/container/small_vector.hpp:85:7:   required from ‘class boost::container::small_vector_allocator<Kernel::Intersection<2>, Eigen::aligned_allocator<void> >’
/usr/local/include/boost/container/allocator_traits.hpp:136:48:   required from ‘struct boost::container::allocator_traits<boost::container::small_vector_allocator<Kernel::Intersection<2>, Eigen::aligned_allocator<void> > >’
/usr/local/include/boost/container/vector.hpp:756:76:   required from ‘class boost::container::vector<Kernel::Intersection<2>, boost::container::small_vector_allocator<Kernel::Intersection<2>, Eigen::aligned_allocator<void> >, void>’
/usr/local/include/boost/container/small_vector.hpp:334:7:   required from ‘class boost::container::small_vector_base<Kernel::Intersection<2>, Eigen::aligned_allocator<Kernel::Intersection<2> > >’
/usr/local/include/boost/container/small_vector.hpp:489:76:   required from ‘struct boost::container::small_vector_storage_definer<Kernel::Intersection<2>, Eigen::aligned_allocator<Kernel::Intersection<2> >, 4>’
/usr/local/include/boost/container/small_vector.hpp:511:7:   required from ‘class boost::container::small_vector<Kernel::Intersection<2>, 4, Eigen::aligned_allocator<Kernel::Intersection<2> > >’
/home/ec2-user/libfive/libfive/src/render/brep/dc/dc_tree.cpp:490:32:   required from ‘void Kernel::DCTree<N>::evalLeaf(Kernel::XTreeEvaluator*, Kernel::Tape::Handle, const Kernel::Region<N>&, Kernel::DCTree<N>::Pool&, const Kernel::DCNeighbors<N>&) [with unsigned int N = 2; Kernel::Tape::Handle = std::shared_ptr<Kernel::Tape>; Kernel::DCTree<N>::Pool = Kernel::ObjectPool<Kernel::DCTree<2>, Kernel::DCLeaf<2> >]’
/home/ec2-user/libfive/libfive/src/render/brep/dc/dc_tree2.cpp:50:16:   required from here
/usr/local/include/eigen3/Eigen/src/Core/util/Memory.h:728:27: error: forming reference to void
   typedef T&              reference;
                           ^~~~~~~~~
/usr/local/include/eigen3/Eigen/src/Core/util/Memory.h:729:27: error: forming reference to void
   typedef const T&        const_reference;
                           ^~~~~~~~~~~~~~~
make[2]: *** [libfive/src/CMakeFiles/five.dir/render/brep/dc/dc_tree2.cpp.o] Error 1
make[1]: *** [libfive/src/CMakeFiles/five.dir/all] Error 2
make: *** [all] Error 2
```

Here's what I ran to get the EC2 instance set up:

```
# package installs:
yum update -y
yum install -y git gcc gcc-c++ libpng-devel

# cmake:
cd ~
wget https://github.com/Kitware/CMake/releases/download/v3.14.5/cmake-3.14.5.tar.gz
tar -xzf cmake-3.14.5.tar.gz
cd cmake-3.14.5
rm cmake-3.14.5.tar.gz
mkdir build
cd build
../bootstrap --parallel=8
make -j8
make install
cd ~
rm cmake-3.14.5.tar.gz

# Eigen:
cd ~
wget http://bitbucket.org/eigen/eigen/get/3.3.7.tar.gz
tar -xzf 3.3.7.tar.gz
cd eigen-eigen-323c052e1731
mkdir build
cd build
cmake ..
make install
cd ~
rm 3.3.7.tar.gz
export PKG_CONFIG_PATH=/usr/local/share/pkgconfig

# Boost 1.70:
cd ~
wget https://dl.bintray.com/boostorg/release/1.70.0/source/boost_1_70_0.tar.gz
tar -xzf boost_1_70_0.tar.gz
cd boost_1_70_0
./bootstrap.sh
./b2 -j8
./b2 install
cd ~
rm boost_1_70_0.tar.gz

# libfive
cd ~
git clone https://github.com/libfive/libfive.git
cd libfive
mkdir build
cd build
cmake ..
make -j8
```

mkeeter, 2019-06-06:
Interesting!  Does it work if you remove the `aligned_allocator<Intersection>` from the template [here](https://github.com/libfive/libfive/blob/master/libfive/include/libfive/render/brep/dc/intersection.hpp#L39)?  I'm not sure if it's actually necessary: because we have `EIGEN_MAKE_ALIGNED_OPERATOR_NEW` in the `Intersection` class, using the aligned allocator may be unnecessary.

erikstrand, 2019-06-06:
Yup, that does the trick https://github.com/libfive/libfive/pull/282

mkeeter, 2019-06-06:
Merged, thanks!

closed 2019-06-06 by mkeeter