libfive/libfive #281: Does not build on Windows

omgitsraven, 2019-06-06:
There's already another issue about not building on Windows, but my error log is different enough from theirs that I don't want to confuse things in that thread.

I'm running Microsoft Visual Studio Community 2017, Version 15.7.2

I followed all the instructions in the readme, and the result was this:

```
1>------ Build started: Project: ZERO_CHECK, Configuration: Debug Win32 ------
1>Checking Build System
1>CMake does not need to re-run because C:/Users/Andrew/Documents/GitHub/libfive/build/CMakeFiles/generate.stamp is up-to-date.
1>CMake does not need to re-run because C:/Users/Andrew/Documents/GitHub/libfive/build/libfive/CMakeFiles/generate.stamp is up-to-date.
1>CMake does not need to re-run because C:/Users/Andrew/Documents/GitHub/libfive/build/libfive/src/CMakeFiles/generate.stamp is up-to-date.
1>CMake does not need to re-run because C:/Users/Andrew/Documents/GitHub/libfive/build/libfive/test/CMakeFiles/generate.stamp is up-to-date.
2>------ Build started: Project: five-version, Configuration: Debug Win32 ------
2>Building Custom Rule C:/Users/Andrew/Documents/GitHub/libfive/libfive/src/CMakeLists.txt
2>CMake does not need to re-run because C:/Users/Andrew/Documents/GitHub/libfive/build/libfive/src/CMakeFiles/generate.stamp is up-to-date.
3>------ Build started: Project: five, Configuration: Debug Win32 ------
3>Building Custom Rule C:/Users/Andrew/Documents/GitHub/libfive/libfive/src/CMakeLists.txt
3>CMake does not need to re-run because C:/Users/Andrew/Documents/GitHub/libfive/build/libfive/src/CMakeFiles/generate.stamp is up-to-date.
3>base.cpp
3>deck.cpp
3>eval_interval.cpp
3>eval_jacobian.cpp
3>eval_array.cpp
3>eval_deriv_array.cpp
3>eval_deriv.cpp
3>eval_feature.cpp
3>eval_point.cpp
3>tape.cpp
3>feature.cpp
3>heightmap.cpp
3>voxels.cpp
3>contours.cpp
3>edge_tables.cpp
3>manifold_tables.cpp
3>mesh.cpp
3>neighbor_tables.cpp
3>progress.cpp
3>intersection_aligner.cpp
3>Generating Code...
3>Compiling...
3>marching.cpp
3>dc_contourer.cpp
3>dc_mesher.cpp
3>dc_neighbors2.cpp
3>dc_neighbors3.cpp
3>dc_pool2.cpp
3>dc_pool3.cpp
3>dc_tree2.cpp
3>c:\users\andrew\documents\github\libfive\vcpkg\installed\x86-windows-static\include\eigen\src\core\util\memory.h(728): error C2182: 'reference': illegal use of type 'void'
3>c:\users\andrew\documents\github\libfive\vcpkg\installed\x86-windows-static\include\boost\container\allocator_traits.hpp(136): note: see reference to class template instantiation 'Eigen::aligned_allocator<U>' being compiled
3>        with
3>        [
3>            U=void
3>        ]
3>c:\users\andrew\documents\github\libfive\vcpkg\installed\x86-windows-static\include\boost\container\small_vector.hpp(86): note: see reference to class template instantiation 'boost::container::allocator_traits<Eigen::aligned_allocator<U>>' being compiled
3>        with
3>        [
3>            U=void
3>        ]
3>c:\users\andrew\documents\github\libfive\vcpkg\installed\x86-windows-static\include\boost\container\allocator_traits.hpp(136): note: see reference to class template instantiation 'boost::container::small_vector_allocator<T,Eigen::aligned_allocator<U>>' being compiled
3>        with
3>        [
3>            T=Kernel::Intersection<2>,
3>            U=void
3>        ]
3>c:\users\andrew\documents\github\libfive\vcpkg\installed\x86-windows-static\include\boost\container\vector.hpp(756): note: see reference to class template instantiation 'boost::container::allocator_traits<boost::container::small_vector_allocator<T,Eigen::aligned_allocator<U>>>' being compiled
3>        with
3>        [
3>            T=Kernel::Intersection<2>,
[... 171 lines cut ...]
4>c:\users\andrew\documents\github\libfive\libfive\test\simplex.cpp(435): error C2398: Element '1': conversion from 'double' to 'const float &' requires a narrowing conversion
4>c:\users\andrew\documents\github\libfive\libfive\test\simplex.cpp(435): error C2398: Element '2': conversion from 'double' to 'const float &' requires a narrowing conversion
4>c:\users\andrew\documents\github\libfive\libfive\test\simplex.cpp(435): error C2398: Element '3': conversion from 'double' to 'const float &' requires a narrowing conversion
4>c:\users\andrew\documents\github\libfive\libfive\test\simplex.cpp(482): error C2398: Element '3': conversion from 'double' to 'const float &' requires a narrowing conversion
4>c:\users\andrew\documents\github\libfive\libfive\test\simplex.cpp(566): error C2398: Element '1': conversion from 'double' to 'const float &' requires a narrowing conversion
4>c:\users\andrew\documents\github\libfive\libfive\test\simplex.cpp(566): error C2398: Element '2': conversion from 'double' to 'const float &' requires a narrowing conversion
4>c:\users\andrew\documents\github\libfive\libfive\test\simplex.cpp(566): error C2398: Element '3': conversion from 'double' to 'const float &' requires a narrowing conversion
4>c:\users\andrew\documents\github\libfive\libfive\test\simplex.cpp(585): error C2398: Element '3': conversion from 'double' to 'const float &' requires a narrowing conversion
4>c:\users\andrew\documents\github\libfive\libfive\test\simplex.cpp(673): error C2398: Element '1': conversion from 'double' to 'const float &' requires a narrowing conversion
4>c:\users\andrew\documents\github\libfive\libfive\test\simplex.cpp(673): error C2398: Element '2': conversion from 'double' to 'const float &' requires a narrowing conversion
4>c:\users\andrew\documents\github\libfive\libfive\test\simplex.cpp(673): error C2398: Element '3': conversion from 'double' to 'const float &' requires a narrowing conversion
4>solver.cpp
4>surface_edge_map.cpp
4>transformed_oracle.cpp
4>tree.cpp
4>voxels.cpp
4>xtree.cpp
4>mesh_checks.cpp
4>shapes.cpp
4>Generating Code...
4>Done building project "libfive-test.vcxproj" -- FAILED.
5>------ Skipped Build: Project: ALL_BUILD, Configuration: Debug Win32 ------
5>Project not selected to build for this solution configuration 
========== Build: 2 succeeded, 2 failed, 0 up-to-date, 1 skipped ==========
```

erikstrand, 2019-06-06:
Error C2182 (illegal use of type 'void') looks like the same problem as https://github.com/libfive/libfive/issues/279. Matt's suggestion worked for me -- does that error go away if you build https://github.com/libfive/libfive/pull/282?

For error C2398 it looks like your compiler wants the numbers in tests written as `1.3f` instead of `1.3`. This is weird but apparently Visual Studio has been this way for a while: https://stackoverflow.com/questions/33605632/error-83-error-c2398-conversion-from-double-to-float-requires-a-narrowing-c.

omgitsraven, 2019-06-06:
Your pull request took care of the first chunk of errors, yes! Thank you!!

Will try sticking `f`s on the constants now…

mkeeter, 2019-06-06:
Yeah, it looks like the error is caused when we're trying to create an `Eigen::Vector3f` with an initializer list: from reading those errors, it looks like `{0, 0, 0}` works fine, and `{0, 0, 0.1f}` works fine, but `{0, 0, 0.1}` throws that error.

I'd recommend changing the function signatures in `test/util/shapes.hpp` to take `Eigen::Vector2d`, since that's easier than scattering `f` everywhere we call `box`; then add `f` to any remaining sites with errors.

omgitsraven, 2019-06-06:
Changing every occurrence of `Eigen::Vector3f` to `Eigen::Vector2d` in `test/util/shapes.hpp` didn't make any of the narrowing conversion errors go away, and in fact introduced a new error claiming that `sphere` had ambiguous overloads

omgitsraven, 2019-06-06:
OK, adding `f` to any non-integers definitely got rid of the 'narrowing conversion' errors, thank you!

I've now just got two instances of a new error:

```
1>------ Build started: Project: five-version, Configuration: Debug Win32 ------
2>------ Build started: Project: libfive-test, Configuration: Debug Win32 ------
2>main.cpp
2>eval_feature.cpp
2>eval_point.cpp
2>hybrid_meshing.cpp
2>indexes.cpp
2>mesh.cpp
2>simplex.cpp
2>shapes.cpp
2>Generating Code...
2>marching.obj : error LNK2001: unresolved external symbol "public: static struct Kernel::MarchingTable<2> Kernel::MarchingTable<2>::mt" (?mt@?$MarchingTable@$01@Kernel@@2U12@A)
2>marching.obj : error LNK2001: unresolved external symbol "public: static struct Kernel::MarchingTable<3> Kernel::MarchingTable<3>::mt" (?mt@?$MarchingTable@$02@Kernel@@2U12@A)
2>C:\Users\Andrew\Documents\GitHub\libfive\build\Output\Debug\libfive-test.exe : fatal error LNK1120: 2 unresolved externals
2>Done building project "libfive-test.vcxproj" -- FAILED.
3>------ Skipped Build: Project: ALL_BUILD, Configuration: Debug Win32 ------
3>Project not selected to build for this solution configuration 
========== Build: 1 succeeded, 1 failed, 2 up-to-date, 1 skipped ==========
```

mkeeter, 2019-06-07:
This seems like an excuse to switch to a nicer singleton for the `Marching<N>` tables, which I've been meaning to do for a while – implemented in b709dbd92a585f6fc496d95ce4b082d9df1851d4, can you see if that fixes things?

omgitsraven, 2019-06-07:
It builds!!
```
Build: 3 succeeded, 0 failed, 1 up-to-date, 1 skipped
```

Now we're on to libfive-test.exe aborting shortly after launch:

```
OracleClause::serialize: no installed "ST"
  You may need to call OracleClause::install.

~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
libfive-test.exe is a Catch v2.2.3 host application.
Run with -? for options

-------------------------------------------------------------------------------
Archive::serialize
  With an oracle
-------------------------------------------------------------------------------
c:\users\andrew\documents\github\libfive\libfive\test\archive.cpp(108)
...............................................................................

c:\users\andrew\documents\github\libfive\libfive\test\archive.cpp(118): FAILED:
  REQUIRE( out.str() == expected )
with expansion:
  "T"""" "ST"  " == "T"""" "ST""hi"  "

OracleClause::deserialize: no installed "ST"
  You may need to call OracleClause::install.
Deserializer: failed to deserialize Oracle "ST"
Deserializer: expected tag == 'T' || tag == 't' at deserializer.cpp:49
Deserializer::deserializeString: expected opening "
Deserializer::deserializeString: expected opening "
Deserializer: expected op < Opcode::LAST_OP at deserializer.cpp:82
Assertion failed: false, file c:\users\andrew\documents\github\libfive\libfive\src\tree\opcode.cpp, line 63
```

I'm not sure if that's a matter for its own thread?

mkeeter, 2019-06-07:
Oh yeah, that's totally different (and not core functionality) – unless you're building custom Oracles, it's not a critical test failure.  I'll close this issue, since we've now fixed "building on Windows".

closed 2019-06-07 by mkeeter