libfive/libfive #283: How to build Studio on Windows?

omgitsraven, 2019-06-07:
Sorry if this is an obvious question, but I don't know my way around Visual Studio at all:

The readme says to "open `build\libfiv.sln` and build the solution", which now works... but it doesn't produce a `studio` folder in the `build` folder like it does on Mac.

Does Studio exist for Windows, and if it does, how do I build it?

mkeeter, 2019-06-19:
Not yet – the blocker is building Guile Scheme on Windows.

Here's my pile of links from last time I looked into it:
https://github.com/Madsy/guile-automatic-build (out of date / no longer working)
https://lists.gnu.org/archive/html/guile-user/2013-08/msg00073.html (earliest attempt)
https://lists.gnu.org/archive/html/guile-devel/2016-07/msg00052.html (more recent)
https://lists.nongnu.org/archive/html/guile-devel/2018-01/msg00008.html (my thread)

[Here](https://github.com/mkeeter/guile-mingw/) is my repo with patches, but it obviously never made it to a working state.

Guile is pleasant to work with and we have a bunch of code that uses it, but the cross-platform story is lacking.  Maybe switching to Chez / Chibi / Racket / Python would make sense – I'm not sure of the right path here.

We could also look at making Studio multi-lingual, but that's blocked by having a cross-language standard library format for shapes and transforms (in my unpublished mental roadmap).

andybak, 2020-03-14:
This project: https://github.com/zalo/libfive-unity

has a working build in dll form and a partial interface to Unity C#

It's not a lot but it does allow some access to libfive functionality from WIndows (and is obviously of great interest if you're already using C# or Unity)

I looked into exposing more functionality to C# but I got stuck fairly early on and never got a response to this issue: https://github.com/zalo/libfive-unity/issues/1