libfive/libfive #284: Mesh Quality

miho, 2019-06-19:
### Description

First of all, nice project! 

I was wondering whether it is to be expected that the exported meshes contain intersecting triangles or don't preserve sharp features. 

I fully understand that meshing iso surfaces is a hard task if sharp features should be preserved. Before I go into detail, I would like to know whether my expectations are too high or whether I have to do something special prior to exporting the mesh to improve the mesh quality.

With the tutorial.io file (export-to-stl) I experienced the following issues:

DC: ->       intersecting triangles
ISO: -> .    intersections & wrinkles
Hybrid: -> sharp feature detection fails and/or intersections 

I noticed you previously worked on the cubical marching squares algorithm. Why did you abandon it?

### Versions

- Operating system: macOS 10.14.x
- `libfive` commit hash: cf94bdb1cfc04a096c3179bbafd59920b88112a2
  - Is this the latest commit to `master`?  [yes]

miho, 2019-06-20:
Thanks for your detailed response. I agree that the CMS paper lacks some important details. In general I think that the authors of such papers should provide plain implementations of the core algorithms that are presented in the publications. 

The "escaping vertex" problem is indeed a weakness in the original paper. But the partial implementations that exist look promising. But unfortunately, they do lack a proper feature preservations. So there's more work to do.

Anyway, here's an example where the hybrid method causes trouble in the tutorial.io:

[image]

I could fix the feature detection just by increasing the resolution. The other problems remain at different resolution/quality levels.

I will try the suggested branch.

Have you tried manifold dual contouring? There's someone [[1]](https://www.reddit.com/r/dualcontouring/comments/3y4rhe/100_manifold_dual_contouring_implementation/) who claims he has created a fully working implementation. Unfortunately, it's not a clean minimalistic implementation. It's mixed into the visualization code. That's why I didn't try it yet.

[1]:

https://www.reddit.com/r/dualcontouring/comments/3y4rhe/100_manifold_dual_contouring_implementation/

mkeeter, 2019-06-20:
The DC implementation in `libfive` is actually manifold dual contouring 😃 

It's not a full implementation, because I'm not merging cells that contain multiple vertices – that's theoretically possible, but has to be handled with care to avoid breaking manifoldness guarantees.

Aveline67, 2019-10-29:
If meshing is hard, what about raytracing ? How hard would it be to raytrace the tree ?

doug-moen, 2019-10-29:
@Aveline67 My project [Curv](https://github.com/curv3d/curv) is quite similar to libfive, in that both use the signed distance field representation, but Curv directly raytraces the signed distance field on the GPU, instead of first creating a mesh.

Meshing is hard, in the sense that creating a watertight, manifold, defect free mesh, while supporting sharp feature detection, is a research problem that hasn't been solved in 20 years. Even with a perfect solution, you are still just creating a polyhedral approximation to your shape that doesn't capture fine details below the grid resolution.

Within certain limits, raytracing is a solved problem. You can accurately render a signed distance field, pixel perfect, at any zoom level. And the code is much easier to write than a good mesher.

It turns out that neither approach dominates the other. Libfive and Curv excel at rendering different classes of signed distance fields.

Libfive's main limitation is that the signed distance field is a closed form expression: no conditionals or iteration or recursion. The language subset for computing signed distance fields is not Turing complete.

Curv has a Turing complete distance field language. Iteration is required for most interesting 3D fractals, and is also needed for various noise primitives, such as fractal noise and cellular noise. Curv also supports full colour models and full colour 3D printing. This is stuff that greatly interests me, so Curv is a better tool for my use cases.

Curv's main limitation is that it uses Sphere Tracing as the ray tracing algorithm. This places a requirement on the signed distance function: it must be "Lipschitz(1)-continuous", which means the gradient cannot exceed 1 or the raytracer will overshoot the surface. This means Curv can't render many of the algebraic surfaces found in recreational math, such as the Klein bottle example in libfive. Other raytracing algorithms don't have the Lipshitz(1) requirement, but instead either require the SDF to be C1 continuous, or they use a grid and can't accurately render fine details, like Libfive.

caioaao, 2020-07-08:
I started exploring after an issue with importing an STL file I generated with libfive into Blender and thought it'd be relevant for the mesh quality discussion. This is what I get from FreeCAD analyzer:
[image]
the scheme file used to generate the STL: https://gist.github.com/caioaao/6aad00147096f05f0f20412dc693c010
(libfive commit hash: e1da2fb6c2171a399c9a109e99e83d704310a6e2)

It looks like it's not generating a fully manifold mesh or am I misunderstanding this?

ianmackenzie, 2022-03-29:
I ran into this today - I've been experimenting with trying to use `libfive` to generate tetrahedral meshes suitable for structural finite element analysis, but both [gmsh](https://gmsh.info/) and [tetgen](https://wias-berlin.de/software/tetgen/index.html) complain about self-intersecting triangles when I try to use them on the STL files output by `libfive`. 

I'm happy to contribute if there's a path towards generating non-self-intersecting meshes suitable for tetrahedralization, but based on the above discussion it sounds like it's not at all obvious how that could work - is `libfive` perhaps not the best fit for this use case?

If it's at all useful, the following test file shows a bunch of self-intersecting triangles when I open it in [MeshLab](https://www.meshlab.net/):

```lisp
(extrude-z (triangle [0 0] [20 0] [10 20]) 0 1)

(set-quality! 8)
(set-resolution! 10)
(set-bounds! [-10 -10 -10] [30 30 30])
```

[image]