libfive/libfive #285: Import other files as modules?

lshoravi, 2019-06-19:
Is it possible to define modules, and import other files as modules? I would like to separate different structures from each other and import them instead, but Studio thinks that import and define-module aren't valid symbols.

I'm running Opensuse TW, libfive/studio running on latest master, guile 2.9.2

mkeeter, 2019-06-19:
Not yet!

Right now, the Guile interpreter uses a sandboxed for safety using [`(ice-9 sandbox)`](https://www.gnu.org/software/guile/manual/html_node/Sandboxed-Evaluation.html), which doesn't bring in those symbols.  I don't know enough about the import system to say whether it could be implemented in a safe way – if you want to dig into this, I'd be happy to accept a PR.

lshoravi, 2019-06-26:
I'll look into it

Renha, 2019-11-20:
Hi! Since I've seen cqparts I'm a bit worried about studio usability for bigger assemblies. So how the progress?

SGSSGene, 2021-01-29:
Hey, I started looking into this (I am very unfamiliar with scheme/guile).
What I wonder is how important is the "safety" of the sandbox?

So one simple thing we could do is simply allow `load` in the sandbox, or maybe also `use-modules`. I could imagine that this is helpful, but it compromises safety a little, but I am not really sure if this even matters here.

SGSSGene, 2021-01-29:
So I added a solution that uses `load`. Checkout #373 . Any thoughts or suggestions for improvements?

mkeeter, 2021-01-30:
Thanks, I will check it out!

More broadly, I go back and forth about the importance of sandboxing.  It was one of the main motivations for using Guile (since Python is nigh-impossible to sandbox), but using Guile sets up an annoying high barrier to entry vs Python – for adoption, it might be better to support Python at the cost of the sandbox.

The right answer is probably some kind of cache which stores hashes of "trusted" files, and doesn't auto-evaluate anything that's untrusted (but once you mark it as trusted, accepts it and future edits to it).  It seems like it would be tricky to get right in the UI, so it hasn't been high on my list.