libfive/libfive #288: How do you set  variables with the C-Api ?

rsaccon, 2019-07-04:
I am using libfive from Rust, with automatically generated bindings ([bindgen](https://github.com/rust-lang/rust-bindgen)) to the C-Api. All is straightforward for simple things like defining simple shapes, manipulating them and rendering them. But now I would like to introduce variables and I have no clue how that can be done using the C-Api. I found [this issue](https://github.com/libfive/libfive/issues/202) which explains it well for C++, I am aware of the C-Api commands for variables, but I did not find any example or documentation on how to glue things together and got confused by the C-Api code comments: 
```
TODO:  Free variables are treated as zero
```

mkeeter, 2019-07-04:
Glad to see people using `libfive` from Rust!  I've been threatening to rewrite it in Rust for years, though it's still a ways off – without const generics, it's hard to replace Eigen for matrix math.

Right now, the C API doesn't expose any way to set variables.  If you want to open a PR, it should be pretty easy to add new functions of the form
```
libfive_tree_eval_f_vars(libfive_tree t, libfive_vec3 p,
                         libfive_id* var_ids, float* var_values,
                         size_t var_count);
```

rsaccon, 2019-07-04:
Thanks a lot. Let's see how far I get ...

rsaccon, 2019-07-16:
Ok, got it working, but had to dive a bit into the internals of libfive and encountered there more questions than answers. And my messy code needs some serious cleanup before I can open a PR.

I basically just extended the C Api with some functions for creating an Evaluator, updating its variables and deleting it (My understanding of the purpose of variables is, that if the tree structure does not get changed, the evaluators can be reused for re-rendering, as in Studio where one can mouse-drag things)

However you suggested to add a function `libfive_tree_eval_f_vars`. But that does not expose the Evaluator in order to reuse it. Even worse, I do not really understand the purpose of the `libfive_tree_eval_...` functions. I assume it is to detect whether a specific position or region is inside or outside the shape defined be the tree, but at the current stage of my exploration into libfive, I have not had the need yet for that functionality.

And one more question about the C Api: what is the function `libfive_tree_constant_vars` good for ? I tried to understand the studio source code. Are there anywhere variables turned into constants ?

mkeeter, 2019-07-17:
`libfive_tree_constant_vars` is used to help shape certain kinds of direct modeling.

For example, given this code

```
(define r #1)
(define x #0)

(difference
  (sphere r [x 0 0])
  (half-space [1 0 0] [x 0 0]))
```
This makes a cut-in-half sphere:
[image]

If you drag on the sphere's curved surface, both `r` and `x` will change (because the position of the sphere's surface is determined by both values).

Using `libfive_tree_constant_vars` (which is named `constant` in the scheme bindings), we can make a small modification:
```
(difference
  (sphere r [(constant x) 0 0])
  (half-space [1 0 0] [x 0 0]))
```
With this change in place, dragging on the sphere's curved surface will only change the value of `r`, not `x` (and dragging on the flat side will only change `x`).

rsaccon, 2019-07-17:
Ah, thanks a lot, now all makes sense !

rsaccon, 2019-07-19:
Made a [PR](https://github.com/libfive/libfive/pull/292). It just covers my current very basic needs, for rendering there is is only: 
```
bool libfive_evaluator_save_mesh(libfive_evaluator evaluator, libfive_region3 R, const char *f);
```
For the future I see three ways for possibly improving the Rust developer experience:

- Just continue to extend the current C-Api (that is what I did here) as needed.
- Use directly the C++ api. I tried to create bindings automatically with [bindgen](https://github.com/rust-lang/rust-bindgen), but could not get it working (for the bindings to C it works perfectly)
- Make a new C-Api, thinner but complete, just exposing data structures and ways to instantiate the C++Classes, but no or very little business logic in the C++ implementation.

Even better of course would be if `const generics` find its way into stable Rust and and the whole libfive would be rewritten in Rust, as you threatened above !

Feel free to close this, as my initial issue is resolved now.

mkeeter, 2019-07-19:
Yup, the C API right now is "whatever I needed to make `libfive-guile` bindings work".  I've been considering a rewrite, but it hasn't bubbled to the top of my priorities.  Thanks for the PR!

closed 2019-07-19 by mkeeter