libfive/libfive #289: Examples not working (graphics issue?)

kasbah, 2019-07-09:
### Description

[Description of the issue]

Examples are not working for me.

I tried to run this example:

```scheme
(set-bounds! [-10 -10 -10] [10 10 10])
(set-quality! 8)
(set-resolution! 10)

(difference (sphere 1 [0 0 0])
  (sphere 0.6 [0 0 0])
  (cylinder-z 0.6 2 [0 0 -1])
  (reflect-xz (cylinder-z 0.6 2 [0 0 -1]))
  (reflect-yz (cylinder-z 0.6 2 [0 0 -1])))
```
And it never renders. The spinner keeps spinning. No errors are reported.

The other [examples](https://libfive.com/examples/) don't seem to render anything intentionally. 

I tried copying some examples from the Studio page. 

```scheme
(set-bounds! [-10 -10 -10] [10 10 10])
(set-quality! 8)
(set-resolution! 10)
(define r (+ 1 2 3))
(sphere #[0 0 0] 10)
```
Gives errors

```
goops-error: No applicable method for #<<generic> - (10)> in call (- #<<shape> 55a92e110300> #[0 0 0])

Stack trace:
0: (sphere (vec3 0 0 0) 10)
```

[Steps to reproduce]
Try to run the examples  on my computer: Ubuntu 18.04 64-bit 

glxinfo: 

```
OpenGL version string: 3.0 Mesa 19.0.2
```
<details>

```

name of display: :0
display: :0  screen: 0
direct rendering: Yes
server glx vendor string: SGI
server glx version string: 1.4
server glx extensions:
    GLX_ARB_create_context, GLX_ARB_create_context_profile, 
    GLX_ARB_create_context_robustness, GLX_ARB_fbconfig_float, 
    GLX_ARB_framebuffer_sRGB, GLX_ARB_multisample, 
    GLX_EXT_create_context_es2_profile, GLX_EXT_create_context_es_profile, 
    GLX_EXT_fbconfig_packed_float, GLX_EXT_framebuffer_sRGB, 
    GLX_EXT_import_context, GLX_EXT_libglvnd, GLX_EXT_texture_from_pixmap, 
    GLX_EXT_visual_info, GLX_EXT_visual_rating, GLX_INTEL_swap_event, 
    GLX_MESA_copy_sub_buffer, GLX_OML_swap_method, GLX_SGIS_multisample, 
    GLX_SGIX_fbconfig, GLX_SGIX_pbuffer, GLX_SGIX_visual_select_group, 
    GLX_SGI_make_current_read, GLX_SGI_swap_control
client glx vendor string: Mesa Project and SGI
[... 544 lines cut ...]
  k.first.first->region.upper.transpose() :=     0  1.25 0.625
  (a.first - b.first).norm() := 0.3124999997

-------------------------------------------------------------------------------
Solver::findRoot
  Sum-of-squares performance
-------------------------------------------------------------------------------
/home/keb67/projects/bath/libfive/libfive/test/solver.cpp:85
...............................................................................

benchmark name                                  iters   elapsed ns      average 
-------------------------------------------------------------------------------
SOS root                                            1       383527   383.527 µs 

===============================================================================
test cases:    179 |    177 passed | 1 failed | 1 failed as expected
assertions: 334811 | 334809 passed | 1 failed | 1 failed as expected

[1]+  Done                    ./Studio  (wd: /home/keb67/projects/bath/libfive/build/studio)
(wd now: /home/keb67/projects/bath/libfive/build)
```

</details>

If this is a graphics issue on Linux, report the results of updating your graphics drivers (and switching to binary graphics drivers if applicable). (Can't do that right now).

scheming-pony, 2019-07-09:
FWIW:

I'm using Ubuntu 18.04.1--glxinfo gives: "OpenGL version string: 3.0 Mesa 18.0.5".  I'm running Revision: de40706 of Studio.

The first example you tried renders OK for me.  However, I get the same Goops error for the sphere example.

scheming-pony, 2019-07-10:
I updated to Revision: 652ab87 of Studio--I now get the same behavior as your first example.  No rendering, just the spinner.  Revision: de40706 is from Oct. 2018, sorry I don't have a newer working revision for Git bisection.

mkeeter, 2019-07-10:
I believe this is a duplicate of #259.  If the fix described there (of changing the algorithm) doesn't work, then re-open the issue.

The videos on the website are using an older build, and may not work exactly as written; I've added a comment to that effect, but don't have time to re-record them now.

closed 2019-07-10 by mkeeter